OpenIPC/firmware #1404: GK7205V210+sc223a (branded sc5239s) i2c not working

miguelangel-nubla, 2024-04-16:
Device [https://www.aliexpress.com/item/2251832668606248.html](https://www.aliexpress.com/item/2251832668606248.html) 4MP 48V POE, sensor branded as sc5239s which should be a sc223a.

Flashed with `burn` just fine, but the sensor is not detected on ipctool.

```
root@openipc-gk7205v210:~# dmesg
Booting Linux on physical CPU 0x0
Linux version 4.9.37 (runner@fv-az1501-923) (buildroot-gcc-13.2.0) #1 Tue Apr 16                                                                                               17:38:40 UTC 2024
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Goke GK7205V200 DEMO Board
cma zone is not set!
Memory policy: Data cache writeback
On node 0 totalpages: 8192
free_area_init_node: node 0, pgdat c05072d8, node_mem_map c1fbb000
  Normal zone: 64 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 8192 pages, LIFO batch:0
CPU: All CPU(s) started in SVC mode.
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: mem=32M console=ttyAMA0,115200 panic=20 rootfstype=squashfs                                                                                               root=/dev/mtdblock3 init=/init mtdparts=sfc:256k(boot),64k(env),2048k(kernel),5                                                                                              120k(rootfs),-(rootfs_data) mmz_allocator=gk
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 27056K/32768K available (3856K kernel code, 150K rwdata, 912K rodata, 19                                                                                              2K init, 181K bss, 5712K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc2800000 - 0xff800000   ( 976 MB)
    lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03cc3a8   (3857 kB)
      .init : 0xc04b2000 - 0xc04e2000   ( 192 kB)
      .data : 0xc04e2000 - 0xc0507ae0   ( 151 kB)
       .bss : 0xc0509000 - 0xc0536468   ( 182 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
Gic dist init...
arm_arch_timer: Architected cp15 timer(s) running at 50.00MHz (phys).
clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, m                                                                                              ax_idle_ns: 440795202655 ns
sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
Switching to timer-based delay loop, resolution 20ns
clocksource: arm,sp804: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 63                                                                                              7086815595 ns
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value calculated using timer frequency.. 100.0                                                                                              0 BogoMIPS (lpj=500000)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x40008200 - 0x40008258
devtmpfs: initialized
VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911                                                                                              2604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
Serial: AMBA PL011 UART driver
12040000.uart: ttyAMA0 at MMIO 0x12040000 (irq = 21, base_baud = 0) is a PL011 r                                                                                              ev2
console [ttyAMA0] enabled
12041000.uart: ttyAMA1 at MMIO 0x12041000 (irq = 22, base_baud = 0) is a PL011 r                                                                                              ev2
12042000.uart: ttyAMA2 at MMIO 0x12042000 (irq = 23, base_baud = 0) is a PL011 r                                                                                              ev2
ssp-pl022 12070000.spi: ARM PL022 driver, device ID: 0x00041022
ssp-pl022 12070000.spi: mapped registers from 0x12070000 to c286f000
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
[... 250 lines cut ...]
firmware:
  kernel: "4.9.37 (Tue Apr 16 17:38:40 UTC 2024)"
  toolchain: buildroot-gcc-13.2.0
```

Found https://github.com/OpenIPC/ipctool/issues/112 but seems i2c is not working

```
root@openipc-gk7205v210:~# i2cdetect -yr 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@openipc-gk7205v210:~# ipctool i2cdump 0x60 0x3100 0x310F
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
3100: FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF  FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF  |  ................
```

I restored the backup and everything works fine, so I guess hardware is ok.
Flashing back to OpenIPC same situation.

miguelangel-nubla, 2024-04-16:
Setting the sensor manually `fs_setenv sensor sc223a` results in:

```
Booting Linux on physical CPU 0x0
Linux version 4.9.37 (runner@fv-az915-692) (buildroot-gcc-13.2.0) #1 Tue Apr 16 17:38:58 UTC 2024
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Goke GK7205V200 DEMO Board
cma zone is not set!
Memory policy: Data cache writeback
On node 0 totalpages: 8192
free_area_init_node: node 0, pgdat c05072d8, node_mem_map c1fbb000
  Normal zone: 64 pages used for memmap
  Normal zone: 0 pages reserved
  Normal zone: 8192 pages, LIFO batch:0
CPU: All CPU(s) started in SVC mode.
pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
pcpu-alloc: [0] 0
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: mem=32M console=ttyAMA0,115200 panic=20 rootfstype=squashfs root=/dev/mtdblock3 init=/init mtdparts=sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data) mmz_allocator=gk
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 27056K/32768K available (3856K kernel code, 150K rwdata, 912K rodata, 192K init, 181K bss, 5712K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc2800000 - 0xff800000   ( 976 MB)
    lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03cc3a8   (3857 kB)
      .init : 0xc04b2000 - 0xc04e2000   ( 192 kB)
      .data : 0xc04e2000 - 0xc0507ae0   ( 151 kB)
       .bss : 0xc0509000 - 0xc0536468   ( 182 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
Gic dist init...
arm_arch_timer: Architected cp15 timer(s) running at 50.00MHz (phys).
clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
Switching to timer-based delay loop, resolution 20ns
clocksource: arm,sp804: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 637086815595 ns
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value calculated using timer frequency.. 100.00 BogoMIPS (lpj=500000)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x40008200 - 0x40008258
devtmpfs: initialized
VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
Serial: AMBA PL011 UART driver
12040000.uart: ttyAMA0 at MMIO 0x12040000 (irq = 21, base_baud = 0) is a PL011 rev2
console [ttyAMA0] enabled
12041000.uart: ttyAMA1 at MMIO 0x12041000 (irq = 22, base_baud = 0) is a PL011 rev2
12042000.uart: ttyAMA2 at MMIO 0x12042000 (irq = 23, base_baud = 0) is a PL011 rev2
ssp-pl022 12070000.spi: ARM PL022 driver, device ID: 0x00041022
ssp-pl022 12070000.spi: mapped registers from 0x12070000 to c286f000
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
clocksource: Switched to clocksource arch_sys_counter
NET: Registered protocol family 2
TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[... 158 lines cut ...]
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
goke-i2c 12060000.i2c: wait idle abort!, RIS: 0x611
```

nekromant, 2024-11-11:
```
ioctl SPI_IOC_WR_MODE err, value = 11 ret = -1
```
Seen this stuff on one of my cams. Most likely due to weird spi flash and an error during sensor autodetect. 
Try running fw_setenv sensor your_sensor
This may fix it.