OpenIPC/firmware #1408: Ingenic T21 - No network

fcollingwood, 2024-04-19:
After flashing, wired ethernet is not working anymore (Neither is wireless, it's not even showing in ifconfig but I don't care about that). I think the relevant error message from the following is this:

> Net:   ====>PHY not found!
> Board Net Initialization Failed
> Jz4775-9161
> Card did not respond to voltage select!

but I have no idea how to fix it. Happy to run any diag testing if required.

U-Boot SPL 2013.07-g90873ad (Mar 28 2024 - 10:45:51)
Timer init
CLK stop
PLL init
pll_init:399
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
fbdiv = 36 , refdiv = 1 , fdivq = 2 ,pllod = 1 range = 3
cppcr is 02300860
CPM_CPAPCR 0470484d
fbdiv = 38 , refdiv = 1 , fdivq = 2 ,pllod = 1 range = 3
cppcr is 02500860
CPM_CPMPCR 04a0484d
fbdiv = 50 , refdiv = 1 , fdivq = 2 ,pllod = 1 range = 3
cppcr is 03100860
CPM_CPVPCR 0310086d
cppcr 0x9a773310
apll_freq 864000000
mpll_freq 900000000
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 450000000
cclk  864000000
l2clk 432000000
h0clk 180000000
h2clk 180000000
pclk  90000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00d73011
T31_0x5: 00000007
T31_0x15: 0000000c
T31_0x4: 00000000
T31_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T31_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T31_118: 0000003c
T31_158: 0000003c
T31_190: 00000020
T31_194: 0000001e
jz-04 :  0x00000051
jz-08 :  0x000000a0
jz-28 :  0x00000024
DDR PHY init OK
INNO_DQ_WIDTH   :00000003
INNO_PLL_FBDIV  :00000014
INNO_PLL_PDIV   :00000005
INNO_MEM_CFG    :00000051
INNO_PLL_CTRL   :00000018
INNO_CHANNEL_EN :0000000d
INNO_CWL        :00000006
INNO_CL         :00000007
DDR Controller init
DDRC_STATUS         0x80000001
DDRC_CFG            0x0a288a40
DDRC_CTRL           0x0000011c
DDRC_LMR            0x00400008
DDRC_DLP            0x00000000
DDRC_TIMING1        0x040e0706
DDRC_TIMING2        0x02150607
DDRC_TIMING3        0x2006051b
DDRC_TIMING4        0x17240031
DDRC_TIMING5        0xff060405
DDRC_TIMING6        0x32150505
DDRC_REFCNT         0x00da4801
DDRC_MMAP0          0x000020fc
DDRC_MMAP1          0x00002400
DDRC_REMAP1         0x03020d0c
DDRC_REMAP2         0x07060504
DDRC_REMAP3         0x0b0a0908
DDRC_REMAP4         0x0f0e0100
DDRC_REMAP5         0x13121110
DDRC_AUTOSR_EN      0x00000000
sdram init finished
SDRAM init ok
board_init_r
image entry point: 0x80100000

U-Boot 2013.07-g90873ad (Mar 28 2024 - 10:45:51)

Board: ISVP (Ingenic XBurst T21 SoC)
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 476k for U-Boot at: 83f88000
Reserving 32832k for malloc() at: 81f78000
Reserving 32 Bytes for Board Info at: 81f77fe0
Reserving 124 Bytes for Global Data at: 81f77f64
Reserving 256k for boot params() at: 81f37f64
Stack Pointer at: 81f37f48
Now running in RAM - U-Boot at: 83f88000
MMC:   msc: 0
SF: Detected W25Q64
In:    serial
Out:   serial
Err:   serial
Net:   ====>PHY not found!
Board Net Initialization Failed
Jz4775-9161
Card did not respond to voltage select!

Press ENTER to interrupt boot in 1...

hafo821, 2024-09-25:
I do have a same problem, network stopped working after bootloader replacement 😶🙄

Oelraucher, 2024-10-09:
Same problem here, on a "Mibao" cam with T21 chip.

Pr0v0st, 2024-11-21:
same here. happy to help troubleshoot. Wansview W6.

fluffy-kaiju, 2025-10-24:
Find here the step I've done for the moment.
If I can do anything to help troubleshoot, don't hesitate to reach me.

### Flash without network access

I've successfully flash [fullflash_202507242127_GD25Q64CSIG_8MB_T21N_SC2335_MT7601U_OpenIPC.bin](https://github.com/flyrouter/flyrouter/releases/download/v2025.07.27/fullflash_202507242127_GD25Q64CSIG_8MB_T21N_SC2335_MT7601U_OpenIPC.bin) via [Serial connection](https://github.com/OpenIPC/wiki/blob/master/en/help-uboot.md#flashing-full-image-via-serial-connection) as we don't have any network access with the current firmware.

### eth0 (lan)

The camera automatically connected to my network by DHCP with the RJ45 port.

```
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP qlen 1000
    link/ether 00:05:dd:b1:b0:89 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.66/24 brd 192.168.0.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::205:ddff:feb1:b089/64 scope link 
       valid_lft forever preferred_lft forever
``` 

### Try to enable wlan0 (wifi)

I've tried to connect the camera to my wifi. First problem, I can't use `fw_setenv` 

```
root@openipc-t21:/etc/network/interfaces.d# fw_setenv wlanpass xxxx
MTD erase error on /dev/mtd0: Invalid argument
Error: can't write fw_env to flash
```

So I harcoded my credentials in `/etc/network/interfaces.d/wlan0`

```
iface wlan0 inet dhcp
    #pre-up wpa_passphrase "$(fw_printenv -n wlanssid)" "$(fw_printenv -n wlanpass)" > /tmpf
    pre-up wpa_passphrase "retracted" "retracted" > /tmp/wpa_supplicanf
    pre-up sed -i 's/#psk.*/scan_ssid=1/g' /tmp/wpa_supplicant.conf
    pre-up wpa_supplicant -B -i wlan0 -D nl80211,wext -c /tmp/wpa_supplicant.conf
    post-down killall -q wpa_supplicant

```

But it failed to start up :/

```
root@openipc-t21:/etc/network/interfaces.d# ifup wlan0 
Successfully initialized wpa_supplicant
nl80211: 'nl80211' generic netlink not found
Failed to initialize driver 'nl80211'
wlan0: Unsupported driver 'nl80211,wext'
wlan0: CTRL-EVENT-DSCP-POLICY clear_all
```

### Web interface

The web interface is accessible. But the camera previews show nothing :/

[image]
[image]
[image]
[image]

closed 2026-09-26 by openipc-ai