OpenIPC/firmware #1412: hi3516cv300 support is broken

carbofos, 2024-04-21:
На нескольких платах Longse после обновления с обнулением оверлея перестал работать h264/h265/mp4.  При этом ffplay просто подвисает как это происходит при попытке воспроизводить отключенный stream1. 

Однако по HTTP отлично работает MJPEG.

Две платы с разными сенсорами ведут себя одинаково: imx291 и imx323.
Я выяснил, даты когда эта платформа была сломана. Причем ломалась она в два этапа:
14 февраля 2024 все работает 
15 февраля стрим всё еще работает, но жутко тормозит и рассыпается на артефакты
16 февраля ничего не работает кроме MJPEG.

Лог рабочей системы:
```
17:07:58  <       majestic> [    main] main@125                      Majestic Ultimate for HiSilicon, version HEAD+1c1718e, built on 2022-11-05
17:07:58  <       majestic> [app_conf] load_config@89                Using /etc/majestic.yaml as main configuration
17:07:58  <       majestic> [watchdog] watchdog_start@101            Detected Hisilicon Watchdog
17:07:58  <       majestic> [watchdog] watchdog_start@121            Watchdog timeout set to 10 seconds
17:07:59  <       majestic> [     sdk] start_sdk@402                 Using /etc/sensors/imx290_i2c_lvds_1080p_line.ini sensor configuration file
17:07:59  <       majestic> [     sdk] start_sdk@411                 App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
17:07:59  <       majestic> [     sdk] start_sdk@414                 Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
17:07:59  <      hi_logger> [    venc] VENC_VpssSend@1545            venc check create:venc 1 chn not create!
17:07:59  <      hi_logger> [    vpss] VpssUserGetFrame@365          vpss(0, 0) user get frame time out, s32MilliSec=10000
17:07:59  <      hi_logger> [    vpss] VpssUserGetFrame@365          vpss(0, 0) user get frame time out, s32MilliSec=10000
17:07:59  <      hi_logger> [    vpss] VpssUserGetFrame@365          vpss(0, 0) user get frame time out, s32MilliSec=10000
17:07:59  <      hi_logger> [     viu] VIU_DRV_DisableChn@1344       !! Disable ViChn:0 timeout 120ms!!!
17:07:59  <       majestic> [     sdk] start_sdk@439                 sensor imx290
17:07:59  <       majestic> [     sdk] start_sdk@444                   input_mode LVDS, WDR NONE
17:07:59  <       majestic> [     sdk] start_sdk@451                   dev [1920x1080]@0x30 25fps, GBRG
17:07:59  <       majestic> [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx290_i2c_lvds.so
17:07:59  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem before allocation: 90028KB
17:07:59  <       majestic> [     sdk] dump_vb_configuration@2135    VB configuration:
17:07:59  <       majestic> [     sdk] dump_vb_configuration@2145      [0]: 3110400 x 9
17:07:59  <       majestic> [     sdk] dump_vb_configuration@2145      [1]: 3110400 x 1
17:07:59  <       majestic> [     sdk] dump_vb_configuration@2145      [2]: 622080 x 1
17:07:59  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem after allocation: 59028KB
17:07:59  <       majestic> [     sdk] init_sensor@2330              Sensor driver loaded
17:07:59  <       majestic> [     sdk] init_vi@2738                  SDK is in 'offline' mode
17:07:59  <       majestic> [    puts]                               linear mode
17:07:59  <       majestic> [    puts]                               ===IMX291 1080P 30fps 12bit LINE Init OK!===
17:07:59  <       majestic> [     sdk] log_venc_chn@1244             H.265 1920x1080 25fps 4096Kbit  25 GOP
17:07:59  <       majestic> [     sdk] create_vpss_chn@991             new venc: 0   vpss_grp: 0,   vpss_chn: 0
17:07:59  <     iq_monitor> [     sdk] monitor_thread@331            Loading IQ profile /etc/sensors/iq/default.ini
17:07:59  <       majestic> [     sdk] start_video@1885                venc took 8148KB of MMZ [1518]
17:07:59  <       majestic> [     sdk] log_venc_chn@1244             H.265 704x576 15fps 1024Kbit  15 GOP
17:07:59  <       majestic> [     sdk] create_vpss_chn@991             new venc: 1   vpss_grp: 0,   vpss_chn: 1
17:07:59  <       majestic> [     sdk] start_video@1885                venc took 1632KB of MMZ [297]
17:07:59  <       majestic> [image_tu] start_image_params_tuning@63  Image tuning task started
17:07:59  <       majestic> [     sdk] start_sdk@546                 HiSilicon SDK started
17:07:59  <       majestic> [     hal] hisi_free_mem_ex@755          Free MMZ mem finally: 44960KB
17:07:59  <       majestic> [   httpd] new_http_server@370           HTTP server started on 0.0.0.0:80
17:07:59  <       majestic> [    rtsp] rtsp_init@31                  RTSP server started on port 554
17:07:59  <       majestic> [   netip] netip_start@2051              NETIP server started on port 34567

```

Лог поломанной системы
```
17:18:29  <       majestic> [    main] main@240                      Majestic Ultimate for HiSilicon, version master+253341c, built on 2024-04-21
17:18:30  <       majestic> [watchdog] watchdog_start@91             Detected Hisilicon Watchdog
17:18:30  <       majestic> [watchdog] watchdog_start@112            Watchdog timeout set to 300 seconds
17:18:30  <       majestic> [   httpd] start_web_server@379          Set server path: /var/www
17:18:30  <       majestic> [   httpd] new_http_server@367           HTTP server started on 0.0.0.0:80
17:18:30  <       majestic> [    main] init_sdk@66                   Loading main configuration: /etc/majestic.yaml
17:18:30  <       majestic> [     sdk] start_sdk@395                 Using /etc/sensors/imx291_i2c_lvds_1080p_line.ini sensor configuration file
17:18:30  <       majestic> [     sdk] start_sdk@400                 App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
17:18:30  <       majestic> [     sdk] start_sdk@405                 Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
17:18:30  <       majestic> [     sdk] start_sdk@426                 sensor imx291
17:18:30  <       majestic> [     sdk] start_sdk@429                   input_mode LVDS, WDR NONE
17:18:30  <       majestic> [     sdk] start_sdk@434                   dev [1920x1080]@0x30 25fps, GBRG
17:18:30  <       majestic> [  sensor] try_to_load@19                trying to load /usr/lib/sensors/libsns_imx291_i2c_lvds.so
17:18:30  <       majestic> [     hal] hisi_free_mem_ex@752          Free MMZ mem before allocation: 90028KB
17:18:30  <       majestic> [     sdk] dump_vb_configuration@2266    VB configuration:
17:18:30  <       majestic> [     sdk] dump_vb_configuration@2269      [0]: 3136320 x 10
17:18:30  <       majestic> [     hal] hisi_free_mem_ex@752          Free MMZ mem after allocation: 59392KB
17:18:30  <       majestic> [     sdk] init_sensor@2461              Sensor driver loaded
17:18:30  <       majestic> [     sdk] init_vi@2909                  SDK is in 'offline' mode
17:18:30  <       majestic> [    puts]                               linear mode
17:18:30  <       majestic> [    puts]                               ===IMX291 1080P 30fps 12bit LINE Init OK!===
17:18:30  <       majestic> [     sdk] log_venc_chn@1245             h264 1920x1080@20fps 4096Kbit 20GOP
17:18:30  <     iq_monitor> [     sdk] monitor_thread@317            Loading IQ profile /etc/sensors/iq/default.ini
17:18:30  <       majestic> [     sdk] create_vpss_chn@990             venc_chn: 0, vpss_grp: 0, vpss_chn: 0
17:18:30  <       majestic> [     hal] HiSi_HAL_VENC_CreateChn@245   Cannot create VENC chn 0
17:18:30  <       majestic> [     hal] HiSi_HAL_VENC_CreateChn@245   ERR_VENC_ILLEGAL_PARAM: The parameter is invalid
17:18:30  <       majestic> [     sdk] start_sdk@542                 HiSilicon SDK started
17:18:30  <       majestic> [     hal] hisi_free_mem_ex@752          Free MMZ mem finally: 55104KB
17:18:31  <       majestic> [    rtsp] rtsp_init@28                  RTSP server started on port 554

```

skilur, 2024-04-21:
Kindly show the log of `cat /dev/logmpp`

flyrouter, 2024-07-06:
Please check last release

carbofos, 2024-09-14:
> Please check last release

I did update this platform to master 2024-09-13 on two cameras and now everything is perfect.
Thank you!

closed 2024-09-14 by carbofos