OpenIPC/firmware #1418: hi3516ev100 lite 16m no match  sensor driver cause mejestic wrong

catulacnpoly, 2024-04-26:
sensor is sc2235_i2c

"majestic[854]: Majestic Lite for HiSilicon, version master+e066cc6, built on 2024-04-22
majestic[854]: Detected Hisilicon Watchdog
majestic[854]: Watchdog timeout set to 300 seconds
majestic[854]: Set server path: /var/www
majestic[854]: HTTP server started on 0.0.0.0:80
majestic[854]: Loading main configuration: /etc/majestic.yaml
majestic[854]: Autodetected sensor as 'sc2235_i2c'
majestic[854]: No matches for 'sc2235_i2c' in /etc/sensors directory
majestic[854]: Cannot start SDK"

list in etc/sensors:

"root@openipc-hi3516ev100:/etc/sensors# ls
ar0237_i2c_dc_1080p_line.ini    imx385_i2c_1080p.ini
imx290_i2c_lvds_1080p_line.ini  iq
imx291_i2c_lvds_1080p_line.ini  jxf22_i2c_dc_1080p_line.ini
imx307_i2c_lvds_1080p_line.ini  ov2718_i2c_1080p.ini
imx307_i2c_mipi_1080p_line.ini  ov2735_i2c_1080p.ini
imx323_i2c_dc_1080p_line.ini    sc2235p_i2c_dc_1080p_line.ini
imx323_spi_dc_1080p_line.ini    sc2310_i2c_dc_1080p.ini
imx327_i2c_lvds_1080p_line.ini"

"sc2235p_i2c_dc_1080p_line.ini" did not match  "sc2235_i2c"

catulacnpoly, 2024-04-26:
I curl the sc2235_i2c_dc_1080p.ini libsns_sc2235.so from hi3561cv200 pakage,and then the  follow::

majestic[802]: Majestic Lite for HiSilicon, version master+e066cc6, built on 2024-04-22
majestic[802]: Detected Hisilicon Watchdog
majestic[802]: Watchdog timeout set to 300 seconds
majestic[802]: Set server path: /var/www
majestic[802]: HTTP server started on 0.0.0.0:80
majestic[802]: Loading main configuration: /etc/majestic.yaml
majestic[802]: SENSOR=sc2235
majestic[802]: matched sensor config: sc2235_i2c_dc_1080p.ini
majestic[802]: Using /etc/sensors/sc2235_i2c_dc_1080p.ini sensor configuration file
majestic[802]: App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
majestic[802]: Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
majestic[802]: sensor sc2235
majestic[802]:   input_mode CMOS, WDR NONE
majestic[802]:   dev [1920x1080]@0x0 30fps, BGGR
majestic[802]: trying to load /usr/lib/sensors/libsns_sc2235.so
majestic[802]: Free MMZ mem before allocation: 32292KB
majestic[802]: VB configuration:
majestic[802]:   [0]: 3110400 x 3
majestic[802]:   [1]: 4147200 x 1
majestic[802]: Free MMZ mem after allocation: 19116KB
majestic[802]: Sensor driver loaded
majestic[802]: open vi(0,0) err, ret:-1 
majestic[802]: Cannot set VI dev 0 attr
majestic[802]: ERR_VI_SYS_NOTREADY: The VI system is not initialized
majestic[802]: Cannot init VI
majestic[802]: Cannot start SDK

widgetii, 2024-04-26:
hi3516cv300 and hi3516cv200 are different platforms, their sensor drivers are incompatible

catulacnpoly, 2024-04-27:
> hi3516cv300 and hi3516cv200 are different platforms, their sensor drivers are incompatible

what's the diffrence between sc2235 and sc2235p,I use the sc2235p driver of cv300 got the same error log。

majestic[805]: Majestic Lite for HiSilicon, version master+e066cc6, built on 2024-04-22
majestic[805]: Detected Hisilicon Watchdog
majestic[805]: Watchdog timeout set to 300 seconds
majestic[805]: Set server path: /var/www
majestic[805]: HTTP server started on 0.0.0.0:80
majestic[805]: Loading main configuration: /etc/majestic.yaml
majestic[805]: SENSOR=sc2235
majestic[805]: matched sensor config: sc2235p_i2c_dc_1080p_line.ini
majestic[805]: Using /etc/sensors/sc2235p_i2c_dc_1080p_line.ini sensor configuration file
majestic[805]: App was built with MPP version: Hi3516CV300_MPP_V1.0.4.0.B050 Release
majestic[805]: Current MPP version: HI_VERSION=Hi3516CV300_MPP_V1.0.4.0 B050 Release
majestic[805]: sensor sc2235p
majestic[805]:   input_mode CMOS, WDR NONE
majestic[805]:   dev [1920x1080]@0x0 25fps, BGGR
majestic[805]: trying to load /usr/lib/sensors/libsns_sc2235p_i2c_dc.so
majestic[805]: Free MMZ mem before allocation: 32292KB
majestic[805]: VB configuration:
majestic[805]:   [0]: 3110400 x 3
majestic[805]:   [1]: 4147200 x 1
majestic[805]: Free MMZ mem after allocation: 19116KB
majestic[805]: Sensor driver loaded
majestic[805]: open vi(0,0) err, ret:-1 
majestic[805]: Cannot set VI dev 0 attr
majestic[805]: ERR_VI_SYS_NOTREADY: The VI system is not initialized
majestic[805]: Cannot init VI
majestic[805]: Cannot start SDK

catulacnpoly, 2024-04-27:
root@openipc-hi3516ev100:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.18.20 (runner@fv-az651-175) (buildroot-gcc-13.2.0                                                                             ) #2 Thu Apr 25 02:10:11 UTC 2024
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=0005317f
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine model: Hisilicon Hi3516CV300 DEMO Board
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 8192
[    0.000000] free_area_init_node: node 0, pgdat c045a388, node_mem_map c1fba00                                                                             0
[    0.000000]   Normal zone: 64 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 8192 pages, LIFO batch:0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pag                                                                             es: 8128
[    0.000000] Kernel command line: mem=32M console=ttyAMA0,115200 panic=20 root                                                                             =/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=hi_sfc:256k(boot),64k(en                                                                             v),3072k(kernel),10240k(rootfs),-(rootfs_data)
[    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Memory: 27668K/32768K available (3129K kernel code, 171K rwdata,                                                                              976K rodata, 148K init, 239K bss, 5100K reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xc2800000 - 0xff000000   ( 968 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc040aa74   (4107 kB)
[    0.000000]       .init : 0xc040b000 - 0xc0430000   ( 148 kB)
[    0.000000]       .data : 0xc0430000 - 0xc045ae20   ( 172 kB)
[    0.000000]        .bss : 0xc045ae20 - 0xc0496d60   ( 240 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] VIC @c2800000: id 0x00641190, vendor 0x41
[    0.000035] sched_clock: 32 bits at 3000kHz, resolution 333ns, wraps every 14                                                                             31655765682ns
[    0.000808] Console: colour dummy device 80x30
[    0.000866] Calibrating delay loop... 398.13 BogoMIPS (lpj=1990656)
[    0.050024] pid_max: default: 32768 minimum: 301
[    0.050194] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.050214] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.051425] Initializing cgroup subsys devices
[    0.051471] Initializing cgroup subsys freezer
[    0.051497] Initializing cgroup subsys blkio
[    0.051556] CPU: Testing write buffer coherency: ok
[    0.052248] Setting up static identity map for 0x802fa6f0 - 0x802fa748
[    0.054324] devtmpfs: initialized
[    0.056036] VFP support v0.3: not present
[    0.056896] pinctrl core: initialized pinctrl subsystem
[    0.065214] NET: Registered protocol family 16
[    0.066676] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.083274] Serial: AMBA PL011 UART driver
[    0.083649] 12100000.uart: ttyAMA0 at MMIO 0x12100000 (irq = 21, base_baud =                                                                              0) is a PL011 rev2
[    0.356906] console [ttyAMA0] enabled
[    0.388120] SCSI subsystem initialized
[    0.392135] ssp-pl022 12120000.spi: ARM PL022 driver, device ID: 0x00800022
[    0.399210] ssp-pl022 12120000.spi: mapped registers from 0x12120000 to c286c                                                                             000
[    0.431798] usbcore: registered new interface driver usbfs
[    0.437463] usbcore: registered new interface driver hub
[    0.442979] usbcore: registered new device driver usb
[    0.450321] Switched to clocksource arm,sp804
[    0.478437] NET: Registered protocol family 2
[    0.484687] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.491794] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.498210] TCP: Hash tables configured (established 1024 bind 1024)
[    0.504760] TCP: reno registered
[    0.508007] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.513937] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.520642] NET: Registered protocol family 1
[    0.526063] RPC: Registered named UNIX socket transport module.
[... 165 lines cut ...]
[  126.154561] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  126.154754] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  126.154946] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[... cut ...]
[  126.178258] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  126.178451] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  126.178638] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  126.178826] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  126.179014] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  126.179208] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  126.179398] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611
[  126.179587] hibvt-i2c 12110000.i2c: wait idle abort!, RIS: 0x611

catulacnpoly, 2024-04-27:
dmesg after first boot

catulacnpoly, 2024-04-27:
root@openipc-hi3516ev100:~# ipctool
The ipctool installed as remote GitHub plugin
---
chip:
  vendor: HiSilicon
  model: 3516EV100
board:
  vendor: OpenIPC
  version: 2.4.04.25
ethernet:
  mac: "32:10:81:4b:6a:cb"
  u-mdio-phyaddr: 1
  phy-id: 0x02430c54
  d-mdio-phyaddr: 0
rom:
- type: nor
  block: 64K
  partitions:
    - name: boot
      size: 0x40000
      sha1: e210b643
    - name: env
      size: 0x10000
      sha1: 667429e1
      contains:
        - name: uboot-env
          offset: 0x0
    - name: kernel
      size: 0x300000
      sha1: 3b905d0c
    - name: rootfs
      size: 0xa00000
      path: /,squashfs
      sha1: c1244305
    - name: rootfs_data
      size: 0x2b0000
      path: /overlay,jffs2,rw
  size: 16M
  addr-mode: 3-byte
ram:
  total: 32M
firmware:
  u-boot: "2010.06 (Nov 14 2022 - 19:27:26)"
  kernel: "3.18.20 (Thu Apr 25 02:10:11 UTC 2024)"
  toolchain: buildroot-gcc-13.2.0
sensors:
- vendor: SmartSens
  model: SC2235
  control:
    bus: 0
    type: i2c
    addr: 0x60
  vicap-state: down

closed 2026-11-16 by openipc-ai