OpenIPC/firmware #1422: Wyze v3 uboot missing "setnor16m" and other commands

plush, 2024-04-27:
I replaced the firmware on a Wyze v3 camera (model WYZEC3B, SOC Ingenic T31X, 16MB NOR) using the [official installation guide](https://openipc.org/cameras/vendors/ingenic/socs/t31x?mac=a2-a1-15-09-55-97&cip=192.168.1.10&sip=192.168.1.254&net=wifi&rom=nor16m&var=ultimate&sd=sd), but I cannot complete the process as the bootloader is missing `setnor16m` and many other values.

The bootloader environment [looks like this](https://github.com/OpenIPC/firmware/files/15140197/Broken.environment.txt). I have another camera, same model, that I flashed several months ago. That one works perfectly and the environment [looks very different](https://github.com/OpenIPC/firmware/files/15140198/Working.environment.txt), containing e.g. the `setnor16m` that I am missing. I always used the newest prebuilt firmware from the website and I don't have the image that I had flashed on the earlier camera to play around with anymore unfortunately.

I tried copying over the entire environment from my working camera, ran `saveenv` and then `run setnor16m`. I got no errors, but the bootloader is still not finding the partitions it needs, see the attached [boot log](https://github.com/OpenIPC/firmware/files/15140204/Boot.log.txt). I know I am running the OpenIPC bootloader, as the prompt is `OpenIPC #`. I also tried downloading and re-flashing just the bootloader, following the advanced section of the installation guide, but it made no difference.

basilfx, 2024-06-24:
I have the same problem (same output) with the Dafang camera (T20). I believe that [this](https://github.com/OpenIPC/u-boot-ingenic/pull/6/commits/3f37013823eacd844633621e90cd151eb630a048) PR removed these commands, in favor of autodetect which does not seem to work.

jpfree, 2024-06-28:
the same error in t31ZX，
OpenIPC # run setnor16m
## Error: "setnor16m" not defined
OpenIPC #

skilurius, 2024-06-28:
It is a bug with the current website, it will be replaced with a proper site soon.
Use the following link: https://openipc.org/cameras/vendors/ingenic/socs/t31n/download_full_image?flash_size=8&flash_type=nor&fw_release=ultimate

marshallhasq, 2024-06-28:
@jpfree could not install openipc even new links.
when install finally, camera resets every x minutes
move device to thingino firmware, working ok for wyze v3

skilurius, 2024-06-28:
It was a bug with the current website, it will be replaced with a proper site soon.

closed 2024-06-28 by skilurius

plush, 2024-06-30:
What is the workaround while we are waiting for the new website to be released? Are there updated install instructions that don't require "setnor16m" and friends anymore?

RayPS, 2024-07-09:
Same issue: "sertnor16m" not defined
```

```OpenIPC # run setnor16m
## Error: "sertnor16m" not defined
OpenIPC # <INTERRUPT>
OpenIPC # <INTERRUPT>
OpenIPC # <INTERRUPT>
OpenIPC # printenv 
baseaddr=0x21000000
baudrate=115200
bootargs=console=ttyS0,115200 panic=20 root=/dev/mtdblock3 init=/init mtdparts=NOR_FLASH:256k(boot),64k(env),2048k(kernel),${rootmtd}(rootfs),-(rootfs_data) LX_MEM=${memlx} mma_heap=mma_heap_name0,miu=0,sz=${memsz} cma=2M
bootcmd=setenv bootcmd ${cmdnor}; sf probe 0; saveenv; run bootcmd
bootdelay=0
cmdnor=sf probe 0; setenv setargs setenv bootargs ${bootargs}; run setargs; sf read ${baseaddr} ${kernaddr} ${kernsize}; bootm ${baseaddr}
ethact=sstar_emac
ethaddr=00:00:23:34:45:66
gatewayip=192.168.1.1
ipaddr=192.168.1.10
kernaddr=0x50000
kernsize=0x200000
memlx=0x3FE0000
memsz=0x2000000
netmask=255.255.255.0
rootaddr=0x250000
rootmtd=5120k
rootsize=0x500000
sdcard=setenv updatetool fatload mmc 0
serverip=192.168.1.1
soc=ssc337
stderr=serial
stdin=serial
stdout=serial
totalmem=64M
ubnor=sf probe 0; sf erase 0x0 ${kernaddr}; sf write ${baseaddr} 0x0 ${kernaddr}
uknor=${updatetool} ${baseaddr} uImage.${soc}; sf probe 0; sf erase ${kernaddr} ${kernsize}; sf write ${baseaddr} ${kernaddr} ${filesize}
updatetool=tftpboot
urnor=${updatetool} ${baseaddr} rootfs.squashfs.${soc}; sf probe 0; sf erase ${rootaddr} ${rootsize}; sf write ${baseaddr} ${rootaddr} ${filesize}
ver=U-Boot 2015.01 (Jun 02 2024 - 15:01:29)

Environment size: 1317/65532 bytes
OpenIPC #

plush, 2024-07-09:
The official installation guide has been updated to no longer use `setnor16m` and related commands. I tried flashing the latest firmware using the updated guide today - and it worked perfectly. My camera is up and running.