OpenIPC/firmware #1436: RTMP streaming problem

Egho9, 2024-05-08:
Hi everyone. 
RTMP stream dont work with my server. Through wireshark packet capture analysis, I found several problems with rtmp stream.After the rtmp push status is established, the device does not send setDataFrame to the server and in subsequent media data streams, only have aac data，no video data.The RTMP device does not have an AVC sequence header in the stream, so the encoding parameters of the video cannot be determined.
I check and maybe problem in AV_CODEC_FLAG_GLOBAL_HEADER .Find some info how slove it here https://stackoverflow.com/questions/77535076/problems-with-outputting-stream-format-as-rtmp-via-ffmpeg-c-api

widgetii, 2024-05-08:
What software is installed on your RTMP server?

Egho9, 2024-05-09:
we use this https://github.com/nareix/joy4