OpenIPC/firmware #1444: Unable to mount root fs with (driver?) on Ingenic T31X(Wyze V3)

CazYokoyama, 2024-05-14:
Are you able to give me where in kernel mtd driver is loaded? jz_sfc_probe() successfully loads mtd driver as in the dmesg. It should set disk->driverfs_dev->driver, shouldn't it? Does your dmesg have "JZ GMAC disabled"?

dmesg:
[    0.000000] Kernel command line: console=ttyS1,115200 panic=10 root=/dev/mtdblock3 init=/init mtdparts=jz_sfc:256k(boot),64k(env),2048k(kernel),5120k(rootfs),-(rootfs_data) mem=64M rmem=6c
....
[    0.367776] squashfs: version 4.0 (2009/01/31) Phillip Lougher
....
[    0.483702] JZ SFC Controller for SFC channel 0 driver register
[    0.489819] 5 cmdlinepart partitions found on MTD device jz_sfc
[    0.496160] Creating 5 MTD partitions on "jz_sfc":
[    0.501108] 0x000000000000-0x000000040000 : "boot"
[    0.507122] 0x000000040000-0x000000050000 : "env"
[    0.513113] 0x000000050000-0x000000250000 : "kernel"
[    0.519252] 0x000000250000-0x000000750000 : "rootfs"
[    0.525487] 0x000000750000-0x000001000000 : "rootfs_data"
[    0.532099] SPI NOR MTD LOAD OK
[    0.535688] JZ GMAC disabled
...
[    0.618898] List of all partitions:
[    0.622524] 1f00             256 mtdblock0  (driver?)
[    0.633610] 1f01              64 mtdblock1  (driver?)
[    0.644490] 1f02            2048 mtdblock2  (driver?)
[    0.655380] 1f03            5120 mtdblock3  (driver?)
[    0.666252] 1f04            8896 mtdblock4  (driver?)
[    0.679136] No filesystem could mount root, tried:  squashfs
[    0.684987] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)

uImage and rootfs:
OpenIPC # mw.b ${baseaddr} ff 1000000; fatload mmc 0:1 ${baseaddr} uImage.${soc} ; sf probe 0; sf erase 0x50000 0x300000; sf write ${baseaddr} 0x50000 0x1BE8A6
OpenIPC # mw.b ${baseaddr} ff 1000000; fatload mmc 0:1 ${baseaddr} rootfs.squashfs.${soc} ; sf probe 0; sf erase 0x350000 0xa00000; sf write ${baseaddr} 0x350000 0x4AF000

[root@WCV3:~]# /media/mmc/ipctool-mips32 
chip:
  vendor: Ingenic
  model: T31X
ethernet:
  mac: "7c:78:b2:7f:48:43"
rom:
- type: nor
  block: 32K
  partitions:
    - name: boot
      size: 0x40000
      sha1: dda0626f
    - name: kernel
      size: 0x1f0000
      sha1: 6887b828
    - name: rootfs
      size: 0x3d0000
      sha1: d368892a
    - name: app
      size: 0x3d0000
      path: /system,squashfs
      sha1: 04b813a8
    - name: kback
      size: 0x1f0000
      sha1: 6a9dfced
    - name: aback
      size: 0x3d0000
      sha1: e5b26a97
    - name: cfg
      size: 0x60000
      path: /configs,jffs2,rw
    - name: para
      size: 0x10000
      sha1: 472a55b0
  size: 16M
ram:
  total: 128M
  media: 32M
firmware:
  kernel: "3.10.14__isvp_swan_1.0__ (PREEMPT Wed Apr 14 17:44:51 CST 2021)"
  toolchain: gcc version 4.7.2 (Ingenic r2.3.3 2016.12) 
  libc: uClibc 0.9.33.2
  main-app: /system/bin/iCamera

I am not familiar with OpenIPC yet. I may have a trivial mistake. Thank you.

ckuethe, 2024-05-18:
I just installed openipc on three wyze cam v3's tonight using info in the file below. no problems.

https://github.com/themactep/device-wyze-v3/blob/master/README_RETIRED.md

```
Firmware
Version 2.4.05.18-lite
Build master+0bcb229, 2024-05-18
Majestic master+90c0b57, 2024-05-16
U-Boot U-Boot 2013.07-g90873ad (May 01 2024 - 19:56:15)
```

marshallhasq, 2024-05-18:
@ckuethe could not install openipc
when install finally, camera resets every x minutes
had to move to thingino firmware, working ok for wyze v3

CazYokoyama, 2024-05-18:
Disregard. I moved to thingino. It works fine. Will close.

closed 2024-05-18 by CazYokoyama