OpenIPC/firmware #1447: Inquiry about SDK for Anyka AK3918EV330

hldh214, 2024-05-19:
Hello OpenIPC Team,

I came across the documentation for the Anyka AK3918EV330 [here](https://openipc.org/cameras/vendors/anyka). I would like to inquire about where I can find the SDK related to this SoC.

I currently have an Anyka AK3918EV330 based camera and have obtained root access to it. I am interested in contributing to the development for this platform. Could you please guide me on how I can assist with the development efforts?

widgetii, 2024-05-19:
You may contact Anyka vendor to figure out what they want from you (NDA, money, etc) in exchange for acquiring their SDK

hldh214, 2024-05-20:
I have already emailed their sales team, but I don't have high hopes since their website only provides product information through business partnerships.

I thought the SDK was public according to the content of the linked webpage. Nevertheless, thank you.

closed 2026-11-09 by openipc-ai