OpenIPC/firmware #1456: How to add wifi driver RT2870 on Goke GK7205V200 board

henkiejan1, 2024-05-23:
Hi everyone,

Yesterday arrived my ipcam with a Goke GK7205V200 chip. I flashed without any troubles OpenIPC. But i don't get the wifi working.  I read that i maybe need to copy the wifi driver from the original firmware to OpenIPC but i don't have any idea how. Does someone have  steo by step instructions to get the correct driver and integrate in OpenIPC?

Thank you very much!

henkiejan1, 2024-05-25:
I have made a lot of progress! Can connect to wifi manually if i run the following commands:

`setnetwork -i wlan0 -s SSID -p PASSWORD -m dhcp
echo 9 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio9/direction
echo 0 > /sys/class/gpio/gpio9/value
modprobe mt7601u
ifup wlan0`

So i think gpio9 enables the wifi module. I found the configuration in the network settings so i added the commands what i found here: https://github.com/OpenIPC/wiki/blob/master/en/network-settings.md#wlan0--hi3516ev200gk7205v200gk7205v210--camhixin

And whas hoping wifi would automatic connect, but sadly it doesn't.  I see if i run lsusb after a reboot the wlan module is not detected. But with the manual running comands aboven it does. What goes wrong?????

`auto wlan0
iface wlan0 inet dhcp
    pre-up echo 9 > /sys/class/gpio/export
    pre-up echo out > /sys/class/gpio/gpio9/direction
    pre-up echo 0 > /sys/class/gpio/gpio9/value
    pre-up modprobe mt7601u
    pre-up wpa_passphrase "SSID" "password" >/tmp/wpa_supplicant.conf
    pre-up sed -i '2i \\tscan_ssid=1' /tmp/wpa_supplicant.conf
    pre-up sleep 3
    pre-up wpa_supplicant -B -D nl80211 -i wlan0 -c/tmp/wpa_supplicant.conf
    post-down killall -q wpa_supplicant
    post-down echo 1 > /sys/class/gpio/gpio9/value
    post-down echo 9 > /sys/class/gpio/unexport`

mspnr, 2025-06-27:
Wireless settings guide is for sure missing some parts for those, who never worked with this
https://github.com/OpenIPC/wiki/blob/master/en/wireless-settings.md

Here is my addition - I hope it can be useful for someone:
You need to find appropriate name based on your chip and WiFi driver in one of three files here:
https://github.com/OpenIPC/firmware/tree/master/general/overlay/etc/wireless

For me it was `rtl8733bu-gk7205v200-camhi`, because I have `gk7205v200` written on the chip and `rtl8733bu` on the WiFi module.

Then you can set driver and WiFi credentials using these commands:

```sh
fw_setenv wlandev rtl8733bu-gk7205v200-camhi
fw_setenv wlanssid MySSID
fw_setenv wlanpass MyPassword
```

Then after reboot camera switched to WiFi and is working all right. 
Be careful though - if you select incorrect driver - it will never correct to network even via cable.

widgetii, 2026-08-16:
Sorry for the very long silence here. Reading back, you had already solved the hard part yourself — and the remaining failure has a specific cause worth writing down.

Your manual sequence works, so the driver and the hardware are fine. What fails is that your `/etc/network/interfaces` `pre-up` hooks run **too late**: by the time `ifup wlan0` is processed, the USB device has to already be enumerated. GPIO 9 is the module power/enable line, so at boot the dongle is unpowered, `lsusb` shows nothing, and there is no `wlan0` for `ifup` to bring up. Toggling it by hand afterwards is what makes the device appear — which is exactly what you observed.

The supported mechanism for this is the `wlandev` environment variable, not `interfaces`. It runs `/etc/wireless/usb <profile>` early in boot, before networking. Your board is already covered:

```sh
fw_setenv wlandev mt7601u-gk7205v200-camhi
fw_setenv wlanssid YourSSID
fw_setenv wlanpass YourPassword
reboot
```

That profile is in `general/overlay/etc/wireless/usb` and does precisely your two steps in the right order:

```sh
if [ "$1" = "mt7601u-gk7205v200-camhi" ]; then
	set_gpio 9 0
	modprobe mt7601u
	exit 0
fi
```

Same GPIO 9, same driver. Remove your hand-written `pre-up` lines from `/etc/network/interfaces` first, or they will fight it.

On the RT2870 question in your title: you do not need it. Most dongles sold as "RT2870" on these cameras are actually MT7601U, and your own testing confirms `mt7601u` binds to yours. We do not enable `rt2x00`/`rt2800usb` on Goke targets, and there is no reason to add it for this board.

@mspnr — thank you, that comment is genuinely the missing manual. The same `fw_setenv wlandev` pattern with the right profile name from `general/overlay/etc/wireless/{usb,sdio,modem}`.

References:
- Wireless setup: https://github.com/OpenIPC/wiki/blob/master/en/wireless-settings.md
- Adding a driver we do not ship: https://github.com/OpenIPC/wiki/blob/master/en/adding-wifi-driver.md

If your board needs a GPIO combination that no existing profile covers, the right home for it is a device profile in https://github.com/OpenIPC/builder — PRs there are welcome and are how board-specific bring-up reaches other users.

Closing as answered.

closed 2026-08-16 by widgetii