OpenIPC/firmware #1460: GK7205v200 with rtl8188ftv

Oldroid123, 2024-06-03:
Hello. I installed the ultimate firmware. The problem is that my camera can only connect via the rtl8188ftv wifi module connected via USB. But I don't see it in the system at all. lsusb only detects hubs. I tried to compile the firmware with the rtl8188fu driver myself, but the situation did not change. When I try to measure the voltage at the module contacts, it is equal to 0. How can I get work wifi?
<details><summary>Booting log</summary>
<p>

U-Boot 2016.11-g852b5be (Jan 22 2024 - 21:07:52 +0300)gk7205v200

Relocation Offset is: 0372e000
Relocating to 43f2e000, new gd at 43eedef0, sp at 43eeded0
SPI Nor:  Check Flash Memory Controller v100 ... Found
SPI Nor ID Table Version 1.0
SPI Nor(cs 0) ID: 0xc2 0x20 0x19
Block:64KB Chip:32MB Name:"MX25L(256/257)XX"
SPI Nor total size: 32MB
NAND:  0 MiB
MMC:   sdhci: 0 (SD)
In:    serial
Out:   serial
Err:   serial
RAM size: 64MB
Net:   eth0
Warning: eth0 (eth0) using random MAC address - 96:1c:cb:5a:48:33

Press Ctrl-c to stop autoboot... 2
device 0 offset 0x50000, size 0x300000

SF: 3145728 bytes @ 0x50000 Read: OK
Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.9.37-gk7205v200
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1858024 Bytes = 1.8 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.37 (runner@fv-az1269-466) (buildroot-gcc-13.2.0) #1 Fri May 31 02:02:58 UTC 2024
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Goke GK7205V200 DEMO Board
cma zone is not set!
Memory policy: Data cache writeback
CPU: All CPU(s) started in SVC mode.
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: mem=32M console=ttyAMA0,115200 panic=20 root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=sfc:256k(boot),64k(env),3072k(kernel),10240k(rootfs),-(rootfs_data)
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 27056K/32768K available (3856K kernel code, 150K rwdata, 912K rodata, 192K init, 181K bss, 5712K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc2800000 - 0xff800000   ( 976 MB)
    lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03cc3a8   (3857 kB)
      .init : 0xc04b2000 - 0xc04e2000   ( 192 kB)
      .data : 0xc04e2000 - 0xc0507ae0   ( 151 kB)
       .bss : 0xc0509000 - 0xc0536468   ( 182 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
Gic dist init...
arm_arch_timer: Architected cp15 timer(s) running at 50.00MHz (phys).
clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
Switching to timer-based delay loop, resolution 20ns
clocksource: arm,sp804: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 637086815595 ns
Console: colour dummy device 80x30
[... 392 lines cut ...]
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
goke-i2c 12060000.i2c: wait idle timeout, RIS: 0x10, SR: 0xa0000
root@openipc-gk7205v200:/lib/modules/4.9.37#

</p>
</details>

cronyx, 2024-06-04:
Your board has USB power management, you need to find the GPIO responsible for supplying power. Perhaps a [table with known GPIOs](https://github.com/OpenIPC/wiki/blob/master/en/gpio-settings.md) for your soc model will help you. In any case, you can always find it using the bruteforce method.

Oldroid123, 2024-06-04:
> Your board has USB power management, you need to find the GPIO responsible for supplying power. Perhaps a [table with known GPIOs](https://github.com/OpenIPC/wiki/blob/master/en/gpio-settings.md) for your soc model will help you. In any case, you can always find it using the bruteforce method.

I have enabled all 23 GPIO ON, but it is 0 voltage on wifi module

Oldroid123, 2024-06-05:
Okey, i can power on and power off my wifi with 57 gpio. But if i try to build firmware with driver, i have error. Where i can get driver for my device?

closed 2024-06-14 by Oldroid123

danieltwagner, 2024-07-20:
I found this thread when googling and thought I'd also post that for my gk7205v200 with built-in wifi pulling GPIO 9 low (`echo "0" > /sys/class/gpio/gpio9/value`) worked, much as indicated in [the table with known GPIOs](https://github.com/OpenIPC/wiki/blob/master/en/gpio-settings.md) that @cronyx linked above. Thanks for the helpful pointer.

The resulting deviceid `0bda:f72b` finally pointed me at the fact that there is already an [existing configuration](https://github.com/OpenIPC/firmware/blob/6d2dd985806c47ab60b86e73dee605a5f59464e2/general/overlay/etc/wireless/usb#L52) for my particular board and that [creating a new build with the wifi driver](https://github.com/OpenIPC/wiki/blob/master/en/adding-wifi-driver.md) and then [enabling it](https://github.com/OpenIPC/wiki/blob/master/en/wireless-settings.md) using `fw_setenv` was pretty straight-forward once I knew where to look. Maybe this will help someone else arriving via search results.