OpenIPC/firmware #1474: Compile issue(Cannot execute cross-compiler '/home/.../arm-openipc-linux-musleabi-gcc.br_real')

asyncbridge, 2024-06-18:
Hello, 

According to firmware build guide, I tried to build firmware for Hisilicon hi3516cv300_ultimate.
I got below error and compilation was failed. The environment as follows.
Is some configuration wrong? How to find the root cause of this issue?

- Build env: Ubuntu 20.04.6 LTS

This is the error log.

---
test@pc:~/Desktop/project/openipc-firmware$ make
--- br-ext-chip-hisilicon/configs/hi3516cv300_ultimate_defconfig
make[1]: Entering directory '/home/test/Desktop/project/openipc-firmware/output/buildroot-2024.02.1'
  GEN     /home/test/Desktop/project/openipc-firmware/output/Makefile
#
# configuration written to /home/test/Desktop/project/openipc-firmware/output/.config
#
make[1]: Leaving directory '/home/test/Desktop/project/openipc-firmware/output/buildroot-2024.02.1'
make[1]: Entering directory '/home/test/Desktop/project/openipc-firmware/output/buildroot-2024.02.1'
/usr/bin/make -j1  O=/home/test/Desktop/project/openipc-firmware/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
  GEN     /home/test/Desktop/project/openipc-firmware/output/Makefile
>>> toolchain-external-custom  Configuring
mkdir -p /home/test/Desktop/project/openipc-firmware/output/per-package/toolchain-external-custom/host
rsync -a --hard-links  --link-dest=/home/test/Desktop/project/openipc-firmware/output/per-package/host-ccache/host/ /home/test/Desktop/project/openipc-firmware/output/per-package/host-ccache/host/ /home/test/Desktop/project/openipc-firmware/output/per-package/toolchain-external-custom/host
rsync -a --hard-links  --link-dest=/home/test/Desktop/project/openipc-firmware/output/per-package/host-skeleton/host/ /home/test/Desktop/project/openipc-firmware/output/per-package/host-skeleton/host/ /home/test/Desktop/project/openipc-firmware/output/per-package/toolchain-external-custom/host
rsync -a --hard-links  --link-dest=/home/test/Desktop/project/openipc-firmware/output/per-package/skeleton/host/ /home/test/Desktop/project/openipc-firmware/output/per-package/skeleton/host/ /home/test/Desktop/project/openipc-firmware/output/per-package/toolchain-external-custom/host
mkdir -p /home/test/Desktop/project/openipc-firmware/output/per-package/toolchain-external-custom/target
rsync -a --hard-links  --link-dest=/home/test/Desktop/project/openipc-firmware/output/per-package/host-ccache/target/ /home/test/Desktop/project/openipc-firmware/output/per-package/host-ccache/target/ /home/test/Desktop/project/openipc-firmware/output/per-package/toolchain-external-custom/target
rsync -a --hard-links  --link-dest=/home/test/Desktop/project/openipc-firmware/output/per-package/host-skeleton/target/ /home/test/Desktop/project/openipc-firmware/output/per-package/host-skeleton/target/ /home/test/Desktop/project/openipc-firmware/output/per-package/toolchain-external-custom/target
rsync -a --hard-links  --link-dest=/home/test/Desktop/project/openipc-firmware/output/per-package/skeleton/target/ /home/test/Desktop/project/openipc-firmware/output/per-package/skeleton/target/ /home/test/Desktop/project/openipc-firmware/output/per-package/toolchain-external-custom/target
Cannot execute cross-compiler '/home/test/Desktop/project/openipc-firmware/output/per-package/toolchain-external-custom/host/opt/ext-toolchain/bin/arm-openipc-linux-musleabi-gcc.br_real'
make[2]: *** [package/pkg-generic.mk:273: /home/test/Desktop/project/openipc-firmware/output/build/toolchain-external-custom/.stamp_configured] Error 1
make[1]: *** [Makefile:82: _all] Error 2
make[1]: Leaving directory '/home/test/Desktop/project/openipc-firmware/output/buildroot-2024.02.1'
make: *** [Makefile:25: build] Error 2
---

asyncbridge, 2024-06-25:
My test pc is Nvidia Jetson and arch is aarch64. I think it's related to this issue.

makar0nin, 2024-06-25:
Similar problem.
```shell
>>> toolchain-external-custom  Extracting
mkdir -p /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/host
mkdir -p /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/target
gzip -d -c /srv/develop/openipc-firmware/output/buildroot-2024.02.1/dl/toolchain-external-custom/cortex_a7_thumb2-gcc13-musl-4_9.tgz | tar --strip-components=1 -C /srv/develop/openipc-firmware/output/build/toolchain-external-custom --exclude='usr/lib/locale/*'   -xf -
rm -rf /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/host/opt/ext-toolchain
mkdir -p /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/host/opt/ext-toolchain
mv /srv/develop/openipc-firmware/output/build/toolchain-external-custom/* /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/host/opt/ext-toolchain/
>>> toolchain-external-custom  Patching
>>> toolchain-external-custom  Configuring
mkdir -p /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/host
rsync -a --hard-links  --link-dest=/srv/develop/openipc-firmware/output/per-package/host-ccache/host/ /srv/develop/openipc-firmware/output/per-package/host-ccache/host/ /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/host
rsync -a --hard-links  --link-dest=/srv/develop/openipc-firmware/output/per-package/host-skeleton/host/ /srv/develop/openipc-firmware/output/per-package/host-skeleton/host/ /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/host
rsync -a --hard-links  --link-dest=/srv/develop/openipc-firmware/output/per-package/skeleton/host/ /srv/develop/openipc-firmware/output/per-package/skeleton/host/ /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/host
mkdir -p /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/target
rsync -a --hard-links  --link-dest=/srv/develop/openipc-firmware/output/per-package/host-ccache/target/ /srv/develop/openipc-firmware/output/per-package/host-ccache/target/ /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/target
rsync -a --hard-links  --link-dest=/srv/develop/openipc-firmware/output/per-package/host-skeleton/target/ /srv/develop/openipc-firmware/output/per-package/host-skeleton/target/ /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/target
rsync -a --hard-links  --link-dest=/srv/develop/openipc-firmware/output/per-package/skeleton/target/ /srv/develop/openipc-firmware/output/per-package/skeleton/target/ /srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/target
Cannot execute cross-compiler '/srv/develop/openipc-firmware/output/per-package/toolchain-external-custom/host/opt/ext-toolchain/bin/arm-openipc-linux-musleabi-gcc.br_real'
make[2]: *** [package/pkg-generic.mk:273: /srv/develop/openipc-firmware/output/build/toolchain-external-custom/.stamp_configured] Error 1
make[1]: *** [Makefile:82: _all] Error 2
make[1]: Leaving directory '/srv/develop/openipc-firmware/output/buildroot-2024.02.1'
make: *** [Makefile:25: build] Error 2
``` 
`Linux  5.15.0-112-generic #122~20.04.1-Ubuntu SMP  x86_64 x86_64 x86_64 GNU/Linux`

widgetii, 2024-06-25:
You should use Ubuntu 22.04 on top of x86_64 platform

closed 2024-06-25 by widgetii

asyncbridge, 2024-06-25:
Thanks for your answer. It worked well in Ubuntu 22.04 x86_64 platform.