OpenIPC/firmware #1494: GK7205V300 asking for u-boot password

nkajic, 2024-07-19:
**Board info:**
GK7205V300 + Smartsens SC5239P
https://vi.aliexpress.com/item/1005003340814126.html

***Tried but without success***
HI2105CHIP
ubootpwd

***Normal boot***
System startup

Uncompress Ok!

U-Boot 2016.11-g4bd9c94-dirty (Apr 22 2022 - 21:25:43 +0800)gk7205v300

Relocation Offset is: 0771f000
Relocating to 47f1f000, new gd at 47edeef0, sp at 47edeed0
SPI Nor:  Check Flash Memory Controller v100 ... Found
SPI Nor ID Table Version 1.0
@hifmc_spi_nor_probe(), SPI Nor(cs 0) ID: 0xa1 0x40 0x18 <Read>
@hifmc_spi_nor_probe(), SPI Nor(cs 0) ID: 0xa1 0x40 0x18 <Found>
SPI Nor(cs 0) ID: 0xa1 0x40 0x18
eFlashType: 30.
@XmSpiNor_initSr3() need No SR3!
@XmSpiNor_printWps(), WPS Enabled!
Flash Name: XM_FM25Q128A{0xA14018), 0x1000000.
@hifmc_spi_nor_probe(), XmSpiNor_ProtMgr_probe(): OK.
@XmSpiNor_enableQuadMode(), Quad was Disabled, SRx: [2, 0x8].
Block:64KB Chip:16MB Name:"XM_FM25Q128A"
CONFIG_CLOSE_SPI_8PIN_4IO = y.
read->iftype[0: STD, 1: DUAL, 2: DIO, 3: QUAD, 4: QIO]: 1.
lk[7 => 0x1000000], Total Lock Blks: 286 <WPS=1>
SRx val: {[1, 0x20], [1, 0x8], [1, 0x0], [0, 0x0]}, SrVal: 0x700000000000820.
SPI Nor total size: 16MB
In:    serial
Out:   serial
Err:   serial
Net:   eth0
autoboot:  0 
@do_spi_flash_probe() flash->erase_size: 65536, flash->sector_size: 0
device 0 offset 0x40000, size 0x540000

SF: 5505024 bytes @ 0x40000 Read: OK
srcAddr: 0x43000000, dstAddr: 0x42000000, filename: boot/uImage.
created_inode 0x47edfc18
find_squashfs_file: name bin, start_block 0, offset 1824, type 1
find_squashfs_file: name boot, start_block 0, offset 1996, type 1
read inode: name boot, sb 0, of 1996, type 1
find_squashfs_file: name uImage, start_block 0, offset 1856, type 2
read inode: name uImage, sb 0, of 1856, type 2
write_file: regular file, blocks 27
len 1723581
### get_squashfs_file OK: loade 1723581 bytes to 0x42000000
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.9.37
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1723517 Bytes = 1.6 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image ... OK
using: ATAGS
at setup_xminfo_tag() g_nXmBootSysIndex: 0, g_nXmRomfsIndex: 0.

Starting kernel ...

***CTL + C break boot asking password***
System startup

Uncompress Ok!

U-Boot 2016.11-g4bd9c94-dirty (Apr 22 2022 - 21:25:43 +0800)gk7205v300

Relocation Offset is: 0771f000
Relocating to 47f1f000, new gd at 47edeef0, sp at 47edeed0
SPI Nor:  Check Flash Memory Controller v100 ... Found
SPI Nor ID Table Version 1.0
@hifmc_spi_nor_probe(), SPI Nor(cs 0) ID: 0xa1 0x40 0x18 <Read>
@hifmc_spi_nor_probe(), SPI Nor(cs 0) ID: 0xa1 0x40 0x18 <Found>
SPI Nor(cs 0) ID: 0xa1 0x40 0x18
eFlashType: 30.
@XmSpiNor_initSr3() need No SR3!
@XmSpiNor_printWps(), WPS Enabled!
Flash Name: XM_FM25Q128A{0xA14018), 0x1000000.
@hifmc_spi_nor_probe(), XmSpiNor_ProtMgr_probe(): OK.
@XmSpiNor_enableQuadMode(), Quad was Disabled, SRx: [2, 0x8].
Block:64KB Chip:16MB Name:"XM_FM25Q128A"
CONFIG_CLOSE_SPI_8PIN_4IO = y.
read->iftype[0: STD, 1: DUAL, 2: DIO, 3: QUAD, 4: QIO]: 1.
lk[7 => 0x1000000], Total Lock Blks: 286 <WPS=1>
SRx val: {[1, 0x20], [1, 0x8], [1, 0x0], [0, 0x0]}, SrVal: 0x700000000000820.
SPI Nor total size: 16MB
In:    serial
Out:   serial
Err:   serial
Net:   eth0
autoboot:  0 
Password: 
Password: 
Password: 
Password: 

Thanks

Grundefuht, 2024-07-26:
I met same problem. But found that no u-boot is needed

1. upgraded camera via web interface by precompiled "lite" bin.
2. setup image sensor properly. I selected "4l_sc223a_i2c_1080p"

i missed creating first backup, that is not good. but i was lucky

nkajic, 2024-08-30:
Hi @Grundefuht that is interesting. Thanks for feedback.
I will keep issue open until someone finds pwd.

ankurvdev, 2024-10-29:
Try this `#Ux6@9V&4_Rz`

flyrouter, 2025-01-06:
> Try this `#Ux6@9V&4_Rz`

Good day colleague
Thanks for the information.
I would like to know what methods were used to obtain this password.
Was it a leak somewhere or a selection using certain methods?
I have a strong interest in understanding the process.
Thanks.

chris-yooo, 2025-01-30:
works here too for Novatek NT98566

pw: #Ux6@9V&4_Rz

[image]

[image]

ty

jducifuentes, 2025-03-02:
Working here also:
Sony Imx335 + Goke Gk7205V300

```
System startup

Uncompress Ok!

U-Boot 2016.11-g4bd9c94-dirty (Apr 22 2022 - 21:25:43 +0800)gk7205v300

Relocation Offset is: 0771f000
Relocating to 47f1f000, new gd at 47edeef0, sp at 47edeed0
SPI Nor:  Check Flash Memory Controller v100 ... Found
SPI Nor ID Table Version 1.0
@hifmc_spi_nor_probe(), SPI Nor(cs 0) ID: 0xef 0x40 0x18 <Read>
@hifmc_spi_nor_probe(), SPI Nor(cs 0) ID: 0xef 0x40 0x18 <Found>
SPI Nor(cs 0) ID: 0xef 0x40 0x18
eFlashType: 3.
@XmSpiNor_initSr3() need No SR3!
@XmSpiNor_printWps(), WPS Enabled!
Flash Name: XM_W25Q128FV, W25Q128JV{0xEF4018), 0x1000000.
@hifmc_spi_nor_probe(), XmSpiNor_ProtMgr_probe(): OK.
@XmSpiNor_enableQuadMode(), Disable Quad Failed, SRx: [2, 0x2].
Block:64KB Chip:16MB Name:"XM_W25Q128FV,W25Q128JV"
CONFIG_CLOSE_SPI_8PIN_4IO = y.
read->iftype[0: STD, 1: DUAL, 2: DIO, 3: QUAD, 4: QIO]: 1.
lk[7 => 0x1000000], Total Lock Blks: 286 <WPS=1>
SRx val: {[1, 0x20], [1, 0x2], [1, 0x64], [0, 0x0]}, SrVal: 0x700000000640220.
SPI Nor total size: 16MB
In:    serial
Out:   serial
Err:   serial
Net:   eth0
autoboot:  0
Password:
Password:
Password:
Password: ************
goke #
```

widgetii, 2026-08-16:
Closing this one out — it's been answered, and the answer deserves to be stated clearly at the end rather than buried mid-thread.

### The password

```
#Ux6@9V&4_Rz
```

Credit to @ankurvdev. Independently confirmed in this thread on:

- Goke GK7205V300 + IMX335 (@jducifuentes)
- Goke GK7205V500, TPTek 5MP (@TaylorTWBrown)
- Novatek NT98566 (@chris-yooo)
- LTIP-14DES-P (@ouinouin)

That spread — three different SoC vendors — says it's an OEM-level default baked into a shared bootloader codebase, not something per-device.

@flyrouter, on provenance: I don't know how it was originally obtained and I'm not going to speculate. It circulated in the community and it reproduces, which is as much as I can honestly say.

### If it doesn't work for you

Don't guess further — go underneath the bootloader instead.

Every Goke/Hi35xx SoC has a mask-ROM bootrom that listens on UART0 for a serial download handshake at power-on, *before* U-Boot runs. It's in silicon: a vendor can password-lock their bootloader, but they cannot disable the bootrom. Upload our U-Boot into RAM over the same UART and you get an unlocked console on a camera whose flash still holds the locked vendor bootloader. Nothing is written, so there's nothing to brick — and it works regardless of passwords, `autoboot: 0`, or anything else the vendor did.

**https://openipc.github.io/defib/** does it in the browser — Chrome, Edge or Opera (WebSerial; Firefox and Safari won't work). No install:

1. **Chip Model** → `gk7205v300`
2. **U-Boot image** → *Download from OpenIPC* — SHA-256 verified against our releases
3. **Serial Port** → your USB-UART adapter
4. Tick **"Send Ctrl-C after upload (enter U-Boot console)"**
5. **Start Recovery** — when it prints `Waiting for bootrom... power-cycle the device now!`, cut power and reapply

Then hit **Dump Flash** to back up the stock firmware before writing anything.

### One thing to watch on your board

Your log shows the vendor bootloader locked the whole chip:

```
lk[7 => 0x1000000], Total Lock Blks: 286 <WPS=1>
```

With `WPS=1` the flash uses per-block lock bits, and all 286 blocks are locked. If an erase later looks like it succeeded but nothing changed, that's the cause. The `defib agent` path in the CLI issues the global unlock and clears `SR3.WPS` before erasing.

@Grundefuht's route is also worth remembering for anyone reading later: if the vendor web UI still accepts a firmware upload, you can often flash the `lite` build straight through it and never touch UART at all.

I'll fold this password and the bootrom method into the wiki's [bootloader page](https://github.com/OpenIPC/wiki/blob/master/en/help-uboot.md), which currently mentions neither. Thanks all for the confirmations — that cross-vendor evidence is what made this conclusive.

widgetii, 2026-08-16:
Resolved: bootloader password `#Ux6@9V&4_Rz`, confirmed on Goke GK7205V300/V500, Novatek NT98566 and LTIP-14DES-P. For boards where it doesn't apply, the bootrom route via https://openipc.github.io/defib/ bypasses the locked bootloader entirely.

closed 2026-08-16 by widgetii