OpenIPC/firmware #1496: Novatek 98562 Openipc Support

vk2indi, 2024-07-22:
As I have downloaded the Openipc files for NT98562 and need to know how can i flash that files to my Hardware.

widgetii, 2024-07-22:
Novatek support is broken currently

vk2indi, 2024-07-23:
As i have the hardware and the SDK any way to get support or anything from my side to support for OpenIPC support on  Novatek

widgetii, 2024-07-23:
You can fix current platform issues if you have developer skills

MaizeShark, 2025-08-02:
> As i have the hardware and the SDK any way to get support or anything from my side to support for OpenIPC support on Novatek

Hi @vk2indi,
I'm in the exact same boat as you! I also have hardware based on the NT98566 and have been digging through forums for days trying to get all the necessary developer information. It's been a real struggle.

Seeing that you have the SDK is amazing news. I've been trying to get my hands on it to help contribute and potentially fix the broken builds for our hardware, but as you know, it's incredibly hard to obtain.

**Could you please, please, please consider sharing the SDK?** It would be an absolutely massive help, not just for me, but potentially for the entire OpenIPC community trying to get these powerful chips fully supported. Even just having access to the driver source code and the reference build environment would unlock so much potential.

I'd be incredibly grateful for any help you can provide. Thank you so much in advance!

widgetii, 2026-10-22:
Update on the current state. The "broken" comment above is from July 2024 and is no longer accurate.

**What works today.** The nightly builds `openipc.nt98562-nor-lite.tgz` and `openipc.nt98566-nor-lite.tgz` from the same tree. The two kernel configs differ only in the SoC name. On an XM-vendor NT98566 board (128 MB, 8 MB NOR, SC501AI) the 2026-10-21 nightly boots, detects the sensor and streams H.265 at 2880×1620. We don't have an NT98562 board in the lab, so the NT98562 image is built but untested on hardware. A report from you would be very welcome.

**Scope:**
- XM-vendor boards whose U-Boot prompt is `nvt@na51089:`. The vendor U-Boot stays on flash; only the kernel and rootfs are replaced.
- Sensors wired up in the image: **SC401AI, SC500AI and SC501AI**. For any other sensor the camera boots and networks, but there is no video yet. In that case please post the stock firmware's boot log and `ipcinfo --long-sensor` from OpenIPC.

**Flashing.** You need a UART console and a TFTP server.

0. Back up the whole flash first, with a programmer or `cat /dev/mtdblockN` from the stock firmware's shell. U-Boot can't write a backup to the network.
1. Download https://github.com/OpenIPC/firmware/releases/download/latest/openipc.nt98562-nor-lite.tgz and put `uImage.nt98562` and `rootfs.squashfs.nt98562` from it in your TFTP root.
2. Interrupt U-Boot (press Ctrl-C repeatedly while powering on) and run:

```
setenv ipaddr 192.168.1.10
setenv serverip 192.168.1.254
setenv soc nt98562
setenv osmem 32M
setenv bootargs 'earlyprintk console=ttyS0,115200 mem=${osmem} panic=20 nprofile_irq_duration=on root=/dev/mtdblock3 rootfstype=squashfs init=/init mtdparts=spi_nor.0:64k(loader),256k(boot),2048k(kernel),5120k(rootfs),-(rootfs_data)'
setenv bootcmd 'setenv setargs setenv bootargs ${bootargs};run setargs;sf probe 0;sf read 0x03100000 0x50000 0x200000;nvt_boot'
setenv uk 'mw.b 0x01000000 ff 1000000;tftp 0x01000000 uImage.${soc} && sf probe 0 && sf erase 0x50000 0x200000 && sf write 0x01000000 0x50000 ${filesize}'
setenv ur 'mw.b 0x01000000 ff 1000000;tftp 0x01000000 rootfs.squashfs.${soc} && sf probe 0 && sf erase 0x250000 0x500000 && sf write 0x01000000 0x250000 ${filesize}'
saveenv
run uk
run ur
reset
```

   Use your own network's addresses for `ipaddr` and `serverip`. Both `run` steps must end with `Written: OK` before you `reset`; if a download fails, the step stops before erasing anything.
3. After the first boot, run `firstboot` once to format the settings partition. The camera then starts *unclaimed*: open its web page to set the root password and accept the Majestic licence, or log in over SSH or serial, which asks for the same.

Later updates are just `run uk; run ur; reset`, or `sysupgrade` from a running camera.

**Memory.** Keep the `${osmem}` placeholder in `bootargs` exactly as above. U-Boot lays memory out from `osmem` while the kernel uses `mem=`, and if the two disagree the camera hangs at boot or crashes when the video drivers load. To give Linux more RAM, change only `osmem` (`fw_setenv osmem 54M`). Details: https://github.com/OpenIPC/wiki/blob/master/en/install-novatek.md.

None of this needs the vendor SDK. Adding another sensor needs its driver wired into `load_novatek` plus a majestic sensor profile; if you have such a board and can test, open an issue with the details above.