OpenIPC/firmware #1501: Cann't see video on Preview. Using Hi3516ev200 with sc4236.

shenzihang, 2024-07-25:
I used Hi3516ev200 with sc4236, but something may have wrong...

Here are something about the information, I cann't see any vidio on Preview until now T_T.

--- /rom/etc/majestic.yaml
+++ /etc/majestic.yaml
@@ -26,8 +26,9 @@
   fps: 15
 jpeg:
   enabled: true
-  qfactor: 50
-  fps: 5
+  qfactor: 12
+  fps: 2
+  rtsp: true
 osd:
   enabled: false
   font: "/usr/share/fonts/truetype/UbuntuMono-Regular.ttf"

# cat /etc/majestic.yaml
system:
  webPort: 80
  httpsPort: 443
  logLevel: debug
isp:
  antiFlicker: disabled
image:
  mirror: false
  flip: false
  rotate: 0
  contrast: 50
  hue: 50
  saturation: 50
  luminance: 50
video0:
  enabled: true
  codec: h264
  fps: 20
  bitrate: 4096
  rcMode: vbr
  gopSize: 1.0
video1:
  enabled: false
  codec: h264
  size: 704x576
  fps: 15
jpeg:
  enabled: true
  qfactor: 12
  fps: 2
  rtsp: true
osd:
  enabled: false
  font: "/usr/share/fonts/truetype/UbuntuMono-Regular.ttf"
  template: "%d.%m.%Y %H:%M:%S"
  posX: 16
  posY: 16
audio:
  enabled: false
  volume: 30
  srate: 8000
  codec: opus
  outputEnabled: false
  outputVolume: 30
rtsp:
  enabled: true
  port: 554
nightMode:
  colorToGray: true
  irCutSingleInvert: false
  lightMonitor: false
  lightSensorInvert: false
motionDetect:
  enabled: false
  visualize: false
  debug: false
records:
  enabled: false
  path: "/mnt/mmcblk0p1/%F/%H/%M.mp4"
  maxUsage: 95
outgoing:
  enabled: false
watchdog:
  enabled: true
  timeout: 300
hls:
  enabled: false

majestic[820]: Majestic Lite for HiSilicon, version master+4d20dd9, built on 2024-07-20
majestic[820]: Detected Watchdog
majestic[820]: Watchdog timeout set to 300 seconds
majestic[820]: Set server path: /var/www
majestic[820]: HTTP server started on :::80
majestic[820]: Loading main configuration: /etc/majestic.yaml
majestic[820]: SENSOR=sc4236
majestic[820]: matched sensor config: sc4236_i2c_3M.ini
majestic[820]: Using /etc/sensors/sc4236_i2c_3M.ini sensor configuration file
majestic[820]: App was built with MPP version: Hi3516EV200_MPP_V1.0.1.2.B030 Release
majestic[820]: Current MPP version: HI_VERSION=Hi3516EV200_MPP_V1.0.1.2 B030 Release
majestic[820]: sensor stSnsSc4236Obj
majestic[820]:   input_mode MIPI, WDR NONE
majestic[820]:   dev [2304x1296]@200x20 20fps, BGGR
majestic[820]: trying to load /usr/lib/sensors/libsns_sc4236.so
majestic[820]: Free MMZ mem before allocation: 32188KB
majestic[820]: VB configuration:
majestic[820]:   [0]: 4478976 x 3
majestic[820]:   [1]: 3732480 x 1
majestic[820]: Free MMZ mem after allocation: 15408KB
majestic[820]: Sensor driver loaded
majestic[820]: SDK is in 'VI_ONLINE_VPSS_ONLINE' mode
majestic[820]: linear mode
majestic[820]: ==============================================================
majestic[820]: ===Smart sc4236 sensor 1296P30fps(MIPI) init success!=====
majestic[820]: ==============================================================
majestic[820]: h264 2304x1296@20fps 4096kbit 20gop
majestic[820]:   venc_chn: 0, vpss_grp: 0, vpss_chn: 0
majestic[820]: Loading IQ profile /etc/sensors/iq/default.ini
majestic[820]:   minQp: 28, maxQp: 42
majestic[820]:   venc took 11412KB of MMZ [2187]
majestic[820]: MJPEG 2304x1296@2fps 20q 
majestic[820]:   venc_chn: 1, vpss_grp: 0, vpss_chn: 1
majestic[820]: Cannot create VENC chn 1
majestic[820]: ERR_VENC_NOMEM: The memory fails to be allocated due to some causes such as insufficient system memory
majestic[820]: HiSilicon SDK started
majestic[820]: Free MMZ mem finally: 528KB
majestic[820]: RTSP server started on port 554
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0
majestic[820]: Timeout from venc channel 0

shenzihang, 2024-07-25:
Here are more information.
root@openipc-hi3516ev200:~# cat /dev/logmpp
<3>[(null)] Mmz malloc failed!
<3>[ jpege] [Func]:jpege_create_chn [Line]:320 [Info]:jpege_create_chn: BufSize  2991736 too large, alloc failed!
<3>[  venc] [Func]:VencCreateChnByType [Line]:4381 [Info]:jpeg create chn error
<3>[  venc] [Func]:VencCreateChn [Line]:4657 [Info]:chn:1 Creat chnl failed!!
<3>[  vpss] [Func]:VPSS_ReleaseOnlineVB [Line]:2507 [Info]:Chn:1 release VB timeout!

widgetii, 2026-08-15:
Do you have the same problem with the recent firmware?

5Breeze, 2026-08-25:
> 你最近固件也有同样的问题吗？

Sorry, it has been a long time. My device has been sold :(

5Breeze, 2026-08-25:
> Do you have the same problem with the recent firmware?

The question may be the vram of hi3516ev200, it can not support the large size of video

closed 2026-11-16 by openipc-ai