OpenIPC/firmware #1510: Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)

wiredolphin, 2024-08-02:
Hi everyone.

First of all thank you very much to all the contributors for the great job done so far within the **OpenIPC** project.

U-Boot version:
`U-Boot 2013.07-g90873ad (Jul 26 2024 - 19:14:52)`

I'm trying to add a wifi driver for a Realtek wifi module which ships with my camera board,
equipped with an **Ingenic T31X** SoC. 

When the rebuild procedure ends, I'm trying to flash the new updated kernel using
 the instructions available here:
[https://github.com/OpenIPC/wiki/blob/master/en/adding-wifi-driver.md](https://github.com/OpenIPC/wiki/blob/master/en/adding-wifi-driver.md)

That means, for a 16M flash chip (**GD25Q128**), I'll use:

```
mw.b ${baseaddr} 0xff 0x300000
tftp ${baseaddr} uImage.${soc}
sf probe 0; sf erase 0x50000 0x300000; sf write ${baseaddr} 0x50000 ${filesize}

mw.b ${baseaddr} 0xff 0x500000
tftp ${baseaddr} rootfs.squashfs.${soc}
sf probe 0; sf erase 0x350000 0xa00000; sf write ${baseaddr} 0x350000 ${filesize}
```

All the command above ends without any error but when I reset, I just get a
rebooting infinite loop with a 10 seconds timeout.

The below lines seems to be relevant to the cycling reboot behaviour:

```
[ 2.611062] List of all partitions:
[ 2.614606] 1f00 256 mtdblock0 (driver?)
[ 2.619672] 1f01 64 mtdblock1 (driver?)
[ 2.624806] 1f02 2048 mtdblock2 (driver?)
[ 2.629866] 1f03 5120 mtdblock3 (driver?)
[ 2.635006] 1f04 8896 mtdblock4 (driver?)
[ 2.640056] No filesystem could mount root, tried: squashfs
[ 2.645766] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(31,3)
```

I alse found that the "Upgrading from SD card" method ([https://github.com/OpenIPC/wiki/blob/master/en/sysupgrade.md#upgrading-from-sd-card](https://github.com/OpenIPC/wiki/blob/master/en/sysupgrade.md#upgrading-from-sd-card)) just works.

Thus if I issue:
```
soc=$(fw_printenv -n soc)
sysupgrade --kernel=/mnt/mmcblk0p1/uImage.${soc} --rootfs=/mnt/mmcblk0p1/rootfs.squashfs.${soc} --force_ver -z
```
I'm able to update my kernel. 

I'm just courious to know If anyone else has struggled with this issue.

I just feel updating from u-boot prompt more comfortable than
using the SD card and also wish to understand what is
happening under the hood when the above issue comes in.
Thanks in advance

skilurius, 2024-08-02:
All kernels are now 2MB in size, the flashing command can be changed to:
```
tftp ${baseaddr} rootfs.squashfs.${soc}
sf probe 0; sf erase 0x250000 0xA00000; sf write ${baseaddr} 0x250000 ${filesize}
```

closed 2024-08-11 by skilurius

wiredolphin, 2024-08-15:
Thank You for your explanation