OpenIPC/firmware #1521: Majestic IRCUT: Push-pull mode required

deece, 2024-08-15:
My IPCAM requires that the IRCUT lines be the inverse of each other. If not, I get weird horrible noises coming from the mechanism. The current behavior is that they are only the inverse of each other for a small amount of time, before they are both set to 0:

```
majestic[817]: Change mode to DAY
majestic[817]: set_gpio(54, 0)
majestic[817]: set_gpio(53, 1)
majestic[817]: set_gpio(54, 0)
majestic[817]: set_gpio(53, 0)

majestic[817]: Change mode to NIGHT
majestic[817]: set_gpio(54, 1)
majestic[817]: set_gpio(53, 0)
majestic[817]: set_gpio(54, 0)
majestic[817]: set_gpio(53, 0)

```

I can stop the noise by manually setting the appropriate GPIO:
```
echo 1 >/sys/class/gpio/gpio54/value
```

Could we please have a way to suppress the clearing of the IRCUT GPIOs?

deece, 2024-08-16:
In the meantime, here is a patch that users can apply to the binary to disable clearing the IRCUT lines.

It NOPs out the call to schedule that clears the IRCUT lines after a delay.

To create it, I opened the binary in Ghidra, and searched for GPIO. Once I found the variables that I believed held the GPIO numbers (those were used in a printf to /sys/class/gpio), I renamed them to IRCUT1 & IRCUT2, then searched for references to them. I found a call that wrote an argument to them, which I renamed write_gpio, and then looked for references to that. From there, I found a function that set both IRCUTs to 0. I renamed that to clear_ircut, and looked for references to it. There was only one, so I replaced that with a NOP and confirmed it works.

```
majestic[817]: Change mode to NIGHT
majestic[817]: set_gpio(54, 1)
majestic[817]: set_gpio(53, 0)
majestic[817]: Change mode to DAY
majestic[817]: set_gpio(54, 0)
majestic[817]: set_gpio(53, 1)
majestic[817]: Change mode to NIGHT
majestic[817]: set_gpio(54, 1)
majestic[817]: set_gpio(53, 0)
majestic[817]: Change mode to DAY
majestic[817]: set_gpio(54, 0)
majestic[817]: set_gpio(53, 1)
```

[majestic-master+11648f1-dont_clear_ircut.bspatch.gz](https://github.com/user-attachments/files/16631628/majestic-master%2B11648f1-dont_clear_ircut.bspatch.gz)

deece, 2024-08-16:
Monitoring the power usage of the camera shows that with one side high all the time, there is a significant increase in power consumption, and the lens gets pretty warm, so this request is incorrect and should be ignored.

I'll have to figure out why there is audible sound from the camera when both IRCUT lines are at 0.

closed 2024-08-16 by deece

kavcr, 2025-06-18:
Looks like we need to have a more flexible IRCUT output settings.
For example, a pretty common IR Filter Driver AP1511A uses two control lines: one for direction (FBC = forward/backward control) and one for enabling its output (ENB, active low).
With current Majestic IRCUT behavior it's impossible to disable this driver after moving IR filter, as Majestic either leaves both GPIOs in low state, ether switches single GPIO only.