OpenIPC/firmware #1532: GK7202v300 Install issue

kishorpise, 2024-08-21:
Hello All,
I have this board GK7202v300.

https://www.aliexpress.com/item/3256806069871187.html?spm=a2g0o.order_list.order_list_main.16.297d1802jtxpzG

Trying to install the OPENIPC, looks like its supported. So downloaded firmware files as shown below. 
use burn command for the same.

1) It looks like bootloader is locked. Some times it worked with Cnt+C asking password. Now its not asking password. There is no way to stop. Below is log.

```

System startup

Uncompress Ok!

U-Boot 2016.11-gc642d21-dirty (Feb 21 2024 - 14:46:11 +0800)gk7202v300

Relocation Offset is: 036dc000
Relocating to 43edc000, new gd at 43e9bef0, sp at 43e9bed0
SPI Nor:  Check Flash Memory Controller v100 ... Found
SPI Nor ID Table Version 1.0
@hifmc_spi_nor_probe(), SPI Nor(cs 0) ID: 0xa1 0x40 0x17 <Read>
@hifmc_spi_nor_probe(), SPI Nor(cs 0) ID: 0xa1 0x40 0x17 <Found>
SPI Nor(cs 0) ID: 0xa1 0x40 0x17
eFlashType: 29.
Flash Name: XM_FM25Q64{0xA14017), 0x800000.
@hifmc_spi_nor_probe(), XmSpiNor_ProtMgr_probe(): OK.
@XmSpiNor_enableQuadMode(), Quad was Disabled, SRx: [2, 0x0].
Block:64KB Chip:8MB Name:"XM_FM25Q64"
CONFIG_CLOSE_SPI_8PIN_4IO = y.
read->iftype[0: STD, 1: DUAL, 2: DIO, 3: QUAD, 4: QIO]: 1.
lk=>6, 0x400000.
SRx val: {[1, 0x38], [1, 0x0], [0, 0x0], [0, 0x0]}, SrVal: 0x300000000000038.
SPI Nor total size: 8MB
MMC:   Card did not respond to voltage select!
No SD device found !
sdhci: 0
In:    serial
Out:   serial
Err:   serial
Net:   eth0
@do_auto_update() device name mmc!
Card did not respond to voltage select!
Unknow device type, name:mmc!
autoboot:  0
@do_spi_flash_probe() flash->erase_size: 65536, flash->sector_size: 0
device 0 offset 0x40000, size 0x2e0000

SF: 3014656 bytes @ 0x40000 Read: OK
srcAddr: 0x43000000, dstAddr: 0x42000000, filename: boot/uImage.
created_inode 0x43e9de40
find_squashfs_file: name bin, start_block 0, offset 1796, type 1
find_squashfs_file: name boot, start_block 0, offset 1876, type 1
read inode: name boot, sb 0, of 1876, type 1
find_squashfs_file: name uImage, start_block 0, offset 1828, type 2
read inode: name uImage, sb 0, of 1828, type 2
write_file: regular file, blocks 4
len 1717845
### get_squashfs_file OK: loade 1717845 bytes to 0x42000000
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.9.37
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1717781 Bytes = 1.6 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image ... OK
using: ATAGS
at setup_xminfo_tag() g_nXmBootSysIndex: 0, g_nXmRomfsIndex: 0.

Starting kernel ...

```

[image]

2) Since There was issue of password. thought of direcltly flashing without back up. Below is output 

[image]

Can some one help me to resolve this issue ? This is windows, Burn, Python

suchair, 2024-09-17:
Hello.
I have the same problems with goke gk7205v300. Manufacturer Usafeqlo. Are there any solutions?

xakersolovei, 2025-05-13:
Подскажите можите скинуть фото куда кидали UART, у меня такая же платка но выод с консоли другой и cmd тоже заблочено

widgetii, 2026-08-16:
Sorry this sat unanswered so long. Both problems in your report have fixes, and the tooling has changed a lot since you filed.

### Your board

`gk7202v300`, 8 MB NOR (`XM_FM25Q64`, JEDEC `0xa1 0x40 0x17`). Supported.

### 1. You can't stop the bootloader — so don't try

`autoboot: 0` means there is literally no window to interrupt. Ctrl-C will never work, and no password guess will help you on a build that doesn't even prompt.

Go underneath it instead. Every Goke/Hi35xx SoC has a mask-ROM bootrom that listens on UART0 for a serial download handshake at power-on, *before* the bootloader gets control. It's in silicon — the vendor can lock their U-Boot however they like, but they cannot disable the bootrom. You upload our U-Boot into RAM and get an unlocked console on a camera whose flash still holds the locked vendor bootloader. Nothing gets written, so there's nothing to brick.

### 2. `burn` on Windows — use the browser instead

`burn` is the old tool. Its replacement is [defib](https://github.com/OpenIPC/defib), and there's now a web build that removes the entire Python-and-drivers problem you hit:

**https://openipc.github.io/defib/** — Chrome, Edge or Opera (WebSerial; Firefox and Safari won't work). Nothing to install.

1. **Chip Model** → `gk7202v300`
2. **U-Boot image** → *Download from OpenIPC* — pulled from our releases and SHA-256 verified
3. **Serial Port** → your USB-UART adapter
4. Tick **"Send Ctrl-C after upload (enter U-Boot console)"**
5. **Start Recovery** — when it prints `Waiting for bootrom... power-cycle the device now!`, cut power to the camera and reapply

Start the tool *first*, then power-cycle. It blasts the handshake continuously, which is how it catches the bootrom on fast-booting boards.

> If the U-Boot download step fails, GitHub is refusing the cross-origin fetch for release assets — the page prints the direct URL and the expected SHA-256 so you can fetch it by hand and select the file.

### 3. Back up. You said you skipped it — now it's one button

Once you're at the prompt, hit **Dump Flash**. It reads all 8 MB and saves `flash_gk7202v300_<timestamp>.bin` to your Downloads. Slow (it comes out over `md.b`), but it's a real backup and you currently have none.

### 4. Heads up: your flash is write-protected

This line in your log matters:

```
lk=>6, 0x400000
SRx val: {[1, 0x38], [1, 0x0], [0, 0x0], [0, 0x0]}
```

The vendor bootloader armed the flash's status-register block protection over the lower 4 MB. If `sf erase` later appears to succeed but the data doesn't change, that's why — the write silently no-ops. If you hit it, the `defib agent` path from the CLI clears the BP bits (and SR3.WPS) explicitly before erasing.

### 5. Or just use a programmer

You mentioned wanting to avoid hours of trial and error. If you have a CH341A, the complete image for your exact configuration — bootloader, kernel and rootfs already stitched — is:

```
https://openipc.org/cameras/vendors/goke/socs/gk7202v300/download_full_image?flash_size=8&flash_type=nor&fw_release=lite
```

Note `flash_size=8`, matching your `XM_FM25Q64`. Dump the original chip first.

---

@suchair — same recipe for your gk7205v300, just select `gk7205v300` in the dropdown. Also try the bootloader password `#Ux6@9V&4_Rz` first (see #1494); it's confirmed working on a lot of Goke boards and would save you the trouble.

@xakersolovei — распиновка UART для вашей платы: https://github.com/OpenIPC/wiki/blob/master/en/serial_pins_uart.md — если не найдёте контакты, выложите фото платы с двух сторон в наш Telegram: https://openipc.org/our-channels. Если консоль заблокирована паролем — попробуйте `#Ux6@9V&4_Rz`, либо воспользуйтесь браузерным инструментом выше, который обходит загрузчик целиком.

closed 2026-09-26 by openipc-ai