OpenIPC/firmware #1533: mjsxj03hl youtube streaming

kenkit, 2024-08-21:
Hi, I've just updated to the latest version of OpenIPC but can't get youtube streaming to work. Even night mode is giving me issues, the light sensor is working as expected as I can see the values about 120 in complete darknes with bright light 0. Is that how it should work ?
 
Here is my config
```yaml
system:
  webAdmin: enabled
  buffer: 1024
  httpsPort: 443
  logLevel: debug
  plugins: true
  webPort: 80
image:
  mirror: false
  flip: false
  rotate: none
  contrast: 50
  hue: 50
  saturation: 50
  luminance: 50
osd:
  enabled: true
  template: "%a %e %B %Y %H:%M:%S %Z"
  font: /usr/share/fonts/truetype/UbuntuMono-Regular.ttf
  posX: 10
  posY: 10
nightMode:
  enabled: true
  nightAPI: false
  backlightPin: 39
  irSensorPinInvert: false
  irCutPin1: 49
  irCutPin2: 50
  adcReadout: true
  colorToGray: true
  irCutSingleInvert: false
  lightMonitor: true
  lightSensorInvert: true
  monitorDelay: 0
  lightSensorPin: 60
  maxThreshold: 90
  minThreshold: 50
records:
  enabled: false
  path: /mnt/mmcblk0p1/%Y/%m/%d/%H.mp4
  maxUsage: 95
video0:
  enabled: true
  bitrate: 2046
  codec: h264
  fps: 20
  gopMode: smart
  profile: main
  rcMode: vbr
  size: 1280 x 720
video1:
  enabled: false
  bitrate: 1024
  codec: h264
  fps: 10
  profile: main
  rcMode: vbr
  size: 1280x720
jpeg:
  enabled: true
  fps: 15
  qfactor: 50
  rtsp: true
  size: 1280x720
mjpeg:
  size: 640x360
  fps: 5
  bitrate: 1024
audio:
  enabled: true
  volume: auto
  srate: 8000
  codec: opus
  gain: 20
  outputEnabled: true
  outputGain: 20
  outputVolume: 50
  speakerPinInvert: false
  speakerPin: 63
rtsp:
  enabled: true
  port: 554
hls:
  enabled: true
youtube:
  enabled: true
  key: r7wu-8dvb-de6z-e296-3yt0
motionDetect:
  enabled: false
  visualize: false
  debug: false
  sensitivity: 3
ipeye:
  enabled: false
watchdog:
  enabled: true
  timeout: 10
isp:
  blkCnt: 1
  antiFlicker: 50
  lowDelay: false
outgoing:
  enabled: false
  naluSize: 1200
  server: youtubeurl.hls
```

kenkit, 2024-08-21:
Also can't get the time to show on the stream

widgetii, 2026-08-15:
Do you have issues with recent firmware? If so, please reopen this issue and provide Majestic logs and detailed information about your hardware.

closed 2026-08-15 by widgetii