OpenIPC/firmware #1542: HiSilicon HI3518CV100 Bricked

OK9UWU, 2024-09-02:
Hello.

I have a ethernet only, 8MB flash version of a camera with this SoC.
I flashed the openipc-hi3518cv100-lite-8mb.bin file onto it via TFTP and after executing the command "RESET" the camera seems to be bricked!
No response on ethernet, no response via UART, no activity from the device apart from a dimmly lit "link" LED on the ethernet port..... However no connection gets ever established.
I have a full backup of the factory firmware.

zerog2k, 2024-09-02:
For community support, try the telegram channels.
Likely if no bootloader activity on uart, you will need to look to reflashing SPI flash directly.

OK9UWU, 2024-09-02:
> For community support, try the telegram channels. Likely if no bootloader activity on uart, you will need to look to reflashing SPI flash directly.

I joined the English Telegram group already, but i don't have any rights to post anything yet.

As per flashing the SPI chip directly, yeah i'll try that tomorrow.

OK9UWU, 2024-09-03:
Alright, progress, slow but steady.

I flashed the backup image onto the EEPROM and then reflashed it with openipc-hi3518ev100-lite-8mb.bin via tftp (same steps as for the cv100) and this time it did boot!.... But without the network interface present.

RenaKunisaki, 2024-09-24:
Same problem here. All commands executed OK, but after reset, it just prints some spaces. flashrom doesn't seem to support this particular flash chip either, so I can't restore the backup.

RenaKunisaki, 2024-09-27:
Well I was able to get flashrom to recognize it (by disconnecting vcc, oddly enough) and flash the backup, and that works. Flashing the OpenIPC image again gives the same result.

Also for this model, the Ultimate download link is broken, and redirects endlessly.

RenaKunisaki, 2024-10-02:
Did some more testing:

If I flash everything except the bootloader, it hangs upon booting the kernel.

If I only flash the userspace, it boots with all kinds of errors in the console, which also doesn't accept any input.

closed 2026-11-16 by openipc-ai