OpenIPC/firmware #1556: RTL8812EU init failed on GK7205V300

afilez, 2024-09-22:
Tried link the BL-m8812EU2 WiFi adapter to GK7205V300 

lsusb out:
root@openipc-gk7205v300:~# lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 002: ID 0bda:a81a
Bus 002 Device 001: ID 1d6b:0003

inmod and get dmesg output(debug driver):

RTW: [HALMAC]55772M
HALMAC_MAJOR_VER = 1
HALMAC_PROTOTYPE_VER = 6
HALMAC_MINOR_VER = 7
HALMAC_PATCH_VER = 12
RTW: Chip Version Info: CHIP_8822E_U2_2T2R_RomVer(1)
RTW: config_chip_out_EP OutEpQueueSel(0x07), OutEpNumber(3)
RTW: NR_RECVBUFF: 8, recvbuf_nr: 8
RTW: MAX_RECVBUF_SZ: 20480
RTW: NR_PREALLOC_RECV_SKB: 8
RTW: Init Rx buffer skb ok count=8, fail count=0
RTW: reg 0x4e0, usb write 1 fail, status:-71
RTW: ERROR _halmac_reg_write_32: I/O FAIL!
RTW: reg 0x4e0, usb write 1 fail, status:-71
RTW: ERROR _halmac_reg_write_8: I/O FAIL!
RTW: reg 0x7f, usb read 1 fail, status:-71 value=0x0, vendorreq_times:1
RTW: reg 0x0, usb read 1 fail, status:-71 value=0x0, vendorreq_times:1
RTW: reg 0x2e, usb write 1 fail, status:-71 value=0x15, vendorreq_times:1
RTW: reg 0x4e0, usb write 1 fail, status:-71
random: fast init done
RTW: reg 0x1045, usb write 1 fail, status:-71 value=0x7a, vendorreq_times:1
RTW: reg 0x1080, usb read 4 fail, status:-71 value=0x0, vendorreq_times:1
RTW: SetHwReg: bMacPwrCtrlOn=1
RTW: rtl8822e_fw_dl fw source from array
RTW: reg 0x41a, usb read 1 fail, status:-71 value=0x0, vendorreq_times:1
RTW: reg 0x1703, usb read 1 fail, status:-71 value=0x0, vendorreq_times:1
RTW: reg 0x10d, usb write 1 fail, status:-71 value=0xc0, vendorreq_times:1
RTW: reg 0x422, usb write 1 fail, status:-71 value=0x31, vendorreq_times:1
RTW: reg 0x1208, usb read 4 fail, status:-71 value=0x0, vendorreq_times:1
RTW: reg 0x1200, usb write 4 fail, status:-71 value=0x18780030, vendorreq_times:1
RTW: ERROR _halmac_reg_write_32: I/O FAIL!
RTW: reg 0x204, usb write 2 fail, status:-71 value=0x8000, vendorreq_times:1
RTW: reg 0x422, usb read 1 fail, status:-71 value=0x0, vendorreq_times:1
RTW: reg 0x1208, usb read 4 fail, status:-71 value=0x0, vendorreq_times:1
RTW: ###=> urb_write_port_complete status(-71)

... and more

RTW: 0x17f0 0x00000000  0x00000000  0x00000000  0x00000000
RTW: _debug_dlfw_fail: 0x80 = 0x0060 (after 10 secs)
RTW: ERROR rtl8822e_fw_dl Download Firmware from array failed
RTW: SetHwReg: bMacPwrCtrlOn=0
RTW: reg 0x80, usb read 2 fail, status:-71 value=0x0, vendorreq_times:1
RTW: reg 0xef, usb read 1 fail, status:-71 value=0x0, vendorreq_times:1
RTW: reg 0x6, usb read 1 fail, status:-71 value=0x0, vendorreq_times:1
RTW: reg 0x4e0, usb write 1 fail, status:-71
RTW: reg 0x5, usb read 1 fail, status:-71 value=0x0, vendorreq_times:1
RTW: rtw_hal_read_chip_info in 10540 ms
RTW: usb_read_port_cancel
RTW: WARN free_recv_skb_queue not empty, 8
RTW: rtw_usb_primary_adapter_init Failed!
usbcore: registered new interface driver rtl88x2eu
RTW: module init ret=0

afilez, 2024-09-26:
resloved,

closed 2024-09-26 by afilez

widgetii, 2024-09-26:
Can you explain what helped solve the issue?

afilez, 2024-09-26:
I guess the new adapter just set as USB 3.0, and it may need set some working params and store in inner flash. so:

LanfesLabs, 2025-09-05:
Hi guys, does anyone have binary 8812eu.ko for gk7205V300? I got one for gk7205V200 , but malual try to load it as "modprobe 8812eu" results in "Segmentation fault" ((((