OpenIPC/firmware #1579: How To Correct White Balance?

FidgetyRat, 2024-10-13:
I recently installed OpenIPC on a birdhouse camera (based on GK7605V100 sc2239) with no problems. It's up and running, but I cannot find any means of tweaking the white balance. Under the camera's original Chinese firmware, the image looks fine, but under OpenIPC, it's quite yellow/green (see images). The project overall seems really well done, but the documentation is simply just not there in most cases and I can't find any reference on how to tweak the image outside of setting the HUE/Sat/Lum settings in Majestic which won't affect my white balance issues.

OpenIPC
[image]

Original Firmware
[image]

Can anyone make a suggestion on what I need to fix?

FidgetyRat, 2024-10-30:
Nobody?

widgetii, 2026-10-02:
Sorry this sat for so long. We reproduced it on a lab GK7605V100 + SC2239 with a ColorChecker in front of it. The yellow/green you saw has two separate causes.

**1. The SC2239 had no colour calibration of its own on this chip family.** A camera whose sensor has no IQ profile runs `/etc/sensors/iq/default.ini`, and on GK7205V200/GK7605V100 that file is the IMX307's. So your camera ran another sensor's colour matrices. #2515 adds `sc2239.ini`, measured on that lab camera under a tungsten lamp and under window daylight, and the camera picks it up automatically. Exposure, noise reduction and gamma are unchanged; only colour is.

**2. The automatic white balance can mistake the light when the scene itself is mostly one warm colour.** Its one cue is how colourful surfaces are, and a warm-coloured wall in daylight looks exactly like a white wall under a lamp. On the lab camera, a peach-walled room lit by a window made the AWB choose lamp-like gains (red about 30% too low), whatever profile was loaded, and no white-balance setting separates the two cases. A birdhouse interior lit through its entrance hole is that kind of scene. Making the AWB tell the light sources apart needs a new cue, which is still being worked on.

**What you can do on your own camera now**, with the next nightly (these changes were merged today):

- `/metrics` shows what the white balance decided: `isp_awb_color_temp`, `isp_awb_rgain`, `isp_awb_bgain`.
- With a ColorChecker (or any 24-patch chart) in view, the WebUI's raw editor (**Calibrate** tab) measures the chart and can save a white balance and colour matrix to the camera. Several matrices fit a chart equally well, and some of them amplify the sensor's noise; the editor holds the one it picks to the noise of your current picture and says what that cost. On a sensor with no profile of its own, saving creates `/etc/sensors/iq/<sensor>.ini` rather than changing the shared one, and **Put back** removes it. Under the light the camera actually lives in, this fixes the white balance for that light.

If you still have this camera, a snapshot from the current nightly (`/image.jpg`) and the `isp_awb_*` lines from `/metrics` would show which of the two you are seeing now.

closed 2026-11-09 by openipc-ai