OpenIPC/firmware #1589: Hi3516ev300 cam and Hi3536dv100 nvr, can't get osd working

Warfull, 2024-10-25:
NVR is Hi3536DV100 , master+8d995dt, 2024-10-20
Cam is Hi3516ev300, master+874aa01, 2024-10-21
Camera module is IVG-85HG50PYA-S.
Osd is drawn but no data update.
I can see trafic on the uart port with oscillioscope, both tx and rx signal carry data.
I have tryed all possible configs in telemetry.conf,, vdec.conf
router=0,1,2

On the cam side Mavlink stream is present :

root@openipc-hi3516ev300:/usr/bin# mavfwd -m /dev/ttyAMA0 -o 0.0.0.0:14550 -v
Verbose mode!
Listening on /dev/ttyAMA0...
Detected MAVLink ver: 1.0  (254)
System_id = 1
Mavlink msg 30 no: 132
1 Pckts / 36 bytes sent
Mavlink msg 30 no: 134
1 Pckts / 68 bytes sent
Mavlink msg 30 no: 136
1 Pckts / 68 bytes sent
Mavlink msg 30 no: 138
1 Pckts / 68 bytes sent
Mavlink msg 30 no: 140
1 Pckts / 68 bytes sent
...

On nvr side :

root@openipc-hi3536dv100:/# vdec -c h265 -d stream --osd --mavlink-port 14750
[HI_VERSION=Hi3536D_MPP_V1.0.1.0 B040 Release]
> VDEC picture block size = 7856640, 330240
HDMI start success
> VO: 1280x720 @ 60
> VDEC Protocol = Type: H265, PPS: 64, SLICE: 100, SPS: 16
ERROR: Unable to bind MavLink port: Address already in use
fbg_fbdevSetup: '/dev/fb0' (1280x720 (1280x720 virtual) 32 bpp (8/16, 8/8, 8/0) (8294400 smen_len) 5120 line_length)
Components: 4
^C
root@openipc-hi3536dv100:/# killall vdec
root@openipc-hi3536dv100:/# vdec -c h265 -d stream --osd --mavlink-port 14750
[HI_VERSION=Hi3536D_MPP_V1.0.1.0 B040 Release]
> VDEC picture block size = 7856640, 330240
HDMI start success
> VO: 1280x720 @ 60
> VDEC Protocol = Type: H265, PPS: 64, SLICE: 100, SPS: 16
fbg_fbdevSetup: '/dev/fb0' (1280x720 (1280x720 virtual) 32 bpp (8/16, 8/8, 8/0) (8294400 smen_len) 5120 line_length)
Components: 4
^C

Warfull, 2024-11-04:
I have tried many things :
in telemetry.conf if I set router=0, I have an error message saying "mavfwd not found", I don't remember exactly if its on drone side or gs side.
If I set router=1 it seems ok but on the gs side I have an error saying "Unable to decrypt packet 0x......."

Warfull, 2024-11-06:
"Unable to decrypt packet" message come from wfb_rx and it seems it is not the problem, the message stop when a SESSION_PACKET arrive initializing the session_key.

lida2003, 2024-11-10:
Ah.... I saw this. @Warfull Did you get osd on the ground (3536) working?

As I saw it's working on airunit https://github.com/OpenIPC/msposd/issues/18

Warfull, 2024-11-11:
No way to get the osd working.
I use msposd and it's working fine except for AHI.

Warfull, 2025-01-27:
Now it's working.
On the camera :
router=0 (mavfwd)
and on the ground station
router=1 (mavlink-routerd)
It seems to me a little weird but if I set same config on both it does not work.

On FC:
SERIALx_PROTOCOL=2 (Mavlink2)
SERIALx_BAUD=115
SRx_EXT_STAT=1
SRx_EXTRA1=5
SRx_EXTRA2=5
SRx_EXTRA3=5

closed 2025-01-27 by Warfull