OpenIPC/firmware #1602: Add support for Hi3519DV500 cpu and IMX585 sensor

akzami, 2024-11-08:
Currently there is no support for some modern chips like Hi3519DV500.

patrickelectric, 2025-03-27:
That would be nice!

eldinmiller, 2025-11-14:
Here here!

widgetii, 2026-10-20:
Status update on Hi3519DV500 (and the Hi3516DV500 die).

**The SoC is supported now**, verified in qemu-hisilicon only -- nobody here has a board yet:

- Kernel: split NOR layout, DRAM measured by the kernel (up to 2 GiB) with the media zone laid out automatically (OpenIPC/linux#102, #107, #109).
- U-Boot: built from OpenIPC/u-boot's `openipc-hisilicon` tree (#32, #35), signed boot images per board on the firmware release: `boot-hi3519dv500-dmeb-nor.bin`, `boot-hi3519dv500-dmebpro-nor.bin`. SD/USB recovery is in review (OpenIPC/u-boot#38).
- Firmware: `openipc.hi3519dv500-nor-ultimate-split.tgz` from the nightly (OpenIPC/firmware#2638); flash layout 512K boot + 256K env, firmware from `0xC0000` -- see the [NOR split layout](https://github.com/OpenIPC/wiki/blob/master/en/nor-split-layout.md) wiki page. Install instructions and full images per board on openipc.org: [dmeb](https://openipc.org/cameras/vendors/hisilicon/socs/hi3519dv500-dmeb), [dmebpro](https://openipc.org/cameras/vendors/hisilicon/socs/hi3519dv500-dmebpro); `defib install` supports it too.

**Sensors**: the image ships the drivers built from OpenIPC/openhisilicon for the sensors HiSilicon's SDK supports on this chip -- SC450AI, SC850SL, OS04A10 (and slave), OS08A20, IMX347 (slave), IMX515, GST 412C.

**Not there yet**:

- **No streamer on this SoC**: majestic and divinus have no aarch64 Hi3519DV500 build, so the image boots and loads the media stack, but nothing captures or streams video yet.
- **IMX585** has no driver for this platform: it is not in the vendor SDK's support list, and openhisilicon has none. **IMX678** exists in openhisilicon only for GK7206V100's SDK and would need porting to the DV500 API.

Both need someone with a Hi3519DV500 board and the sensor to bring up and test. If you have one and can run a test build, say so here.