OpenIPC/firmware #1603: MSPOSD on the ground needs wfb_tx support

lida2003, 2024-11-10:
@MarioFPVdev  Here is a new feature from msposd, just a couple of days ago, which might need wfb support. I think this feature is a little bit late than your PR. Please check below info: 

- MSPOSD does not require telemetry_tx or telemetry_rx (https://github.com/OpenIPC/firmware/pull/1597)

MSPOSD draws OSD on airunit doesn't need wfb support. And it has a ground OSD function, which was implemented days ago on x86.

I'm trying this on Jetson board(https://github.com/OpenIPC/msposd/issues/17) now, but it needs time for configuration or some tuning later.

The msposd readme says: 

> This mode uses one MSPOSD running on the air unit, collecting MSP DisplayPort data, aggregating it and sending it via wfb_ng to the ground.
On the ground another instance of MSPOSD renders these data and overlays it on the screen.
Tested on x86 Ubuntu 6.5.0.28
On the cam msposd can be started like this to forward via port 14550 (default mavlink, should not be used)
msposd --channels 7 --master /dev/ttyS2 --baudrate 115200 --out 127.0.0.1:14550 --matrix 11 --ahi 3 -r 30
On the ground:
msposd --master 127.0.0.1:14550  --osd -r 50 --ahi 3 --matrix 11 -v

MarioFPVdev, 2024-11-10:
GS-MSPOSD for x86 is not a new feature. We are aware of that and once we have MSPOSD running on our supported Groundstation platforms we will commit an update that will support it. We will update the community once it is done. Stay tuned.

lida2003, 2025-01-27:
It works on latest firmware.

closed 2025-01-27 by lida2003