OpenIPC/firmware #1610: Asecam / Jieuno (fullhan / vatilon) integration

mtrakal, 2024-11-20:
Hello, I would like to help / provide some small info for cameras.

SoC of my cameras is fh8856v200 or fh8856/IF56N

There is my repo with extracted partitions, etc: https://github.com/mtrakal/ipc-vatilon

I'm not able to help with integration itself, but if you will need anything get from cameras or be a beta tester with firmware I can be that person :).

darrylb123, 2024-11-26:
I have a number of the ASECAM Vatilon PB1 cameras. They can be affected by something on the network that causes them to all reboot at once. This can continue for some time, all rebooting regularly. I am willing to assist where I can. I upgraded the firmware using the Windows tool. It didn't help with the reboots but it may be the way to install alternative firmware.

mtrakal, 2024-11-28:
> I have a number of the ASECAM Vatilon PB1 cameras. They can be affected by something on the network that causes them to all reboot at once. This can continue for some time, all rebooting regularly. [...]

In the camera WebGui is possible to setup autoreboot. Maybe you have it enabled? 
I have no issue with reboots of camera when this settings is disabled. But I have it enabled to reboot once per day to cleanup possible FW memleaks, sync time again, etc.

[image]

darrylb123, 2024-11-28:
No, it's not set to auto-reboot. I can crash all the cameras together if I
use wsdd on the same  network.

mtrakal, 2024-11-29:
> I upgraded the firmware using the Windows tool. It didn't help with the reboots but it may be the way to install alternative firmware.

Yeah, I already caught FW upgrade packets, but it's a bit useless because it's encrypted. So App and Cam changed the keys and session and encrypted the rest of the communication. So until we know the algorithm and how the key is build, we are lost I think this simple way for upgrade FW, but we have 2 others  ⏬.

Discover camera is based on Broadcast on Network L2 (ARP broadcast on same network area):

```json
{
    "DeviceType": "IPC",
    "ChannelNum": 1,
    "DeviceModel": "PB4",
    "DeviceModel2": "PB",
    "DeviceName": "PB4",
    "CloudID": "__CLEARED__",
    "SerialNo": "__CLEARED__",
    "NetType": 1,
    "MAC": "__CLEARED__",
    "DHCP": 0,
    "Address": "10.0.0.33",
    "Submask": "255.255.255.0",
    "GateWay": "10.0.0.254",
    "DNS": "10.0.0.252",
    "TCPPort": 8600,
    "RTSPPort": 554,
    "HTTPPort": 80,
    "Language": "English",
    "SoftwareVersion": "V1.13.23-20240524",
    "DeviceRunTime": 78,
    "QWTEnable": false,
    "IPAdaptive": 1,
    "LampType": 1,
    "DayNightSwitch": 0,
    "P2PType": 4,
    "P2PPID": "__CLEARED__",
    "P2PQRCode": "__CLEARED__",
    "LensCount": 1,
    "StreamCount": 2,
    "StreamNumPerChannel": 2,
    "NetTypeSet": 1,
    "NetProtocolSet": 143,
    "Status": 2,
    "Version": 1,
    "Scode": "__CLEARED__"
}
```

After that application sends UDP packet to camera IP address with body:
```json
{
  "operation": "login",
  "param": {
    "user": "JmsRxTtVEW+J5da5aSt3lw==",
    "password": "bC+UKGMPDDiynIUE9ViJzw==",
    "heartbeat_timeinterval": 1,
    "heartbeat_timeoutcounts": 3,
    "sessionid": 179511328,
    "encryptionkey": "EOtDcH4DFPJZj2CDhLFNAg==",
    "roletype": 5
  }
}
```

So we know, that decrypted `user = admin` and `password = 123456`.

Response is:
```json
{
  "operation": "login_response",
  "code": 0,
  "msg": "successful ",
  "sessionid": 179511328,
  "encryptionkey": "HO55v18ZnC9cL6pJe0U9Tg=="
}
````

And the rest is encryped with new key.

---

But we already have ROOT access to telnet, so I don!t think, that we need use standard way how to upgrade firmware, when you can (I supose) re-mount partition as write or use dd to rewrite flash memory. Not sure, but expecting, that with root bash / bussybox it is possible. 

---

And second tested variant - the author of [articles on Reddit](https://www.reddit.com/r/hardwarehacking/comments/1cuu1wf/hacking_an_asecam_ip_camera_part_1/) already used UART and flash his own custom FW (changed default password to telnet with custom one). So flash new FW is not the main problem I think :)

darrylb123, 2024-11-29:
I opened one up and took some pics. Trouble is, after I checked the chip
numbers I didn't get a pic of the camera chip. Only the flash and ethernet.
I've ordered a new one to have a decent look around inside.

L4ky, 2024-12-28:
Hi, i have a few Simicam cameras from Aliexpress.
They are Vatilon PB4 boards and various firmware version:

-  V1.13.23-20240524 ( linux-4.9-11 , uboot-2016-15 )
-  V1.14.10-20240725 ( linux-4.9-11 , uboot-2016-15 )

On Vatilon website there is 1.15.11 firmware version but still didn't tried it.

I have root access via telnet on port 2360.

Let me know if i can contribute in some way.

attila-lendvai, 2025-02-23:
i also have two ASECAM cameras that say they are PB1 in their web UI.

i'm relatively experienced in lowlevel hw hacking, i can also help, but i'm very new to openipc.

attila-lendvai, 2025-02-24:
i have two of these: https://www.aliexpress.com/item/1005006826151802.html

i have used `nmap` to scan the ports, and 2360 was not listed as open, but i can indeed telnet to that port.

but the `admin` `123456` doesn't work there (it works through the web UI, though).

any other ideas for login/pw?

L4ky, 2025-02-25:
> i have two of these: https://www.aliexpress.com/item/1005006826151802.html
> 
> i have used `nmap` to scan the ports, and 2360 was not listed as open, but i can indeed telnet to that port.
> 
> but the `admin` `123456` doesn't work there (it works through the web UI, though).
> 
> any other ideas for login/pw?

Try root/ipc@hs66

attila-lendvai, 2025-02-25:
> Try root/ipc@hs66

it worked! :) thank you!

oglabogla, 2025-10-07:
hello everyone i have a chinese camera which runs a V1.14.91-20241120, but everytime my nvr reboots (once a week) the camera fails to connect and needs a manual password input
can someone supply me with the V1.12 version
thank you

eribito, 2026-11-11:
Adding data for two more units on this board, in case it helps.

I have two of these — board **IF56N** (FH8856V200), both still on stock firmware. Everything below was read over the stock root telnet (read-only); I haven't flashed OpenIPC on them.

- `/proc/driver/chip`: `chip_name: FH8856`, `ddr_size: 1024Mbit` (128 MB)
- `bootargs` (= `/proc/cmdline`): `mem=36M console=ttyS0,115200 root=/dev/mtdblock2 rootfstype=squashfs mtdparts=spi_flash:320K(uboot),2560K(kernel),4096K(appfs),832K(custom),320K(config),64K(data)`
- There's a real, CRC-valid U-Boot environment inside mtd0 at `0x40000`–`0x50000` (a default env is also baked into the code). `bootargs`/`bootcmd` live there, so `mem=` and `mtdparts` are editable through the env — no bootloader rebuild needed. Stock boots `mem=36M`; whether `mem=32M` is still required looks build-dependent now (PR #2821 reworks the Fullhan pool sizing).
- Sensor: GC4653 — inferred from `/usr/app/sensor/gc4653_mipi_*.hex` (the driver is in the firmware; not physically confirmed)
- `/usr/app/product` (software-reported): `product=IF56N`, `ota_url=/fh8856/IF56N/IF56N.bin`
- Stock userland: Linux 3.0.8 "FHLinux", busybox, `thttpd/2.29`; root telnet on 2360 (factory creds already noted in this thread)
- Flash: 8 MB SPI NOR, partitioned as above; `cat /dev/mtd0..5` gives a clean full image (NOR, no OOB/bad-block handling needed)

If you need to back up the stock flash first and your unit is network-isolated (no route back out) with a stripped busybox (no `od`/`base64`/`nc`), you can still dump the 8 MB over the existing root telnet. busybox has `inetd`, so serve the mtd devices on a high port and pull them from your side. Do this **only on a trusted/restricted network** — it exposes the whole flash (configs/credentials included) to anyone who can reach the port — and tear it down afterwards:

```sh
# on the camera (telnet just sets this up; the transfer is a separate TCP connection):
echo '#!/bin/sh' > /tmp/d.sh
echo 'exec cat /dev/mtd0 /dev/mtd1 /dev/mtd2 /dev/mtd3 /dev/mtd4 /dev/mtd5 2>/dev/null' >> /tmp/d.sh
echo '29000 stream tcp nowait root /bin/sh sh /tmp/d.sh' > /tmp/i.conf
busybox inetd /tmp/i.conf
# from your machine:  nc <camera-ip> 29000 > flash_8M.bin   (expect 8388608 bytes)
# when done:  kill the inetd you started, rm /tmp/i.conf /tmp/d.sh, and confirm the port is closed
```

Happy to run read-only diagnostics on these if it helps. I'd only attempt a flash on the bench, with UART and SPI-programmer recovery on hand.

eribito, 2026-11-11:
The stock firmware does load it from a script, and `/proc/driver/vmm` exists on the stock 3.0.8 kernel. Both read-only, same IF56N units.

`/usr/app/driver/load_driver.sh` (run from `/usr/app/run.sh`):

```sh
insmod vmm.ko mmz=anonymous,0,0xa2400000,90M anony=1
insmod rtvbus.ko ra=0xa7e00000 rs=65536 fn=rtthread_arc_human_detect.bin fa=0xa7f00000
```

So with `mem=36M` the vendor gives the pool everything from `0xA2400000` up to `0xA7E00000` (90 MB) and keeps the top 2 MB (`0xA7E00000`–`0xA7FFFFFF`) for `rtvbus`, an RT-Thread image on the ARC core that does human detection. 36 + 90 + 2 = 128 MB. If the 64M cap is ever raised, that top 2 MB is worth keeping out of the pool.

`/proc/driver/vmm` live, with the stock app streaming main + sub:

```
VMM ZONE:  phys(0xA2400000~0xA7DFFFFF), size: 92160KB,  name: "anonymous"
  BLOCK00:  phys(0xA2400000~0xA242CFFF), size: 180KB,   name: "vpu-sys"
  BLOCK01:  phys(0xA242D000~0xA3C8DFFF), size: 24964KB, name: "vpu-ch-0"
  BLOCK02:  phys(0xA3C8E000~0xA3DDFFFF), size: 1352KB,  name: "vpu-ch-1"
  BLOCK03:  phys(0xA3DE0000~0xA417AFFF), size: 3692KB,  name: "h264-sys"
  BLOCK04:  phys(0xA417B000~0xA41D0FFF), size: 344KB,   name: "h265-sys"
  BLOCK05:  phys(0xA41E0000~0xA5E76FFF), size: 29276KB, name: "isp"
  BLOCK06:  phys(0xA5E77000~0xA7962FFF), size: 27568KB, name: "venc-ch-0"
  BLOCK07:  phys(0xA7963000~0xA7B0BFFF), size: 1700KB,  name: "venc-ch-1"
  (BLOCK08–15: OSD, 4–32KB each)
  BLOCK16:  phys(0xA7B2C000~0xA7B8EFFF), size: 396KB,   name: "PD"
  BLOCK17:  phys(0xA7B8F000~0xA7BC0FFF), size: 200KB,   name: "def_mmb"
  BLOCK18:  phys(0xA7BC1000~0xA7BF2FFF), size: 200KB,   name: "def_mmb"
  BLOCK19:  phys(0xA7BF3000~0xA7CEAFFF), size: 992KB,   name: "HUAMN_DETECT_CA"
 total size=90.000MB,used=88.859MB,remain=1.140MB,zone_number=1,block_number=20
```

Those 89 MB are inflated by the stock pipeline. The same script sets:

```sh
echo "vi_2560_1440"    > /proc/driver/vpu
echo "cap_0_3840_2160" > /proc/driver/vpu
echo "cap_1_800_576"   > /proc/driver/vpu
echo 'support4k_on'    > /proc/driver/vpu
echo 'ispp_three_buf'  > /proc/driver/isp
```

The sensor comes in at 2560×1440 and the main channel is upscaled to 3840×2160 (vpu-ch-0 is about 3840×2160×1.5×2).

The pool follows that boot-time `cap_0`, not the stream setting. I switched the main stream to 2560×1440 in the stock web UI on both units (ffprobe confirms 2560×1440 on the wire) and `/proc/driver/vmm` didn't move at all: same 88.859 MB, same block sizes. So on stock the buffers are sized at driver load, and the 89 MB is what `cap_0_3840_2160` costs.

Scaling vpu-ch-0 and venc-ch-0 to a 2560×1440 `cap_0` and keeping the rest gives roughly 60 MB. That's a back-of-envelope figure, not a measurement, but it suggests two streams at native resolution would be tight under a 64M cap, with the ISP's 29 MB as the largest fixed block.

Still no flashing on my side.

darrylb123, 2026-11-16:
I have a uboot prompt on my Asecam with PB1.
I have a tftp server set up with the openipc installation files.
There are no instructions for the FH88856v200 and I'm not on Telegram.
I see int the boot messages that the kernel is loaded from 0xA1000000use default flash ops...
spi_flash_probe_default multi wire open flag is 0
8192 KiB default_flash at 0:0 is now current device
## Booting kernel from Legacy Image at a1000000 ...
   Image Name:   Linux-3.0.8
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2393484 Bytes = 2.3 MiB
   Load Address: a0008000
   Entry Point:  a0008000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK
OK
What is the install procedure?

eribito, 2026-11-16:
Thanks for chasing this down. I can confirm it from my side: my two IF56N units run exactly the firmware in the PB1 dump, so whatever holds for that dump holds for them.

I ran `git hash-object` on my stock flash backups and compared the result with the blob SHAs GitHub lists for `pb1/mtd_complete` in mtrakal/ipc-vatilon. The first three partitions are byte-identical, on both cameras:

| partition | git blob |
| --- | --- |
| mtd0 (uboot, 320K) | `d06233410eadfa460cb8c7ef532dba64288ea163` |
| mtd1 (kernel, 2560K) | `0bc44d210e58bc5fa636cd784e3312f10bc58575` |
| mtd2 (appfs, 4096K) | `7b9bd1099febfff758fd30ad5b8545a29a64b1f1` |

Only custom/config/data differ, and those are per-device state.

#2872 has only run in emulation so far, so here is what the stock 3.0.8 kernel reports on the real chips (read-only, over the stock telnet).

`/proc/driver/chip`, the same on both units:

```
chip_name       : FH8856
ddr_size        : 1024Mbit
plat_id         : 0x17092901
pkg_id          : 0x7840d
```

`plat_id` matches the QEMU value in #2872, but `pkg_id` doesn't (`0xd` there, `0x7840d` here). I don't know if that comes from the model or from how the two kernels print it.

`/proc/cpuinfo` on one unit: `Hardware : FH8852`, `ARMv6-compatible processor rev 7 (v6l)`.

A correction to my first comment: I wrote "IF56N (FH8856V200)". I took the V200 from the opening post here, not from the chip, which only reports `FH8856`. I haven't read the marking on the package.

Both cameras are still on stock firmware, and I'd only flash one on the bench with UART and an SPI programmer.

mtrakal, 2026-11-16:
@openipc-ai 

Some requested info for PB4 (I hope so, that I choose correct camera):

```
# cat /usr/app/product
product=PB
uboot=15
kernel=11
appfs=11410
ota_url=/fh8856v200/PB/PB.bin

# cat /proc/driver/chip
chip_name       : FH8856V200
ddr_size        : 1024Mbit
plat_id         : 0x19112201
pkg_id          : 0x7850b

# cat /proc/meminfo
MemTotal:          32256 kB
MemFree:           13464 kB
MemAvailable:      15600 kB
Buffers:              92 kB
Cached:             8544 kB
SwapCached:            0 kB
Active:             4356 kB
Inactive:           7488 kB
Active(anon):       3220 kB
Inactive(anon):     5420 kB
Active(file):       1136 kB
Inactive(file):     2068 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:          3224 kB
Mapped:             5008 kB
Shmem:              5432 kB
Slab:               3332 kB
SReclaimable:        680 kB
SUnreclaim:         2652 kB
KernelStack:         800 kB
PageTables:          396 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:       16128 kB
Committed_AS:      68932 kB
VmallocTotal:     999424 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB

# cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 2.00
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : FH8856V200
Revision        : 0000
Serial          : 0000000000000000

# cat /proc/driver/vmm
[... 140 lines cut ...]
         Cur VO Mode: SCAN     Rotate: 0
      STATUS:
         PIC_FINISH_CNT:         55220050
         VPU_ABONDON_FRM_CNT:    0(0 - 0)
         CHANGE_PARAM_ABOND_CNT: 0
         VPU_OUTPUT_FRAME_RATE:1250/107(11.66)
         NEXT_EXPECT_PTS:  0x2ccbf6eeabb PTS_NOW: 0x2ccbf6f5701

  [CHAN 2]:
      MEM INFO:
         [BUF 0]DATA   ADDR:0xa3e55000 SIZE:0x56400 USECNT:0
      PARAM INFO:
         PIC Width: [640] Height: [368]
         Max Process Size: [width: 640] [height: 368]
         Cur VO Mode: SCAN     Rotate: 0
      STATUS:
         PIC_FINISH_CNT:         55220050
         VPU_ABONDON_FRM_CNT:    0(0 - 0)
         CHANGE_PARAM_ABOND_CNT: 0
         VPU_OUTPUT_FRAME_RATE:10000/799(12.50)
         NEXT_EXPECT_PTS:  0x0 PTS_NOW: 0x2ccbf6f5701

  [CHAN 3]:  NOT CREATED
------------------------------------------------------------
```