OpenIPC/firmware #1616: No Video Preview. ssc377d lite with imx335

jawadbrayer, 2024-11-25:
I am trying to get the video preview, but without success.
[image]

[image]

**logs of majestic started manually: testing with RTSP stream also fails** 
```
root@openipc-ssc377d:~# majestic -s
08:04:00  <       majestic> [    main] main@237                      Majestic Lite for SigmaStar, version master+8bf12fe, built on 2024-11-24
08:04:01  <       majestic> [watchdog] watchdog_start@94             Detected infinity Watchdog
08:04:01  <       majestic> [watchdog] watchdog_start@115            Watchdog timeout set to 300 seconds
08:04:01  <       majestic> [   httpd] start_web_server@365          Set server path: /var/www
08:04:01  <       majestic> [   httpd] new_http_server@353           HTTP server started on 0.0.0.0:80
08:04:01  <       majestic> [    main] init_sdk@66                   Loading main configuration: /etc/majestic.yaml
08:04:01  <       majestic> [     sdk] start_sdk@1534                Initializing Sigmastar SDK, mi_sys_version_3.011
08:04:01  <       majestic> [     sdk] mi_snr_get_resolution@505     Sensor index 0: 2592x1944@25fps
08:04:01  <       majestic> [     log] printf@292                    [MI_ISP_CUS3A_Enable] AE = 1, AWB = 1, AF = 1
08:04:01  <       majestic> [     log] printf@292                    [MI_ISP_CUS3A_Enable] AE = 1, AWB = 1, AF = 1
08:04:01  <       majestic> [     log] printf@292                    [MI_ISP_CUS3A_Enable] AE = 1, AWB = 1, AF = 1
08:04:01  <       majestic> [     sdk] mi_venc_create_chn@1001       Channel 0: 2592x1944@20fps
08:04:01  <       majestic> [     sdk] mi_venc_create_chn@1002       - [H264, 4096Kb, RC1, GOP20]
08:04:01  <       majestic> [     sdk] mi_venc_create_chn@1066       - [minQp: 12, maxQp: 48, qpDelta: -4, slice: 0]
08:04:01  <       majestic> [     sdk] mi_venc_create_jpeg@899       Channel 2: 2592x1944@5fps
08:04:01  <       majestic> [     sdk] mi_venc_create_jpeg@900       - [JPEG, Q50, 4900Kb]
[    AeInit] FPS (Init, Input, Output, Cur, Deband) = (   27,   27,   27,   27,   27)
[    AeInit] Use AE Short Histogram = Valid
[    AeInit] --> Change  :      1,    CurYx10 :      0, CurY1x10 :      0   , Targetx10 :    470, BVx16384 : 111344, FNx10 :  16, IspGain :   1024, SensorGain :  10240, Shutter  :     72 us, Step      :   8080 ns

08:04:02  <       majestic> [     sdk] mi_isp_get_file@148           Load isp file /etc/sensors/imx335.bin
08:04:02  <       majestic> [     log] printf@292                    [MAJOR]: ispversion(1) in sdk, ispversion(1) in binfile.
08:04:02  <       majestic> [     log] printf@292                    [MINOR]: ispversion(2) in sdk, ispversion(2) in binfile.
[   pAE_SetPlainShortExpTbl]-------------HW:1024/1445888/72/333333-----TBL:1024/8192/30/1400-----Plain Table is not valid!
08:04:02  <       majestic> [     log] printf@292                    ******************** Load api bin Success - /etc/sensors/imx335.bin ********************
08:04:02  <       majestic> [     sdk] mi_isp_set_image@258          Set dynamic range: 0
08:04:02  <       majestic> [    rtsp] rtsp_init@28                  RTSP server started on port 554
08:04:17  <       majestic> [   night] set_night_mode@128            Change mode to NIGHT
08:04:19  <       majestic> [   night] set_night_mode@128            Change mode to DAY
08:04:19  <       majestic> [     sdk] mi_isp_set_image@258          Set dynamic range: 0
08:04:36  <       majestic> [  client] RtspClient_before@403         RTSP OPTIONS rtsp://192.168.11.101:554/stream=0 CSeq=2 from 192.168.11.104:50548
08:04:36  <       majestic> [  client] RtspClient_before@403         RTSP OPTIONS rtsp://192.168.11.101:554/stream=0 CSeq=3 from 192.168.11.104:50548
08:04:36  <       majestic> [  client] RtspClient_before@403         RTSP DESCRIBE rtsp://192.168.11.101:554/stream=0 CSeq=4 from 192.168.11.104:50548
08:04:40  <       majestic> [  client] RtspClient_before@403         RTSP OPTIONS rtsp://192.168.11.101:554/stream=0 CSeq=5 from 192.168.11.104:50548
08:04:40  <       majestic> [  client] RtspClient_before@403         RTSP OPTIONS rtsp://192.168.11.101:554/stream=0 CSeq=6 from 192.168.11.104:50548
08:04:40  <       majestic> [  client] RtspClient_before@403         RTSP DESCRIBE rtsp://192.168.11.101:554/stream=0 CSeq=7 from 192.168.11.104:50548
08:04:40  <       majestic> [  client] RtspClient_before@403         RTSP SETUP rtsp://192.168.11.101:554/stream=0/video CSeq=8 from 192.168.11.104:50548
08:04:40  <       majestic> [  client] RtspClient_setup@314          UDP RTSP SETUP done, stream_id=0 from 192.168.11.104:50548
08:04:40  <       majestic> [  client] RtspClient_before@403         RTSP PLAY rtsp://192.168.11.101:554/stream=0 CSeq=9 from 192.168.11.104:50548
08:04:51  <       majestic> [  client] RtspClient_before@403         RTSP TEARDOWN rtsp://192.168.11.101:554/stream=0 CSeq=10 from 192.168.11.104:50548
08:04:51  <       majestic> [  client] RtspClient_on_closed_conn@654 RTSP connection closed.
```

**kernel logs :**

```
[1;33m[MI WRN] _MI_SYS_Pass_DebugDumpInputTaskFence[344]: [0:c0aedf80 00000000] fence1 pulseindex2
[0m
[1;33m[MI WRN] _MI_SYS_Pass_DebugDumpInputTaskFence[344]: [1:c1890280 00000000] fence2 pulseindex4
[0m
[1;33m[MI WRN] MI_SYS_Pass_InternalMainWorkerThread[4139]: [scl0_P0_MAIN] NON-SelfDrive module mod34 pass0 input buffer has not been processed for more than 2 seconds!
[0m
[1;33m[MI WRN] _MI_SYS_Pass_TryDequeueInputTaskNoLock[2391]:  [thread:isp0_P0_MAIN] mod[33] dev[0] pass[0] chn[0] inputtask's fence is not finished more than 252830ms
[0m
[1;33m[MI WRN] _MI_SYS_Pass_DebugDumpInputTaskFence[344]: [0:c0aedd00 00000000] fence1 pulseindex-1
[0m
[1;33m[MI WRN] _MI_SYS_Pass_DebugDumpInputTaskFence[344]: [1:c1890300 00000000] fence2 pulseindex-1
[0m
[1;31m[MI ERR] MI_SYS_InferGraph_EnsureInputPortFifoEmpty[315]: Next input(mod33, dev0, pass0, chn0, port0, reject_cmdq0, fifo_cnt:1 2 0) no response in 1000ms!
[0m
[1;33m[MI WRN] _MI_SYS_Pass_TryDequeueInputTaskNoLock[2391]:  [thread:isp0_P0_MAIN] mod[33] dev[0] pass[0] chn[0] inputtask's fence is not finished more than 254860ms
[0m
[... 72 lines cut ...]
    - name: kernel
      size: 0x200000
      sha1: 70486b29
    - name: rootfs
      size: 0x500000
      sha1: 5e7cbd3e
    - name: rootfs_data
      size: 0x8b0000
      path: /overlay,jffs2,rw
  size: 16M
ram:
  total: 128M
  media: 70M
firmware:
  u-boot: "2015.01 (Oct 27 2024 - 16:59:55)"
  kernel: "5.10.61 (PREEMPT Sun Nov 24 18:13:20 UTC 2024)"
  toolchain: buildroot-gcc-13.3.0
sensors:
- vendor: Sony
  model: IMX335
  control:
    bus: 0
    type: i2c
    addr: 0x34
```

carcinogoy, 2024-12-12:
I pulled the drivers from one of my original anjvision cameras via an exploit
[image]

```
majestic[655]: Majestic Lite for SigmaStar, version master+f25b567, built on 2024-12-05
majestic[655]: Detected infinity Watchdog
majestic[655]: Watchdog timeout set to 300 seconds
majestic[655]: Set server path: /var/www
majestic[655]: HTTP server started on 0.0.0.0:80
majestic[655]: Loading main configuration: /etc/majestic.yaml
majestic[655]: Initializing Sigmastar SDK, mi_sys_version_3.011
majestic[655]: Sensor index 0: 2592x1944@25fps
majestic[655]: [MI_ISP_CUS3A_Enable] AE = 1, AWB = 1, AF = 1
majestic[655]: [MI_ISP_CUS3A_Enable] AE = 1, AWB = 1, AF = 1
majestic[655]: [MI_ISP_CUS3A_Enable] AE = 1, AWB = 1, AF = 1
majestic[655]: Channel 0: 2592x1944@10fps
majestic[655]: - [H265, 4096Kb, RC1, GOP50]
majestic[655]: - [minQp: 12, maxQp: 48, qpDelta: -4, slice: 0]
majestic[655]: Channel 2: 2592x1944@5fps
majestic[655]: - [JPEG, Q50, 4900Kb]
majestic[655]: Load isp file /etc/sensors/imx335.bin
majestic[655]: [MAJOR]: ispversion(0) in sdk, ispversion(1) in binfile.
majestic[655]: [MINOR]: ispversion(-964716594) in sdk, ispversion(2) in binfile.
majestic[655]: ooooops,SDK & iqbinfile major version is not match.
majestic[655]: warning warning!!!SDK & iqbinfile minor version is not match.
majestic[655]: ******************** Load api bin Fail - ERROR CODE (0xffffffff) ********************
majestic[655]: Error loading isp file
majestic[655]: Cannot set color to gray
majestic[655]: Error setting isp parameters
majestic[655]: RTSP server started on port 554
```
To get the drivers you can either reverse one of these with binwalk http://www.anjvision.com:8021/firmware/online/public/MCL12D_V0-A-H5/
and check the `config\modules\5.10` folder

Or if you have another camera with the origianl firmware you can exfiltrate the files via
` wget http://CAMERAIP/playback/lib/modules/5.10.61/5.10/mi_common.ko -O mi_common.ko`

Fun fact anjvision camera root password are 
`ANJVISION*SERIAL NUMBER*  --> MD5 --> UPPERCASE --> last 8 chars`
example
`ANJVISIONEF00000000A12D61 --> 44af9a4ceecbad0c7be0a53972fed43a --> 44AF9A4CEECBAD0C7BE0A53972FED43A --> 72FED43A`

A good reason to switch to OpenIPC :)

flyrouter, 2024-12-14:
> A good reason to switch to OpenIPC :)

Thanks for the interesting information and research
Write to me in Telegram messenger on the nickname @FlyRouter and we will discuss testing your device.

tctlrd, 2025-01-24:
I am also getting no video with oIPC. It seems there is no ultimate version available for this SoC? I was able to get root on MD-L12D's firmware by replacing the passwd link with a new passwd file.

jawadbrayer, 2025-03-17:
@tctlrd how did you do it? Did you generate a new firmware image, or was it something else?

tctlrd, 2025-03-18:
@jawadbrayer 
I extracted, modified, re-compressed the filesystem from the firmware image.
I did a dd to replace the filesystem bytes in the firmware image.
Attached is the result. It is not a .txt file, it is a .bin, you can rename it.

[mod.txt](https://github.com/user-attachments/files/19327274/mod.txt)

tctlrd, 2025-04-04:
@jawadbrayer you can reach me at p[@]tactile.red

tctlrd, 2025-11-09:
@carcinogoy what is your process to install the extracted drivers into openipc? My camera is a PTZ branded by Jennov A76WM55-5X-EA.
I am wondering the correct process to implement openipc on this with PTZ onvif support.

widgetii, 2026-11-10:
I checked this against the current firmware on an SSC377QE + IMX335 board. The `ssc377d` and `ssc377qe` lite images ship the same drivers, libraries and IMX335 IQ file. With the current nightly, the sensor comes up at 2592x1944, the IQ file loads (`ispversion(1) in sdk, ispversion(1) in binfile` … `Load api bin Success`) and video works.

Two things have changed since this was reported:
- #1679 replaced the whole infinity6c driver and library set.
- #2343 now loads the IMX335 driver with `lane_num=4`, the same thing Anjvision's stock firmware does on the MCL12D (`imx335_MIPI.ko chmap=1 lane_num=4`).

`ispversion(0) in sdk` with a garbage minor number means the ISP never came up. That's also what you get when the kernel modules and the userspace libraries come from different SDKs, for example after copying the stock `.ko` files over OpenIPC's.

Could you sysupgrade to the current `ssc377d` lite nightly with the stock OpenIPC modules (no swapped `.ko`) and check again? If it still fails, please attach the output of `logread`, `dmesg | grep -iE "isp|sensor|vif|csi"` and `fw_printenv sensor sensor_lanes`.

@byfaruk, the OS04D10 / SSC377DE driver question is a different problem. Please open a separate issue for it so it gets tracked on its own.