OpenIPC/firmware #1617: Ingenic T31 no ethernet

daoudeddy, 2024-11-25:
I've been trying to run OpenIPC on a tuya camera but the thing is im stuck with no ethernet on OpenIPC u-boot nor inside linux
while if i revert it to stock firmware i get my ethernet working 

openipc bootlogs:
```
U-Boot SPL 2013.07-g90873ad (Oct 27 2024 - 16:57:48)
Timer init
CLK stop
PLL init
pll_init:366
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf=118 nr = 1 od0 = 1 od1 = 2
cppcr is 07605100
CPM_CPAPCR 0750510d
nf=84 nr = 1 od0 = 1 od1 = 2
cppcr is 05405100
CPM_CPMPCR 07d0590d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a773310
apll_freq 1404000000 
mpll_freq 1000000000 
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 500000000
cclk  1404000000
l2clk 702000000
h0clk 200000000
h2clk 200000000
pclk  100000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00f73011
T31_0x5: 00000007
T31_0x15: 0000000c
T31_0x4: 00000000
T31_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T31_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T31_118: 0000003c
T31_158: 0000003c
T31_190: 00000020
T31_194: 0000001f
jz-04 :  0x00000051 
jz-08 :  0x000000a0 
jz-28 :  0x00000024 
DDR PHY init OK
INNO_DQ_WIDTH   :00000003
INNO_PLL_FBDIV  :00000014
INNO_PLL_PDIV   :00000005
INNO_MEM_CFG    :00000051
INNO_PLL_CTRL   :00000018
INNO_CHANNEL_EN :0000000d
INNO_CWL        :00000006
INNO_CL         :00000007
DDR Controller init
DDRC_STATUS         0x80000001
DDRC_CFG            0x0a288a40
DDRC_CTRL           0x0000011c
DDRC_LMR            0x00400008
DDRC_DLP            0x00000000
DDRC_TIMING1        0x040e0806
DDRC_TIMING2        0x02170707
DDRC_TIMING3        0x2007051e
DDRC_TIMING4        0x1a240031
DDRC_TIMING5        0xff060405
DDRC_TIMING6        0x32170505
DDRC_REFCNT         0x00f26801
DDRC_MMAP0          0x000020fc
DDRC_MMAP1          0x00002400
DDRC_REMAP1         0x03020d0c
DDRC_REMAP2         0x07060504
DDRC_REMAP3         0x0b0a0908
DDRC_REMAP4         0x0f0e0100
DDRC_REMAP5         0x13121110
DDRC_AUTOSR_EN      0x00000000
sdram init finished
SDRAM init ok
board_init_r
image entry point: 0x80100000

U-Boot 2013.07-g90873ad (Oct 27 2024 - 16:57:48)

Board: ISVP (Ingenic XBurst T31 SoC)
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 479k for U-Boot at: 83f88000
Reserving 32832k for malloc() at: 81f78000
Reserving 32 Bytes for Board Info at: 81f77fe0
Reserving 124 Bytes for Global Data at: 81f77f64
Reserving 256k for boot params() at: 81f37f64
Stack Pointer at: 81f37f48
Now running in RAM - U-Boot at: 83f88000
MMC:   msc: 0
SF: Detected XM25QH64A
In:    serial
Out:   serial
Err:   serial
gpio_request[hichip_mmc_enable] = 62
Net:   ====>phy 0: 0x0-0x128 found
SPEED:1, DUPLEX:2
Jz4775-9161
reading boot.scr
** Unable to read file boot.scr **
```

stock bootlogs:
```
U-Boot SPL 2013.07 (Dec 23 2020 - 12:31:13)
Timer init
CLK stop
PLL init
pll_init:366
pll_cfg.pdiv = 10, pll_cfg.h2div = 5, pll_cfg.h0div = 5, pll_cfg.cdiv = 1, pll_cfg.l2div = 2
nf=84 nr = 1 od0 = 1 od1 = 2
cppcr is 05405100
CPM_CPAPCR 0540510d
nf=84 nr = 1 od0 = 1 od1 = 2
cppcr is 05405100
CPM_CPMPCR 07d0590d
nf=100 nr = 1 od0 = 1 od1 = 2
cppcr is 06405100
CPM_CPVPCR 0640510d
cppcr 0x9a773310
apll_freq 1008000000 
mpll_freq 1000000000 
vpll_freq = 1200000000
ddr sel mpll, cpu sel apll
ddrfreq 500000000
cclk  1008000000
l2clk 504000000
h0clk 200000000
h2clk 200000000
pclk  100000000
CLK init
SDRAM init
sdram init start
ddr_inno_phy_init ..!
phy reg = 0x00000007, CL = 0x00000007
ddr_inno_phy_init ..! 11:  00000004
ddr_inno_phy_init ..! 22:  00000006
ddr_inno_phy_init ..! 33:  00000006
REG_DDR_LMR: 00000210
REG_DDR_LMR: 00000310
REG_DDR_LMR: 00000110
REG_DDR_LMR, MR0: 00f73011
T31_0x5: 00000007
T31_0x15: 0000000c
T31_0x4: 00000000
T31_0x14: 00000002
INNO_TRAINING_CTRL 1: 00000000
INNO_TRAINING_CTRL 2: 000000a1
T31_cc: 00000003
INNO_TRAINING_CTRL 3: 000000a0
T31_118: 0000003c
T31_158: 0000003c
T31_190: 00000021
T31_194: 0000001f
jz-04 :  0x00000051 
jz-08 :  0x000000a0 
jz-28 :  0x00000024 
DDR PHY init OK
INNO_DQ_WIDTH   :00000003
INNO_PLL_FBDIV  :00000014
INNO_PLL_PDIV   :00000005
INNO_MEM_CFG    :00000051
INNO_PLL_CTRL   :00000018
INNO_CHANNEL_EN :0000000d
INNO_CWL        :00000006
INNO_CL         :00000007
DDR Controller init
DDRC_STATUS         0x80000001
DDRC_CFG            0x0a288a40
DDRC_CTRL           0x0000011c
DDRC_LMR            0x00400008
DDRC_DLP            0x00000000
DDRC_TIMING1        0x040e0806
DDRC_TIMING2        0x02170707
DDRC_TIMING3        0x2007051e
DDRC_TIMING4        0x1a240031
DDRC_TIMING5        0xff060405
DDRC_TIMING6        0x32170505
DDRC_REFCNT         0x00f25a01
DDRC_MMAP0          0x000020fc
DDRC_MMAP1          0x00002400
DDRC_REMAP1         0x03020d0c
DDRC_REMAP2         0x07060504
DDRC_REMAP3         0x0b0a0908
DDRC_REMAP4         0x0f0e0100
DDRC_REMAP5         0x13121110
DDRC_AUTOSR_EN      0x00000000
sdram init finished
SDRAM init ok
board_init_r
image entry point: 0x80100000

U-Boot 2013.07 (Dec 23 2020 - 12:31:13)

Board: ISVP (Ingenic XBurst T31 SoC)
DRAM:  64 MiB
Top of RAM usable for U-Boot at: 84000000
Reserving 442k for U-Boot at: 83f90000
Reserving 32784k for malloc() at: 81f8c000
Reserving 32 Bytes for Board Info at: 81f8bfe0
Reserving 124 Bytes for Global Data at: 81f8bf64
Reserving 128k for boot params() at: 81f6bf64
Stack Pointer at: 81f6bf48
Now running in RAM - U-Boot at: 83f90000
MMC:   msc: 0
the manufacturer ef
SF: Detected W25Q128

In:    serial
Out:   serial
Err:   serial
Net:   ====>phy 0:0x0-0x128 found
SPEED:2, DUPLEX:2
====>phy 0 Autonegotiation Complete
Jz4775-9161
```

pictures of the camera:

[image]
[image]

johankvdbp, 2025-01-23:
I have the exact same board. ethernet light on cable is lit indefinite. no lan or wifi. in openipc cli "ifconfig" has no rx. I think wifi and eth drivers are missing. how to fix I don't know sadly...

daoudeddy, 2025-01-23:
> I have the exact same board. ethernet light on cable is lit indefinite. no lan or wifi. in openipc cli "ifconfig" has no rx. I think wifi and eth drivers are missing. how to fix I don't know sadly...

@johankvdbp you can check the [Thingino](https://github.com/themactep/thingino-firmware) repo i already submitted my changes both ethernet and WiFi works well