OpenIPC/firmware #1631: T20L: sc2235: Majestic: Can't find config file under paths

felipecrs, 2024-12-04:
I updated my camera from an ancient OpenIPC version and now Majestic no longer starts:

```
majestic[641]: Majestic Lite for Ingenic, version master+51af9af, built on 2024-11-27
majestic[641]: Detected jz Watchdog
majestic[641]: Watchdog timeout set to 300 seconds
majestic[641]: Set server path: /var/www
majestic[641]: HTTP server started on :::80
majestic[641]: Loading main configuration: /etc/majestic.yaml
majestic[641]: SENSOR=sc2235
majestic[641]: Can't find config file under paths:
majestic[641]: 	
majestic[641]: 	/etc/sensors/
majestic[641]: 	/etc/sensor/
majestic[641]: Cannot start SDK
```

This was the "ancient" OpenIPC version:

[image]

felipecrs, 2024-12-21:
There's no need to add this extra binary if I'm the only one who needs it. I am experimenting with thingino instead.

HwlloChen, 2025-01-31:
Looks like I'm having the same problem as you.
＞﹏＜