OpenIPC/firmware #1654: sigmastar-osdrv-infinity6c missing sc430ai sensor module

mcd1992, 2024-12-31:
The TP-Link Tapo C120 uses a sc430ai sensor. Not sure where to get the updated sigmastar SDK, will attempt to use the ones extracted from factory firmware for now though.

mcd1992, 2025-04-11:
I did but was never able to load them correctly. I forget the details but the .ko shipped by tplink had all the modules linked into 1 so it complained about symbols. And unloading all the openipc ones and attempting to load the tplink one caused a kernel panic.

asyncbridge, 2025-04-15:
I tried to extract the sc430ai.ko and the sensor config file (extention is bin) from c120 rootfs and copy them to proper path as I expected and build the firmware and run flash writing but I got the segmentation error.

1. sc430ai sensor config file
squashfs-root/config/auto_detect_sensor/default_isp/sc430ai_MIPI/isp_default_day.bin
* I'm not sure the sensor config bin file is correct or not.

2. sc430ai ko file
squashfs-root/config/auto_detect_sensor/modules/sc430ai_MIPI.ko

** I tried to copy them to 

** Update 
sigmastar-osdrv-infinity6c/files/script/load_sigmastar to support sc430ai.

set_sensor() {
	case $SENSOR in
		gc4653|imx335|imx415|os04a10|sc4336p|sc401ai|**sc430ai**|sc501ai|sc830ai|sc850sl)
			insmod $MODULE/sensor_${SENSOR}_mipi.ko chmap=1
			;;
		*)
			echo -e "\n\e[1;31mUNSUPPORTED sensor - $SENSOR\e[0m\n" | logger -s -t OpenIPC
			;;
	esac
}

$(TARGET_DIR)/etc/sensors/sc430ai.bin
$(TARGET_DIR)/lib/modules/5.10.61/sigmastar/sensor_sc430ai_mipi.ko

----
OpenIPC: Sensor assigned - sc430ai
8<--- cut here ---
Unable to handle kernel NULL pointer dereference at virtual address 00000009
pgd = 7e6d0037
[00000009] *pgd=00000000
Internal error: Oops: 805 [#1] PREEMPT THUMB2
Modules linked in: drv_ms_cus_sc430ai_MIPI_tp_ww(PO+) mi(PO) 8188fu(O) cfg80211 exfat(O)
CPU: 0 PID: 646 Comm: insmod Tainted: P           O      5.10.61 #2
Hardware name: SStar Soc (Flattened Device Tree)
PC is at DrvSensorSPIAttach+0x2/0x28 [mi]
LR is at DrvRegisterSensorDriverEx+0xcf/0x150 [mi]
pc : [<bf96fc96>]    lr : [<bf96eecb>]    psr: 60000033
sp : c1891e70  ip : bfada764  fp : 00000000
r10: 00000000  r9 : c0605900  r8 : bfbe992d
r7 : bfada148  r6 : bfadb028  r5 : bfada148  r4 : 00000000
r3 : bf96fc71  r2 : 0003a980  r1 : 00000001  r0 : 00000000
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA Thumb  Segment none
Control: 50c5383d  Table: 218e8059  DAC: 00000051
Process insmod (pid: 646, stack limit = 0xf16c0e38)
Stack: (0xc1891e70 to 0xc1892000)
1e60:                                     00000000 bfbec248 c0605900 00000000
1e80: bfbec240 bfbebbe1 bfbe992d bfbef05f bfbef001 ffffe000 00000000 c040b000
1ea0: c0008404 c1890000 00000000 c00097ab 00000000 00000000 00000000 00000000
1ec0: 00000000 00000000 6e72656b 00006c65 00000000 00000000 00000001 c0601e40
1ee0: 00000cc0 c1890000 0000774f c0074fe1 c1891efc 00000000 00000000 c87d034f
1f00: 00000000 bfbec040 b6feec20 c1904a00 0000017b c004af67 bfbec040 0040003e
1f20: 00000000 b6feec20 00000003 c004c6cf c1891f40 7fffffff 00000000 00000002
1f40: c4d54000 c4d562aa c4d563c0 c4d54000 0000448c c4d57f8c c4d57e4c c4d57074
1f60: 00004000 00004670 000030a0 00004b4b 00000000 00000000 00000000 00003090
1f80: 0000001d 0000001e 00000018 00000000 00000013 00000000 b6feec20 00000000
1fa0: 0000017b c0008201 b6feec20 00000000 00000003 b6feec20 00000000 befb6f2c
1fc0: b6feec20 00000000 0000017b 0000017b 8683fbf8 00000000 00000000 00000000
1fe0: befb6f2c befb6d70 00496a69 b6fb10b0 60000030 00000003 00000000 00000000
[<bf96fc96>] (DrvSensorSPIAttach [mi]) from [<bf96eecb>] (DrvRegisterSensorDriverEx+0xcf/0x150 [mi])
[<bf96eecb>] (DrvRegisterSensorDriverEx [mi]) from [<bfbef05f>] (init_module+0x5f/0x1000 [drv_ms_cus_sc430ai_MIPI_tp_ww])
[<bfbef05f>] (init_module [drv_ms_cus_sc430ai_MIPI_tp_ww]) from [<c00097ab>] (do_one_initcall+0x43/0x114)
[<c00097ab>] (do_one_initcall) from [<c004af67>] (do_init_module+0x37/0x13c)
[<c004af67>] (do_init_module) from [<c004c6cf>] (sys_finit_module+0x51/0x5a)
[<c004c6cf>] (sys_finit_module) from [<c0008201>] (ret_fast_syscall+0x1/0x64)
Exception stack(0xc1891fa8 to 0xc1891ff0)
1fa0:                   b6feec20 00000000 00000003 b6feec20 00000000 befb6f2c
1fc0: b6feec20 00000000 0000017b 0000017b 8683fbf8 00000000 00000000 00000000
1fe0: befb6f2c befb6d70 00496a69 b6fb10b0
Code: 2000 f000 bc24 4b05 (6088) 2000 
---[ end trace 1f471130def8537a ]---
Segmentation fault

miumil, 2025-04-16:
I tried to copy and overwrite the related .so files extracted from C120 firmware to $(TARGET_DIR)/usr/lib folder and no segmentation fault error but majestic isn't initialized successfully.

10:01:12  <       majestic> [    main] main@237                      Majestic Lite for SigmaStar, version master+2ca6c07, built on 2025-04-09
10:01:12  <       majestic> [watchdog] watchdog_start@94             Detected SSTAR Watchdog
10:01:12  <       majestic> [watchdog] watchdog_start@115            Watchdog timeout set to 10 seconds
10:01:12  <       majestic> [   httpd] start_web_server@371          Set server path: /var/www
10:01:12  <       majestic> [   httpd] new_http_server@359           HTTP server started on 0.0.0.0:80
10:01:12  <       majestic> [    main] init_sdk@66                   Loading main configuration: /etc/majestic.yaml
10:01:12  <       majestic> [     sdk] start_sdk@1534                Initializing Sigmastar SDK, mi_sys_version_3.011
10:01:12  <       majestic> [     sdk] mi_snr_get_resolution@505     Sensor index 0: 2688x1520@30fps
10:01:12  <       majestic> [     log] printf@292                    [isp] failed to ioctl 0xc02c6923!(I/O error)
10:01:14  <       majestic> [     sdk] mi_venc_create_chn@1001       Channel 0: 2688x1520@20fps
10:01:14  <       majestic> [     sdk] mi_venc_create_chn@1002       - [H264, 4096Kb, RC1, GOP20]
10:01:14  <       majestic> [     sdk] mi_venc_init@1099             Cannot create venc chn0
10:01:14  <       majestic> [     sdk] start_sdk@1572                Error initializing venc
10:01:14  <       majestic> [    main] init_sdk@73                   Cannot start SDK

asyncbridge, 2025-04-16:
I tried to copy and overwrite the related .so files extracted from C120 firmware to $(TARGET_DIR)/usr/lib folder and no segmentation fault error but majestic isn't initialized successfully.

10:01:12 < majestic> [ main] main@237 Majestic Lite for SigmaStar, version master+2ca6c07, built on 2025-04-09
10:01:12 < majestic> [watchdog] watchdog_start@94 Detected SSTAR Watchdog
10:01:12 < majestic> [watchdog] watchdog_start@115 Watchdog timeout set to 10 seconds
10:01:12 < majestic> [ httpd] start_web_server@371 Set server path: /var/www
10:01:12 < majestic> [ httpd] new_http_server@359 HTTP server started on 0.0.0.0:80
10:01:12 < majestic> [ main] init_sdk@66 Loading main configuration: /etc/majestic.yaml
10:01:12 < majestic> [ sdk] start_sdk@1534 Initializing Sigmastar SDK, mi_sys_version_3.011
10:01:12 < majestic> [ sdk] mi_snr_get_resolution@505 Sensor index 0: 2688x1520@30fps
10:01:12 < majestic> [ log] printf@292 [isp] failed to ioctl 0xc02c6923!(I/O error)
10:01:14 < majestic> [ sdk] mi_venc_create_chn@1001 Channel 0: 2688x1520@20fps
10:01:14 < majestic> [ sdk] mi_venc_create_chn@1002 - [H264, 4096Kb, RC1, GOP20]
10:01:14 < majestic> [ sdk] mi_venc_init@1099 Cannot create venc chn0
10:01:14 < majestic> [ sdk] start_sdk@1572 Error initializing venc
10:01:14 < majestic> [ main] init_sdk@73 Cannot start SDK

asyncbridge, 2025-04-17:
I tried to copy related .so files and mi.ko, esp32.ko from above path of extracted squashfs-root to target path and it worked without segmentation fault but I got sdk init error.

squashfs-root/config/venc_fw/chagall.bin -> firmware/general/package/sigmastar-osdrv-infinity6c/files/sensor/firmware
squashfs-root/config/iqfile/isp_api.xml -> firmware/general/package/sigmastar-osdrv-infinity6c/files/sensor/firmware
squashfs-root/lib/modules/5.10.61/mi.ko -> firmware/general/package/sigmastar-osdrv-infinity6c/files/kmod
squashfs-root/lib/modules/5.10.61/esp32.ko -> firmware/general/package/sigmastar-osdrv-infinity6c/files/kmod
squashfs-root/usr/lib/*.so -> firmware/general/package/sigmastar-osdrv-infinity6c/files/lib

------
Starting ntpd: OK
Starting dropbear: OK
Starting crond: OK
Loading vendor modules...
**OpenIPC: Sensor assigned - sc430ai**
Starting majestic: OK

Welcome to OpenIPC

------
majestic[650]: Majestic Lite for SigmaStar, version master+73701aa, built on 2025-04-16
majestic[650]: Detected SSTAR Watchdog
majestic[650]: Watchdog timeout set to 10 seconds
majestic[650]: Set server path: /var/www
majestic[650]: HTTP server started on 0.0.0.0:80
majestic[650]: Loading main configuration: /etc/majestic.yaml
majestic[650]: Initializing Sigmastar SDK, mi_sys_version_3.011
majestic[650]: Sensor index 0: 2688x1520@30fps
**majestic[650]: Error initializing snr
majestic[650]: Cannot start SDK**

NicksonYap, 2025-06-07:
@asyncbridge I think you're heading to the right direction, it has to be driver issue and it seems to be the Linux kernel version (or driver version) mismatch with the other drivers that are existing in OpenIPC for the SSC337 SoC

I have Tapo C120 (EU) v1.0 which should have the same sensor as yours
I'll try installing OpenIPC into it as well

Earlier I've successfully added the driver for SC3336 sensor into OpenIPC 
https://github.com/OpenIPC/firmware/pull/1825
but that was easy and worked immediately without any version mismatched

tdick-stackop, 2025-08-25:
@asyncbridge I am currently experimenting on the same board, slightly different SoC and another sensor, but I face the nearly the exact error message. Could you send me your .ko file please?  I like to compare a few values to mine.

asyncbridge, 2025-09-16:
> [@asyncbridge](https://github.com/asyncbridge) I am currently experimenting on the same board, slightly different SoC and another sensor, but I face the nearly the exact error message. Could you send me your .ko file please? I like to compare a few values to mine.