OpenIPC/firmware #1662: I bought JFTECH GK7205V300 with SONY IMX335 and can not connect to Serial port

tteksoy, 2025-01-13:
I bought JFTECH GK7205V300 with SONY IMX335 and can not connect to Serial port with IFFT adapter.

I was planning to get started with openIPC and dupplicate the recommendations from

https://www.youtube.com/watch?v=acLRpRaEIsM&list=PLiTPsaqtZy-9s6pXaYK5lQjaKCJJPIr4Q

Any recommendation wht I can not connect to serial port

thanks

lida2003, 2025-01-16:
> Any recommendation wht I can not connect to serial port

Did you check RX/TX wire sequence/UART1 or UART0?

Here is simple module first power-on steps, it's written in Chinese, but you can see those steps, or translate it use some web tools.

[Open FPV VTX开源之第一次出图](https://blog.csdn.net/lida2003/article/details/145072199)

tteksoy, 2025-01-17:
> > Any recommendation wht I can not connect to serial port
> 
> Did you check RX/TX wire sequence/UART1(default for MAVLink) or UART0(uboot)?
> 
> Here is simple module first power-on steps, it's written in Chinese, but you can see those steps, or translate it use some web tools.
> 
>     * [Open FPV VTX开源之第一次出图](https://blog.csdn.net/lida2003/article/details/145072199)
> 
>     * [Open FPV VTX开源之默认MAVLink设置](https://blog.csdn.net/lida2003/article/details/145080645)

Thank you very much, I have checked the wiring I will replace the FTDI adapter. but no response from serial port.

widgetii, 2026-10-04:
The wiki has a step-by-step for finding and connecting the UART: **https://github.com/OpenIPC/wiki/blob/master/en/installation.md#step-3-connect-to-uart-port-of-your-camera**, plus the quick receive-only test at https://github.com/OpenIPC/wiki/blob/master/en/serial_pins_uart.md#quick-test.

The usual reasons for a silent port:
- The adapter is set to 5 V instead of 3.3 V.
- TX and RX are not crossed (adapter TX → camera RX).
- GND is not shared.
- The terminal is not at 115200 8N1 with no flow control.

The quick test isolates it: connect only GND and the adapter's RX, power the camera on, and the bootloader log should appear. If nothing appears, you are on the wrong pad or the adapter is faulty. On the FPV modules, U-Boot's console is UART0; UART1 carries MAVLink.

Closing as answered. Please reopen with a photo of the wiring if the quick test still shows nothing.

closed 2026-10-04 by widgetii