OpenIPC/firmware #1674: Modifying u-boot and flashing firmware from inside the limited Linux shell - possible?

qumuq-til, 2025-01-21:
I've tried getting into a boot-password-protected HI3516EV300 device using the burn utility, without success. Then I've tried shortening the pins but it didn't affect u-boot behavior whatsoever. Then I was somehow able to gain a limited shell access using the same pins, so that it drops me into Linux shell after a failed boot. Is it possible to use this shell to reset u-boot password, flash a new generic boot partition or flash the firmware in its entirety? The openipc flashing manual seemingly deals with u-boot commands which I don't have here.

VictorDrijkoningen, 2025-07-11:
Could you list the available commands in this limited linux shell? if you can write to the spi flash, then it should be possible.

Regarding not being able to interrupt U-boot, I had this issue as well, it took me a few tries. In my case i needed to short the correct spi flash chip pins rather quickly after powering the board.

widgetii, 2026-08-16:
Both of your questions have straight answers, and neither of them requires that limited shell.

### "Does the OpenIPC firmware image contain the bootloader?"

**No.** Our release `.tgz` is kernel + rootfs only. From the packaging rule in the firmware tree ([`Makefile`](https://github.com/OpenIPC/firmware/blob/master/Makefile), `REPACK_FIRMWARE`), the archive contains exactly `uImage.<soc>`, `rootfs.squashfs.<soc>` and their `.md5sum` files — no U-Boot.

U-Boot is published as a separate asset. If you're going in with a hardware programmer, you want the **full image** instead, which stitches bootloader + kernel + rootfs into one flashable blob:

```
https://openipc.org/cameras/vendors/hisilicon/socs/hi3516ev300/download_full_image?flash_size=16&flash_type=nor&fw_release=lite
```

(Set `flash_size` to match your chip — 8 or 16.)

### "Can I flash from the limited Linux shell?"

Technically yes, if that shell has `/dev/mtd*` and `flashcp` or `dd`. **I'd strongly advise against it.** Overwriting mtd0 from a rootfs that is itself mounted from the same flash chip is about the most brick-prone operation available on these devices — the moment you touch a block the running system still needs, you're done, and there's no console to recover from.

If you want to explore it anyway, post the output of `cat /proc/mtd` and `busybox --list` and I'll tell you whether it's viable.

### What I'd actually do: skip U-Boot entirely

The reason the burn utility and the pin-shorting both failed you is that they're fighting the vendor bootloader. You don't have to.

Every Hi35xx carries a mask-ROM bootrom that listens on UART0 for a serial download handshake at power-on, *before* U-Boot runs. It's in silicon — the vendor can password-lock their bootloader, but they cannot disable or lock the bootrom. Upload our U-Boot into RAM over UART and you get an unlocked console on a device whose flash still holds the locked vendor bootloader. Nothing is written, so there's nothing to brick.

**https://openipc.github.io/defib/** does this in the browser — Chrome, Edge or Opera (WebSerial; Firefox and Safari won't work). No install, no Python, no drivers, no pin shorting, no password.

1. **Chip Model** → `hi3516ev300`
2. **U-Boot image** → *Download from OpenIPC* — fetched from our releases and SHA-256 verified
3. **Serial Port** → your USB-UART adapter
4. Tick **"Send Ctrl-C after upload (enter U-Boot console)"**
5. **Start Recovery** — when it prints `Waiting for bootrom... power-cycle the device now!`, cut power and reapply

Start the tool *first*, then power-cycle — it blasts the handshake continuously so it catches the bootrom window, which is what the timing games with shorted pins were trying to achieve.

Once you're at the prompt, hit **Dump Flash** to save the stock firmware before you write anything, then follow the TFTP install at https://openipc.org/cameras/vendors/hisilicon/socs/hi3516ev300

> If the U-Boot download step fails, GitHub is refusing the cross-origin fetch for release assets — the page prints the direct URL and expected SHA-256 so you can fetch it manually and select the file.

widgetii, 2026-08-16:
## Correction — the browser tool will not work for your chip

I need to walk back part of my previous comment. I checked defib's internals more carefully afterwards and got this wrong.

**hi3516ev300 needs a "frame-blast" handshake that the browser build does not implement.** Its profile carries a 64-byte `PRESTEP0` block that has to be sent *before* the DDR init step, and the handshake has to be an active `0xAA + HEAD` blast rather than passively waiting for the bootrom's `0x20` markers. The web build does neither — it waits passively and then sends DDR → SPL → U-Boot with the pre-step missing. So the sequence is incomplete for this SoC and the upload will not come up.

I missed it because the profile file for `hi3516ev300` is an alias — it contains the single line `hi3516ev200.json` rather than any data — so my first check for the pre-step read an empty result and I reported it as a plain-handshake chip. It isn't. Tracking the underlying defect as OpenIPC/defib#121.

### What to use instead

The **command-line** defib does implement the frame-blast path, and hi3516ev300 is hardware-verified on it:

```shell
uv tool install defib        # or: pipx install defib
defib burn -c hi3516ev300 -p /dev/ttyUSB0 -t
```

Start it first, then power-cycle the camera — same principle as before. `-t` drops you into the U-Boot console afterwards. On Windows, use the port name directly (`-p COM3`).

To back up the stock firmware before writing anything:

```shell
defib agent read -c hi3516ev300 -p /dev/ttyUSB0 -o flash_backup.bin
```

Everything else in my earlier comment stands — the bootrom genuinely does run ahead of the vendor bootloader and cannot be password-locked, so the method is right; it's only the browser front-end that can't drive your particular SoC yet.

Sorry for the detour.

closed 2026-09-26 by openipc-ai