OpenIPC/firmware #1706: NetIP requests password

J0hNnik, 2025-02-14:
NetIP says wrong password.
In majestic.yaml:

netip:
enabled: true
port: 34567
ignoreSetTime: false
snapshots: true
user: admin
password: ХХХХХ

hi3516ev200
openipc_2.5.02.12-lite

J0hNnik, 2025-03-01:
I'll answer my own question.
Maybe it will be useful to someone.
The password is Dahua hash.

https://github.com/haicenhacks/DahuaHashCreator

flyrouter, 2025-03-08:
Good day
Thank you for your interest in our project.
You need to set not a password but a hash.
For example, 6V0Y4HLF corresponds to the password 12345 and thus by setting it for verification you will be able to make a connection with the video recorder.

J0hNnik, 2025-03-08:
Why do I need 12345?
I want my password.

closed 2026-09-04 by openipc-ai