OpenIPC/firmware #1716: Error in general/overlay/etc/wireless/usb for HI3518EV200 Switcam HS303 V2 (maybe others)

tonyich, 2025-02-19:
On boot i got error:
```Starting network...
Setting GPIO-54 to HI
modprobe: module 8188eu not found in modules.dep
Successfully initialized wpa_supplicant
Could not read interface wlan0 flags: No such device
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Could not read interface wlan0 flags: No such device
```

Check module name:
```~# cat /lib/modules/4.9.37/modules.dep |grep 8188
kernel/drivers/staging/rtl8188eu/r8188eu.ko: kernel/net/wireless/cfg80211.ko
```

module name is **r**8188eu

but in script [/etc/wireless/usb](https://github.com/OpenIPC/firmware/blob/fcae5981fffa5c3f76edb5dcb17a0c5237e2e803/general/overlay/etc/wireless/usb#L152) module name is different: modprobe 8188eu.

cronyx, 2025-02-20:
Hello, it is not error, hs303v2 use external wifi driver. You can see all configs for this camera in the [builder repo.](https://github.com/OpenIPC/builder/blob/master/devices/hi3518ev200_lite_switcam-hs303-v2/general/overlay/usr/share/openipc/customizer.sh)

tonyich, 2025-02-20:
[cronyx](https://github.com/cronyx), In the current version of the script, the camera will be inoperative, which prevents from doing this:
```
# HI3518EV200 Switcam HS303 V2
if [ "$1" = "rtl8188eus-switcam-hs303v2" ]; then
	set_gpio 54 1
	modprobe r8188eu
	exit 0
fi
```

I tested this on my camera - it works.

closed 2025-02-25 by flyrouter

tonyich, 2025-03-26:
I don't understand. I updated the firmware via sysupgrade.
```
root@openipc-hi3518ev200:~# fw_printenv upgrade
upgrade=https://github.com/OpenIPC/builder/releases/download/latest/hi3518ev200_lite_switcam-hs303-v2-nor.tgz
```

and after reboot I see:
modprobe: module 8188eu not found in modules.dep

What am I doing wrong?

tonyich, 2025-04-03:
for those who have encountered the same problem. after April's update problem solved.
_sysupgrade -k -r -n_
and:
```
# cat /etc/*-release
NAME=Buildroot
VERSION=-g171d54c
ID=buildroot
VERSION_ID=2024.02.10
PRETTY_NAME="Buildroot 2024.02.10"
OPENIPC_VERSION=2.5.04.02
GITHUB_VERSION="master+c847c72, 2025-04-02"
BUILD_OPTION=lite
TIME_STAMP=1743564909
# cat /lib/modules/4.9.37/modules.dep |grep 8188
kernel/drivers/staging/rtl8188eu/r8188eu.ko: kernel/net/wireless/cfg80211.ko
extra/8188eu.ko: kernel/net/wireless/cfg80211.ko  #<<<<<Here it is!
```