OpenIPC/firmware #1725: Bad posterization SSC378DE + IMX415 (i2c)

qumuq-til, 2025-02-24:
See attached images, one from Majestic (obviously visible posterization) the other from VLC (less visible, but still present). Tried to tweak some settings with SS IQ Tool, changes are being applied, but none of the ones I have tried would deal with this posterization effect.

[image]
[image]

flyrouter, 2025-02-25:
Before opening a new Issue please review the list and try to supplement the existing ones. Thank you.

qumuq-til, 2025-02-25:
I did search for posterization and found nothing.

flyrouter, 2025-02-25:
> I did search for posterization and found nothing.

Unfortunately, you probably didn't search well enough.

https://github.com/OpenIPC/majestic/issues/213