OpenIPC/firmware #1729: Backup creation always fails with 500 Internal Server Error

qumuq-til, 2025-02-25:
/cgi-bin/ext-backuper.cgi?backup=create accessible from Firmware - Settings - Create Backup always fails.

flyrouter, 2025-02-25:
Good day
Thank you, we will look into this information

flyrouter, 2025-02-25:
Good day
You can request research and correction directly from the author of the solution - @del000

qumuq-til, 2025-02-25:
I don't know if I should (or am entitled to) request anything from anyone directly. It's clearly a bug, which I have reported, and if there's a person working on this part of the code here on GitHub, then you could mark it as a bug and assign that person. If you're fine with leaving it as it is, then who am I to argue.

That is, unless I misunderstand how development is handled for this project.

flyrouter, 2025-02-25:
> I don't know if I should (or am entitled to) request anything from anyone directly. It's clearly a bug, which I have reported, and if there's a person working on this part of the code here on [...]

The component you reported a bug for is maintained by a developer in another repository.
I gave you advice and a chance to pay direct attention to who created this applet so that you could find and submit a request in the right place yourself.