OpenIPC/firmware #1740: imp-control and ingenic-pwm no longer in the firmware?

qumuq-til, 2025-03-06:
This wiki page https://github.com/thegroove/openipc-wiki/blob/master/en/ingenic-research.md mentions the two utilities, imp-control and ingenic-pwm to control an Ingenic-based board, but those utilities don't seem to be present in the Ultimate T40XP firmware. Is that intentional? Any reasonable way to get them back? If those were deprecated, are there any replacement ways to control sensor parameters (IMX415 specifically) in OpenIPC? So basically an alternative for PQTool and IQTool?

qumuq-til, 2025-03-07:
Do I really need to compile everything, including the Ingenic SDK just to make a simple imp-control utility? Or can someone share the necessary files that I could put onto my device?

qumuq-til, 2025-03-07:
Majestic plugin functionality doesn't seem to be included in the firmware either. I don't know how to build those.

flyrouter, 2025-03-08:
Good afternoon
Thank you for your interest in our project
These utilities are not complete for OpenIPC and you can find them on the Thingino project that separated from us last year.

closed 2025-03-08 by flyrouter

qumuq-til, 2025-03-08:
Do I understand it correctly that the whole IMP subsystem is not compatible, libraries included, not only the utilities themselves? I.e. it's just as impossible at the moment to have this functionality via Majestic plugins.