OpenIPC/firmware #1767: Mavlink OSD is not available in the latest firmware, but old version firmware works well

Li-Tianming, 2025-03-21:
The air unit can get data from serial, but the ground station(phone) can not display osd data.
```
root@openipc-ssc338q:~#  mavfwd --baudrate 115200 --channels 8 -p 100 -a 15 -t -
-master /dev/ttyS3 --in 0.0.0.0:14550 --out 10.5.0.1:14551
Monitoring RC channel 8
Aggregate mavlink pckts in packs of 15
Listening on /dev/ttyS3...
Listening on 0.0.0.0:14550...
Found SigmaStart temp sensor
Packets:3  Bytes:26
System_id = 1
Flight Controller Type :PX4

```

The Old version firmware works well.

Li-Tianming, 2025-03-21:
```
 mavfwd --baudrate 115200 --channels 8 -p 100 -a 15 -t --
master /dev/ttyS3 --in 127.0.0.1:14550 --out 127.0.0.1:14551
Monitoring RC channel 8
Aggregate mavlink pckts in packs of 15
Listening on /dev/ttyS3...
Listening on 127.0.0.1:14550...
Found SigmaStart temp sensor
Packets:3  Bytes:6
System_id = 1
Flight Controller Type :PX4

```

Tried old mavfwd ip but still not work.

Li-Tianming, 2025-03-21:
```
wfb-tun   Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.5.0.10  P-t-P:10.5.0.10  Mask:255.255.255.0
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1443  Metric:1
          RX packets:271 errors:0 dropped:4 overruns:0 frame:0
          TX packets:1219 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes:19684 (19.2 KiB)  TX bytes:224897 (219.6 KiB)

```

suiaiyun, 2025-03-21:
这个问题tg问过，并没有人回答

Li-Tianming, 2025-03-26:
I just found solution of using Mavfwd OSD in lasted firmware.

According to this [answer ](https://github.com/OpenIPC/firmware/issues/1762#issuecomment-2741500007) from tipoman9.

`telemetry_tx` and `telemetry_rx` just copy and rename of `wfb_tx` and `wfb_rx`.

In the old firmware, this is how they stared.

```
root@openipc-ssc338q:~# ps -eaf | grep 738
  738 root     wfb_tx -p 0 -u 5600 -R 456000 -K /etc/drone.key -B 20 -M 1 -S 0 -L 0 -G long -k 8 -n 12 -T 0 -i 7669206 -f data wlan0
  859 root     grep 738
root@openipc-ssc338q:~# ps -eaf | grep 746
  746 root     telemetry_tx -p 16 -u 14550 -K /etc/drone.key -B 20 -M 1 -S 1 -L 1 -G long -k 1 -n 2 -T 0 -i 7669206 -f data wlan0
  839 root     grep 746
root@openipc-ssc338q:~# ps -eaf | grep 745
  745 root     telemetry_rx -p 144 -u 14551 -K /etc/drone.key -i 7669206 wlan0
  921 root     grep 745
```

In the new firmware, this is how they stared.
```
root@openipc-ssc338q:~# ps -eaf | grep 771
  771 root      0:25 wfb_tx -K /etc/drone.key -M 2 -B 20 -k 8 -n 12 -S 0 -L 0 -i 7669206 -C 8000 -G short wlan0
  894 root      0:00 grep 771
root@openipc-ssc338q:~#  ps -eaf | grep 774
  774 root      0:00 wfb_rx -K /etc/drone.key -i 7669206 -p 160 -u 5800 wlan0
  905 root      0:00 grep 774
root@openipc-ssc338q:~#  ps -eaf | grep 775
  775 root      0:00 wfb_tx -K /etc/drone.key -M 1 -B 20 -k 8 -n 12 -S 0 -L 0 -i 7669206 -p 32 -u 5801 wlan0
  917 root      0:00 grep 775
root@openipc-ssc338q:~# ps -eaf | grep 776
  776 root      0:00 wfb_tun -a 10.5.0.10/24
  936 root      0:00 grep 776
```

So, if you want to use mavfwd. you should 

```
ps 
882 root      0:00 wfb_tun -a 10.5.0.10/24
910 root      0:00 msposd -b 115200 -c 8 -r 20
```

```
kill 882 
kill 910

# start wfb_tx for 14550 port
wfb_tx -p 16 -u 14550 -K /etc/drone.key -B 20 -M 1 -S 1
-L 1 -G long -k 1 -n 2 -T 0 -i 7669206 -f data wlan0 > /dev/null &

# start wfb_rx for 14551 port
wfb_rx -p 144 -u 14551 -K /etc/drone.key -i 7669206 wlan
0  > /dev/null &

# start mavfwd
mavfwd --channels 8 --master /dev/ttyS3 --baudrate 11520
0 -p 100 -t -a 15 --out 127.0.0.1:14550 --in 127.0.0.1:14551

```
This gonna carsh the lasted Pixlepilot app in your phone. so you need downgrade the pixlepilot

Li-Tianming, 2025-03-26:
https://github.com/OpenIPC/PixelPilot/releases/download/v0.1.2/PixelPilot.apk

suiaiyun, 2025-03-26:
> I just found solution of using Mavfwd OSD in lasted firmware.
> 
> According to this [answer ](https://github.com/OpenIPC/firmware/issues/1762#issuecomment-2741500007) from tipoman9.
> 
> `telemetry_tx` and just copy and rename of and .`telemetry_rx``wfb_tx``wfb_rx`
> 
> In the old firmware, this is how they stared.
> 
> ```
> root@openipc-ssc338q:~# ps -eaf | grep 738
>   738 root     wfb_tx -p 0 -u 5600 -R 456000 -K /etc/drone.key -B 20 -M 1 -S 0 -L 0 -G long -k 8 -n 12 -T 0 -i 7669206 -f data wlan0
>   859 root     grep 738
> root@openipc-ssc338q:~# ps -eaf | grep 746
>   746 root     telemetry_tx -p 16 -u 14550 -K /etc/drone.key -B 20 -M 1 -S 1 -L 1 -G long -k 1 -n 2 -T 0 -i 7669206 -f data wlan0
>   839 root     grep 746
> root@openipc-ssc338q:~# ps -eaf | grep 745
>   745 root     telemetry_rx -p 144 -u 14551 -K /etc/drone.key -i 7669206 wlan0
>   921 root     grep 745
> ```
> 
> In the new firmware, this is how they stared.
> 
> ```
> root@openipc-ssc338q:~# ps -eaf | grep 771
>   771 root      0:25 wfb_tx -K /etc/drone.key -M 2 -B 20 -k 8 -n 12 -S 0 -L 0 -i 7669206 -C 8000 -G short wlan0
>   894 root      0:00 grep 771
> root@openipc-ssc338q:~#  ps -eaf | grep 774
>   774 root      0:00 wfb_rx -K /etc/drone.key -i 7669206 -p 160 -u 5800 wlan0
>   905 root      0:00 grep 774
> root@openipc-ssc338q:~#  ps -eaf | grep 775
>   775 root      0:00 wfb_tx -K /etc/drone.key -M 1 -B 20 -k 8 -n 12 -S 0 -L 0 -i 7669206 -p 32 -u 5801 wlan0
>   917 root      0:00 grep 775
> root@openipc-ssc338q:~# ps -eaf | grep 776
>   776 root      0:00 wfb_tun -a 10.5.0.10/24
>   936 root      0:00 grep 776
> ```
> 
> So, if you want to use mavfwd. you should
> 
> ```
> ps 
> 882 root      0:00 wfb_tun -a 10.5.0.10/24
> 910 root      0:00 msposd -b 115200 -c 8 -r 20
> ```
> 
> ```
> kill 882 
> kill 910
> 
> # start wfb_tx for 14550 port
> wfb_tx -p 16 -u 14550 -K /etc/drone.key -B 20 -M 1 -S 1
> -L 1 -G long -k 1 -n 2 -T 0 -i 7669206 -f data wlan0 > /dev/null &
> 
> # start wfb_rx for 14551 port
> wfb_rx -p 144 -u 14551 -K /etc/drone.key -i 7669206 wlan
> 0  > /dev/null &
> 
> # start mavfwd
> mavfwd --channels 8 --master /dev/ttyS3 --baudrate 11520
> 0 -p 100 -t -a 15 --out 127.0.0.1:14550 --in 127.0.0.1:14551
> ```
> 
> This gonna carsh the lasted Pixlepilot app in your phone. so you need downgrade the pixlepilot

Thank you, this is very helpful to me. But one thing I still don't understand is that the new firmware uses tunneling technology, but it doesn't work for Mavfwd. Is this a bug in the new firmware?

closed 2026-09-26 by openipc-ai