OpenIPC/firmware #1769: GK7205V500 firmware request

MiwaHiro, 2025-03-23:
Greetings,

I bought 5 cameras which have GK7205V500 CPU.
I saw only GK7205V200, GK7205V210 and GK7205V300 are supported in the OPENIPC website (https://openipc.org/cameras/vendors/goke).

I am so hyped by this project that I tried to search everywhere for the GK7205V500 openipc firmware.
After looking for previous issues, I found this merge request for the GK7205V500 (https://github.com/OpenIPC/firmware/pull/1708).

Then I found this binary in the latest releases (https://github.com/OpenIPC/firmware/releases/download/latest/openipc.gk7205v500-nor-lite.tgz)
Inside, there is rootfs.squashfs.gk7205v500 and uImage.gk7205v500 file.
I tried to play with it with my camera and succeeded to boot with /bin/sh, but the credentials openipc/12345 doesn't work when I try to use the command login.
I used this command to boot into shell.
setenv ipaddr xxx.xxx.xxx.xxx; setenv serverip xxx.xxx.xxx.xxx; sf probe 0;setenv bootargs init=/bin/sh mem=38M console=ttyAMA0,115200 root=/dev/mtdblock1 rootfstype=squashfs mtdparts=sfc:0x40000(boot),0x540000(romfs),0x740000(usr),0x180000(web),0x80000(custom),0x140000(mtd);tftp 0x43000000 rootfs.squashfs.gk7205v500;tftp 0x42000000 uImage.gk7205v500;squashfsload;bootm 0x42000000

Therefore I would have some stupid questions...
Is it normal that the default login isn't working when I try to login from shell (having loaded squashfs and uImage on RAM)?
Do I need OPENIPC uboot in order to make this image work or should it work with the original uboot from the camera?

Sorry to ask that but is GK7205V500 in the pipes? 
If needed I can test dev releases with my cameras :)

Thank you in advance, your project is AWESOME!

michalderkacz, 2026-10-16:
I also have a Tptek 5 Mpx camera, unfortunately bootloop affected after some random power outages.

Its serial console is completely silent so I managed to desolder the XM25QH128D flash chip and dumped its content and add a telnetd command to the root squashfs so I was able to login to the camera this way.

The bootloop is caused by the `XmServices_Mgr /usr/sbin/AppRun.sh /mnt/mtd/Config` command. What I understand from its logs there is a problem to communicate with the watchdog and this probably causes reboot after few seconds.

I reinitialized the jffs2 filesystem with the `/sbin/flash_eraseall -j -q /dev/mtd5` command but without any positive effect.

Here is wat the App says about this camera:

ID(Encr):  0x0234(IPC_G5S-WQ)
product type:  "IPC_GK7205V500_G5S-WQ"
DSP chip:  GK7205V500(1)

The onboard WiFi chip is ATBM6062U.

I want to try the OpenIPC on this camera. I've generated the Installation Guide and then downloaded and flashed the `openipc-gk7205v500-nor-ultimate-16mb.bin` image directly to the flash. Unfortunately after soldering the flash back to the board and powering on the camera there is no any activity on the UART and no answer to the `ping 192.168.1.10`. I reverted to the backup image and camera is alive. Flashed the OpenIPC image again, unfortunately with the same result.

Is the generated image the right one for this camera and the way I flash it?

michalderkacz, 2026-10-16:
The camera has ethernet port. I logged to the dumped and patched firmware (telnetd added) this way. But wit the openipc-gk7205v500-nor-ultimate-16mb.bin there is no any console output/reaction, no any Ethernet activity (checked with tcpdump, etherned linked ok, expected see DHCP requests),

closed 2026-11-16 by openipc-ai