OpenIPC/firmware #1772: the FPV function has been cancelled?

njgymb, 2025-03-25:
Why is there no FPV download option for ssc338q on the official website now, only "lite" and "unlimated", does it mean the FPV function has been cancelled?

Li-Tianming, 2025-03-25:
```
https://openipc.org/cameras/vendors/sigmastar/socs/ssc338q/download_full_image?flash_size=16&flash_type=nor&fw_release=fpv
```
You can still download it through url. Hope it's just a display bug.

And RubyFpv also disappeared.

wildblue, 2025-04-03:
majestic source code fails to download. therefore no functione...
/src/openipc/openipc-builder/openipc/general/package/majestic/majestic.mk
MAJESTIC_SITE = https://openipc.s3-eu-west-1.amazonaws.com
MAJESTIC_SOURCE = majestic.$(MAJESTIC_FAMILY).$(MAJESTIC_VARIANT).master.tar.bz2
MAJESTIC_LICENSE = PROPRIETARY
MAJESTIC_LICENSE_FILES = LICENSE

By the way, don`t these variables ever get correct values and where is the mentioned license file?

widgetii, 2026-08-16:
No, FPV was not cancelled. It moved repositories, and the website catalogue simply does not cover it.

FPV images are built and published from https://github.com/OpenIPC/builder, in the `nightly` release. For SSC338Q:

- https://github.com/OpenIPC/builder/releases/download/nightly/openipc.ssc338q-nor-fpv.tgz
- https://github.com/OpenIPC/builder/releases/download/nightly/openipc.ssc338q-nand-fpv.tgz
- https://github.com/OpenIPC/builder/releases/download/nightly/openipc.ssc338q-nor-apfpv.tgz

There are also per-device profiles built from the same place if one matches your hardware — `ssc338q_fpv_caddx-fly`, `ssc338q_fpv_emax-wyvern-link`, `ssc338q_fpv_openipc-mario-aio`, `ssc338q_fpv_openipc-thinker-aio`, `ssc338q_fpv_runcam-wifilink`, `ssc338q_rubyfpv_generic`, and Greg AP-FPV variants.

That is why openipc.org only offers Lite and Ultimate: this repository builds the general-purpose IP camera variants, and the FPV/VTX side lives in the builder repo with its own device profiles and its own release cadence. FPV development is very much alive — see `OpenIPC/adaptive-link`, `OpenIPC/msposd`, `OpenIPC/PixelPilot`, `OpenIPC/aviateur`, `OpenIPC/configurator`.

@wildblue — on your separate point about `majestic.mk`: `MAJESTIC_FAMILY` and `MAJESTIC_VARIANT` are set by the build, not left empty, so the tarball name resolves at build time. If the download fails for you it is a network or a stale-`dl/` problem rather than an unset variable. Majestic is closed-source, which is why the recipe pulls a prebuilt binary; if you hit a concrete download error, please open a separate issue with the failing URL and we will look at it.

Closing — the download you were looking for exists at the links above.

closed 2026-08-16 by widgetii