OpenIPC/firmware #1778: Hi3516CV200 himci error

hiverods, 2025-03-30:
Device HiSilicon HI3516CV200 with 8MB NOR flash
When kernel started on step 
`random: fast init done`

it waiting a long time.
`himci_control_cclk:255: disable or enable clk is timeout!`

How to disable this mmc driver for fast boot?

 ```
ohci-platform 100a0000.ohci: irq 32, io mem 0x100a0000
hub 2-0:1.0: USB hub found
hub 2-0:1.0: 1 port detected
i2c /dev entries driver
usbcore: registered new interface driver i2c-tiny-usb
hisi-i2c-hisilicon 200d0000.i2c: Hisilicon [i2c-0] probed!
hisi-i2c-hisilicon 20240000.i2c: Hisilicon [i2c-1] probed!
hisi-i2c-hisilicon 20250000.i2c: Hisilicon [i2c-2] probed!
himci: mmc host probe
himci: mmc host probe
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
hisi-dmac 10060000.hidma-controller: hidmac probe!
NET: Registered protocol family 10
NET: Registered protocol family 17
random: fast init done
himci_wait_cmd:219: send cmd is timeout!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_set_cclk:297: set card clk divider is failed!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_set_cclk:297: set card clk divider is failed!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_set_cclk:297: set card clk divider is failed!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
devtmpfs: mounted
Freeing unused kernel memory: 160K (c043e000 - c0466000)
This architecture does not have kernel memory protection.
himci_wait_cmd:219: send cmd is timeout!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_set_cclk:297: set card clk divider is failed!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_set_cclk:297: set card clk divider is failed!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_set_cclk:297: set card clk divider is failed!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

^A^Chimci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:219: send cmd is timeout!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_set_cclk:297: set card clk divider is failed!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

random: fast init done
himci_wait_cmd:219: send cmd is timeout!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_set_cclk:297: set card clk divider is failed!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_set_cclk:297: set card clk divider is failed!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_set_cclk:297: set card clk divider is failed!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

himci_wait_cmd:208: Other CMD is running,please operate cmd again!

himci_control_cclk:255: disable or enable clk is timeout!

VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
devtmpfs: mounted
Freeing unused kernel memory: 160K (c043e000 - c0466000)
This architecture does not have kernel memory protection.
himci_wait_cmd:219: send cmd is timeout!

himci_control_cclk:255: disablerandom: crng init done
 or enable clk is timeout!

himci_wait_cmd:208: Other CMD is runttyAMA ttyAMA0: 1 input overrun(s)
ning,plee cmd again!

himci_set_cclk:297: set card clk divider is failed!

himcining,please operate cmd again!
```

pianist, 2025-06-29:
Same for me, 3518ev200

pianist, 2025-06-29:
https://github.com/OpenIPC/firmware/issues/1832

closed, fix by env!

closed 2026-11-16 by openipc-ai