OpenIPC/firmware #1792: Support for GK7102S

mrlulu51, 2025-04-10:
Hello,

I recently bought chinese cameras in the retail store and I wanted to know if the support of the GK7102S is planned, as it is the chip inside those cameras.

If you are in research for testers, I can be here to test the firmware.

Thank you for advise.

csev1755, 2025-06-15:
You can check hardware support here: https://openipc.org/supported-hardware/featured

Right now we have no streamer written and driver binaries for only the gc1034 and sc1135. The best way to accelerate support would likely be providing a firmware dump or donating a sample camera to a developer.

mrlulu51, 2025-06-19:
Oh yeah I can provide you a firmware dump of a Avidsen camera 123981 tomorrow if it could help.
How can I send the file to your team ?

closed 2026-11-09 by openipc-ai