OpenIPC/firmware #1808: Unable to access SD card on SSC378DE

overplex, 2025-05-01:
<details>
  <summary>Works in U-Boot</summary>

```shell
OpenIPC # mmc dev 1
switch to partitions #0, OK
mmc1 is current device

OpenIPC # mmc info
Device: MStar SD/MMC
Manufacturer ID: 2
OEM: 544d
Name: SA16G
Tran Speed: 50000000
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 14.9 GiB
Bus Width: 4-bit

OpenIPC # fatinfo mmc 1
Interface:  MMC
  Device 1: Vendor: Man 000002 Snr b446c400 Rev: 2.1 Prod: SA16G
            Type: Removable Hard Disk
            Capacity: 15256.0 MB = 14.8 GB (31244288 x 512)
Filesystem: FAT32 "NO NAME    "

OpenIPC # fatls mmc 1
            system volume information/
            test dir 1/
            test dir 2/
        3   test file.txt

1 file(s), 3 dir(s)
```
</details>

In the firmware, as I understand, the device /dev/mmcblk0 should appear, but it does not. Also, when loading, there are errors when loading [sdmmc].
The memory card is soldered manually, 3.3 V power is connected externally, because the power from the board drops almost to 0 on pin when connecting the microsd.

<details>
  <summary>dmesg</summary>

```shell
Booting Linux on physical CPU 0x0
Linux version 5.10.61 (runner@fv-az1335-445) (buildroot-gcc-13.3.0) #2 PREEMPT Wed Apr 30 22:49:07 UTC 2025
CPU: ARMv7 Processor [411fd040] revision 0 (ARMv7), cr=50c5383d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
early_atags_to_fdt() success
OF: fdt: Machine model: INFINITY6C SSC027D-S01A
LXmem is 0x8000000 PHYS_OFFSET is 0x20000000
Add mem start 0x20000000 size 0x8000000!!!!

LX_MEM  = 0x20000000, 0x8000000
LX_MEM2 = 0x0, 0x0
LX_MEM3 = 0x0, 0x0
MMU_MEM = 0x0, 0x0
EMAC_LEN= 0x0
DRAM_LEN= 0x0
Memory policy: Data cache writeback
Reserved memory: bypass cma0 node, using cmdline CMA params instead
OF: reserved mem: node cma0 compatible matching fail
no any mmap reserved
deal_with_reserve_mma_heap memblock_reserve success mma_config[0].reserved_start=
0x23a00000 size:4600000

cma: Reserved 2 MiB at 0x23800000
Zone ranges:
  Normal   [mem 0x0000000020000000-0x0000000027ffffff]
[... 439 lines cut ...]
      sha1: 5f635d02
    - name: rootfs
      size: 0x500000
      sha1: dd050705
    - name: rootfs_data
      size: 0xb0000
      path: /overlay,jffs2,rw
  size: 8M
ram:
  total: 128M
  media: 70M
firmware:
  u-boot: "2015.01 (Oct 27 2024 - 17:00:16)"
  kernel: "5.10.61 (PREEMPT Wed Apr 30 22:49:07 UTC 2025)"
  toolchain: buildroot-gcc-13.3.0
  main-app: majestic
sensors:
- vendor: Sony
  model: IMX415
  control:
    bus: 0
    type: i2c
    addr: 0x34
```
</details>

VictorDrijkoningen, 2025-07-11:
What do you mean with "soldered manually"? Could you give more details about the board and where it is soldered to?

Have you checked with another memory card? I think you might have a shorted one as a 3.3v power line should definitely not drop to 0v when "just" an sd card is connected

overplex, 2025-08-25:
It seems the problem cannot be solved.

- I probably burned something when I mixed up the mirror wires at the very beginning, which is why the 3.3 V power supply is not reaching the card. I powered the memory card from DC-DC 5->3.3.
- I tested 4 MicroSD cards of different brands and capacities, all of which work in U-Boot.
- None of the MicroSD cards work in OpenIPC.
    > [sdmmc] ms_sdmmc_probe
    Fail to get pad(131331) form padmux !
    > [sdmmc_0] Probe Platform Devices
- None of the MicroSD cards work in the original firmware.

[image]
[image]
[image]
[image]