OpenIPC/firmware #1816: GK7205V210  + mis2008  artifacts and wrong colors

vobukay, 2025-05-13:
### Ip camera

Cheap ip camera from aliexpress. Original firmware version is V5.00.R02.000659SQ. Sensor has to be SC5239s (SC223a ) based on [Please add GK7205V210+SC5239s platform](https://github.com/OpenIPC/coupler/issues/32)

### OpenIPC

Processor - gk7205v210
Sensor - mis2008
Firmware Version - 2.5.05.12-lite
Build - master+134387d, 2025-05-12
Majestic - master+9f0308f, 2025-05-11

<details><summary>Output image</summary>
<p>
Image has artifacts like a grid  and wrong colors.

- Using lib libsns_mis2008.so from Sensor Library has no effects.
- Changing value of Isp_Bayer in INI file changes color to other side but not normal.

</p>

 [image]
</details> 

<details><summary>Hardware investigation</summary>
<p>
</p>

```shell
root@openipc-gk7205v200:/tmp# ipctool 
The ipctool installed as remote GitHub plugin
Error: unexpected value for SOI == 0xe00
---
chip:
  vendor: Goke
  model: 7205V210
  id: 23c2281881485de3
board:
  vendor: OpenIPC
  version: 2.5.05.12
ethernet:
  mac: "2e:ba:14:41:4f:be"
  u-mdio-phyaddr: 1
  phy-id: 0x20669906
  d-mdio-phyaddr: 0
rom:
- type: nor
  block: 64K
  partitions:
    - name: boot
      size: 0x40000
      sha1: 59b871ea
    - name: env
      size: 0x10000
      sha1: 13aa8cc5
      contains:
        - name: uboot-env
          offset: 0x0
    - name: kernel
      size: 0x200000
      sha1: 46da2b47
    - name: rootfs
      size: 0x500000
      sha1: d30c65ad
    - name: rootfs_data
      size: 0xb0000
      path: /overlay,jffs2,rw
  size: 8M
  addr-mode: 3-byte
ram:
  total: 64M
  media: 32M
firmware:
  kernel: "4.9.37 (Mon May 12 22:59:47 UTC 2025)"
  toolchain: buildroot-gcc-8.4.0
  main-app: majestic
sensors:
- vendor: ImageDesign
  model: MIS2008
  control:
    bus: 0
    type: i2c
    addr: 0x60
  data:
    type: MIPI
    input-data-type: DATA_TYPE_RAW_10BIT
    lane-id:
      - 0
      - 1
    image: 1920x1080
  clock: 27MHz
``` 
```shell
root@openipc-gk7205v200:/tmp# ipctool i2cdump 0x60 0x3000 0x3002
       0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F
3000: 20 08 01 
root@openipc-gk7205v200:/tmp# 
``` 
mis2008 datasheet  and sensor have mismatched value of 0x3001 register   0x9 and  0x8
[image]

</details>

vobukay, 2025-05-13:
some progress 

1. edit mis2008_i2c_1080p.ini  Isp_Bayer = BAYER_RGGB
2. ipctool i2cset 0x60  0x320b 0x8b         (initial 0x88)

[image]

vobukay, 2025-05-17:
recompiled version that is worked on my camera. 
! Mirroring or flipping will change Isp_Bayer
[libsns_mis2008.zip](https://github.com/user-attachments/files/20264642/libsns_mis2008.zip)

boli, 2025-07-28:
I am also having a very similar problem

I had the 'grid' artefacts , but the ipctool command made a big improvement to the image

>ipctool i2cset 0x60 0x320b 0x8b

I'm not sure what daytime images look like - I have only viewed it at night so far.