OpenIPC/firmware #1817: hi3518ev100 build rtl8188eus driver failed

raulmadrid777, 2025-05-13:
I [forked] the repository, modified the config file, and then used GitHub Actions to compile it, but an error occurred during compilation.
[https://github.com/raulmadrid777/openipc-firmware/blob/master/br-ext-chip-hisilicon/configs/hi3518ev100_lite_defconfig](url)

err：
[36211](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36212)
                 from ../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/include/osdep_service.h:50,
[36212](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36213)
                 from ../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/include/drv_types.h:26,
[36213](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36214)
                 from ../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:17:
[36214](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36215)
include/net/cfg80211.h:782:12: note: ‘ssid’ declared here
[36215](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36216)
  782 |         u8 ssid[IEEE80211_MAX_SSID_LEN];
[36216](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36217)
      |            ^~~~
[36217](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36218)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_issue_p2p_provision_request’:
[36218](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36219)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:6054:17: warning: taking address of packed member of ‘struct rtw_ieee80211_hdr’ may result in an unaligned pointer value [-Waddress-of-packed-member]
[36219](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36220)
 6054 |         fctrl = &(pwlanhdr->frame_ctl);
[36220](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36221)
      |                 ^~~~~~~~~~~~~~~~~~~~~~
[36221](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36222)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c: At top level:
[36222](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36223)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8921:48: warning: ‘struct ieee80211_sta_vht_cap’ declared inside parameter list will not be visible outside of this definition or declaration
[36223](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36224)
 8921 | static void rtw_cfg80211_create_vht_cap(struct ieee80211_sta_vht_cap *vht_cap)
[36224](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36225)
      |                                                ^~~~~~~~~~~~~~~~~~~~~
[36225](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36226)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_create_vht_cap’:
[36226](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36227)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8926:16: error: invalid use of undefined type ‘struct ieee80211_sta_vht_cap’
[36227](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36228)
 8926 |         vht_cap->vht_supported = 1;
[36228](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36229)
      |                ^~
[36229](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36230)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8927:16: error: invalid use of undefined type ‘struct ieee80211_sta_vht_cap’
[36230](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36231)
 8927 |         vht_cap->cap = IEEE80211_VHT_CAP_RXLDPC;
[36231](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36232)
      |                ^~
[36232](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36233)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8927:24: error: ‘IEEE80211_VHT_CAP_RXLDPC’ undeclared (first use in this function); did you mean ‘IEEE80211_HT_CAP_RX_STBC’?
[36233](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36234)
 8927 |         vht_cap->cap = IEEE80211_VHT_CAP_RXLDPC;
[36234](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36235)
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
[36235](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36236)
      |                        IEEE80211_HT_CAP_RX_STBC
[36236](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36237)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8927:24: note: each undeclared identifier is reported only once for each function it appears in
[36237](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36238)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8931:28: error: ‘IEEE80211_VHT_MCS_SUPPORT_0_9’ undeclared (first use in this function)
[36238](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36239)
 8931 |                 mcs_map |= IEEE80211_VHT_MCS_SUPPORT_0_9 << (i*2);
[36239](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36240)
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[36240](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36241)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8934:16: error: invalid use of undefined type ‘struct ieee80211_sta_vht_cap’
[36241](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36242)
 8934 |         vht_cap->vht_mcs.rx_mcs_map = cpu_to_le16(mcs_map);
[36242](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36243)
      |                ^~
[36243](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36244)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8935:16: error: invalid use of undefined type ‘struct ieee80211_sta_vht_cap’
[... 3 lines cut ...]
      |                ^~
[36246](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36247)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_init_wiphy’:
[36247](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36248)
../rtl8188eus-openipc-6146193406b62e942d13d4d43580ed94ac70c218/os_dep/linux/ioctl_cfg80211.c:8964:60: error: ‘struct ieee80211_supported_band’ has no member named ‘vht_cap’; did you mean ‘ht_cap’?
[36248](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36249)
 8964 |                         rtw_cfg80211_create_vht_cap(&band->vht_cap);
[36249](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36250)
      |                                                            ^~~~~~~
[... cut ...]
[36255](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36256)
make[1]: Leaving directory '/home/runner/work/openipc-firmware/openipc-firmware/output/buildroot-2024.02.10'
[36256](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36257)
make: *** [Makefile:26: build] Error 2
[36257](https://github.com/raulmadrid777/openipc-firmware/actions/runs/14991768745/job/42116569293#step:6:36258)
Error: Process completed with exit code 2.

raulmadrid777, 2025-05-13:
my config ：
# Architecture
BR2_arm=y

# Toolchain
BR2_TOOLCHAIN_EXTERNAL=y
BR2_TOOLCHAIN_EXTERNAL_CUSTOM=y
BR2_TOOLCHAIN_EXTERNAL_DOWNLOAD=y
BR2_TOOLCHAIN_EXTERNAL_URL="https://github.com/openipc/firmware/releases/download/$(OPENIPC_TOOLCHAIN).tgz"
BR2_TOOLCHAIN_EXTERNAL_CUSTOM_PREFIX="arm-openipc-linux-musleabi"
BR2_TOOLCHAIN_EXTERNAL_HEADERS_3_0=y
BR2_TOOLCHAIN_EXTERNAL_CUSTOM_MUSL=y
BR2_TOOLCHAIN_EXTERNAL_CXX=y

# Kernel
BR2_LINUX_KERNEL=y
BR2_LINUX_KERNEL_CUSTOM_TARBALL=y
BR2_LINUX_KERNEL_CUSTOM_TARBALL_LOCATION="https://github.com/openipc/linux/archive/$(OPENIPC_KERNEL).tar.gz"
BR2_LINUX_KERNEL_USE_CUSTOM_CONFIG=y
BR2_LINUX_KERNEL_CUSTOM_CONFIG_FILE="$(EXTERNAL_VENDOR)/board/$(OPENIPC_SOC_FAMILY)/hi3518ev100.generic.config"
BR2_LINUX_KERNEL_UIMAGE=y
BR2_LINUX_KERNEL_XZ=y

# Filesystem
BR2_PACKAGE_BUSYBOX_CONFIG="$(BR2_EXTERNAL)/package/busybox/busybox.config"
BR2_PACKAGE_UBOOT_TOOLS=y
BR2_PACKAGE_ZLIB=y
# BR2_PACKAGE_IFUPDOWN_SCRIPTS is not set
BR2_PACKAGE_MOSQUITTO=y
# BR2_PACKAGE_MOSQUITTO_BROKER is not set
BR2_PACKAGE_WIRELESS_TOOLS=y
BR2_PACKAGE_WPA_SUPPLICANT=y
BR2_PACKAGE_WPA_SUPPLICANT_CLI=y
BR2_PACKAGE_WPA_SUPPLICANT_PASSPHRASE=y
BR2_TARGET_ROOTFS_CPIO=y
BR2_TARGET_ROOTFS_SQUASHFS=y
BR2_TARGET_ROOTFS_SQUASHFS4_XZ=y

# OpenIPC
BR2_OPENIPC_SOC_VENDOR="hisilicon"
BR2_OPENIPC_SOC_MODEL="hi3518ev100"
BR2_OPENIPC_SOC_FAMILY="hi3516cv100"
BR2_OPENIPC_VARIANT="lite"
BR2_OPENIPC_FLASH_SIZE="8"

# Packages
BR2_PACKAGE_DROPBEAR_OPENIPC=y
BR2_PACKAGE_HISILICON_OSDRV_HI3516CV100=y
BR2_PACKAGE_IPCTOOL=y
BR2_PACKAGE_JSONFILTER=y
BR2_PACKAGE_LIBCURL_OPENIPC=y
BR2_PACKAGE_LIBCURL_OPENIPC_CURL=y
# BR2_PACKAGE_LIBCURL_OPENIPC_PROXY_SUPPORT is not set
# BR2_PACKAGE_LIBCURL_OPENIPC_COOKIES_SUPPORT is not set
# BR2_PACKAGE_LIBCURL_OPENIPC_EXTRA_PROTOCOLS_FEATURES is not set
BR2_PACKAGE_LIBEVENT_OPENIPC=y
BR2_PACKAGE_LIBOGG_OPENIPC=y
BR2_PACKAGE_LINUX_FIRMWARE_OPENIPC=y
#BR2_PACKAGE_LINUX_FIRMWARE_OPENIPC_MEDIATEK_MT7601U=y
BR2_PACKAGE_LINUX_FIRMWARE_OPENIPC_RTL_8188EUS=y
BR2_PACKAGE_MAJESTIC_FONTS=y
BR2_PACKAGE_MAJESTIC_WEBUI=y
BR2_PACKAGE_MAJESTIC=y
BR2_PACKAGE_MBEDTLS_OPENIPC=y
BR2_PACKAGE_MOTORS=y
#BR2_PACKAGE_MT7601U_OPENIPC=y
#BR2_PACKAGE_RTL8188ETV_OPENIPC=y
BR2_PACKAGE_RTL8188EUS_OPENIPC=y
BR2_PACKAGE_OPUS_OPENIPC=y
BR2_PACKAGE_VTUND_OPENIPC=y
BR2_PACKAGE_YAML_CLI=y

zerog2k, 2025-07-09:
due to extremely old kernel (3.x) on hi3518ev100 (which is hi3516cv100 arch), lots of things wont build on them.
This workaround _works_ for me, but overall I have stability (crashing) issues with majestic hi3518ev100/ov9712 combo.

```
BR2_PACKAGE_LINUX_FIRMWARE_OPENIPC=y
BR2_PACKAGE_LINUX_FIRMWARE_OPENIPC_MEDIATEK_MT7601U=n
BR2_PACKAGE_LINUX_FIRMWARE_OPENIPC_RTL_8188EU=n
BR2_PACKAGE_RTL8188EU=y
```

closed 2026-10-16 by openipc-ai