OpenIPC/firmware #1827: Password protect U-boot help

49handyman, 2025-06-06:
I'll assume no one has the u-boot password for the CamHiPro cameras? Model: 350c5mp PTZ Camera. HI3516cv300

It repeadetly requests u-boot password:

`
### Please input uboot password: ###
****************************************************
### Please input uboot password: ###
*************************************
### Please input uboot password: ###
`

See below for the rest of the boot text.

I've worked on reversing many XM model many years ago. Even compiled firmware back in 2015 but have forgot much of what i learned back then. Many bricked & trashed cameras later. 

Can anyone tell me for sure if i can burn your U-boot  or some generic binary directy to a chip and reinstall, so I can tftp the proper firmware on to this camera for testing. 

if so, exactly what location does it get burned to. I have all the equipment and extra chips. But, I'd like to bypass all the hours of testing if on of the developers already has the knowledge quickly at hand.

As i recall, i had problems burning files directly to chip. they were designed to be written by tftp or in proper locations or formats. I remember the compiler builds a few types of u-boot binaries and I dont remember which was needed.

I could pull a u-boot off another chipset but I dont have anything close to this on. U-boot might get loaded and running even if the Linux kernel doesnt.

If some can let me know if they have or can tell me how to create a generic boot loader and where on the chip to burn it. I asume u-boot would be at 0x000000

Appreciate any help.
Doug

`
System startup

Uncompress Ok!

U-Boot 2016.11 (Apr 01 2019 - 18:57:48 +0800)hi3516ev300

Relocation Offset is: 0775e000
Relocating to 47f5e000, new gd at 47f1def0, sp at 47f1ded0
SPI Nor:  Check Flash Memory Controller v100 ... Found
SPI Nor ID Table Version 1.0
SPI Nor(cs 0) ID: 0x20 0x70 0x18
Block:64KB Chip:16MB Name:"XM25QH128A"
SPI Nor total size: 16MB
NAND:  0 MiB
In:    serial
Out:   serial
Err:   serial
Net:   eth0
Hit any key to stop autoboot:  0
device 0 offset 0x40000, size 0x1a0000

SF: 1703936 bytes @ 0x40000 Read: OK
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   Linux-4.9.37
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1664694 Bytes = 1.6 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Loading Kernel Image ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.9.37 (root@localhost.localdomain) (gcc version 6.3.0 (HC&C V100R002C00B032_20190114) ) #5 Sat Dec 7 01:22:26 HKT 2019
CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c53c7d
CPU: div instructions available: patching division code
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt:Machine model: Hisilicon HI3516EV300 DEMO Board
Memory policy: Data cache writeback
CPU: All CPU(s) started in SVC mode.
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16764
Kernel command line: mem=66M console=ttyAMA0,115200 root=/dev/mtdblock2 rootfstype=jffs2 rw mtdparts=hi_sfc:256K(boot),1664K(kernel),14464K(rootfs)
PID hash table entries: 512 (order: -1, 2048 bytes)
Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
Memory: 62380K/67584K available (3238K kernel code, 118K rwdata, 784K rodata, 140K init, 136K bss, 5204K reserved, 0K cma-reserved)
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xc4800000 - 0xff800000   ( 944 MB)
    lowmem  : 0xc0000000 - 0xc4200000   (  66 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0331ca0   (3240 kB)
      .init : 0xc03f7000 - 0xc041a000   ( 140 kB)
      .data : 0xc041a000 - 0xc0437820   ( 119 kB)
       .bss : 0xc0439000 - 0xc045b344   ( 137 kB)
SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
NR_IRQS:16 nr_irqs:16 16
Gic dist init...
arm_arch_timer: Architected cp15 timer(s) running at 50.00MHz (phys).
clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb8812736b, max_idle_ns: 440795202655 ns
sched_clock: 56 bits at 50MHz, resolution 20ns, wraps every 4398046511100ns
Switching to timer-based delay loop, resolution 20ns
clocksource: arm,sp804: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 637086815595 ns
Console: colour dummy device 80x30
Calibrating delay loop (skipped), value calculated using timer frequency.. 100.00 BogoMIPS (lpj=500000)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0x40008200 - 0x40008258
VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
futex hash table entries: 256 (order: -1, 3072 bytes)
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
Serial: AMBA PL011 UART driver
12040000.uart: ttyAMA0 at MMIO 0x12040000 (irq = 21, base_baud = 0) is a PL011 rev2
console [ttyAMA0] enabled
12041000.uart: ttyAMA1 at MMIO 0x12041000 (irq = 22, base_baud = 0) is a PL011 rev2
12042000.uart: ttyAMA2 at MMIO 0x12042000 (irq = 23, base_baud = 0) is a PL011 rev2
ssp-pl022 12070000.spi: ARM PL022 driver, device ID: 0x00800022
ssp-pl022 12070000.spi: mapped registers from 0x12070000 to c4873000
ssp-pl022 12071000.spi: ARM PL022 driver, device ID: 0x00800022
ssp-pl022 12071000.spi: mapped registers from 0x12071000 to c4877000
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
clocksource: Switched to clocksource arch_sys_counter
NET: Registered protocol family 2
TCP established hash table entries: 1024 (order: 0, 4096 bytes)
TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
TCP: Hash tables configured (established 1024 bind 1024)
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
workingset: timestamp_bits=30 max_order=14 bucket_order=0
squashfs: version 4.0 (2009/01/31) Phillip Lougher
jffs2: version 2.2 (NAND) (ZLIB) (RTIME) (c) 2001-2006 Red Hat, Inc.
io scheduler noop registered
io scheduler deadline registered (default)
io scheduler cfq registered
pl061_gpio 120b0000.gpio_chip: PL061 GPIO chip @0x120b0000 registered
pl061_gpio 120b1000.gpio_chip: PL061 GPIO chip @0x120b1000 registered
pl061_gpio 120b2000.gpio_chip: PL061 GPIO chip @0x120b2000 registered
pl061_gpio 120b3000.gpio_chip: PL061 GPIO chip @0x120b3000 registered
pl061_gpio 120b4000.gpio_chip: PL061 GPIO chip @0x120b4000 registered
pl061_gpio 120b5000.gpio_chip: PL061 GPIO chip @0x120b5000 registered
pl061_gpio 120b6000.gpio_chip: PL061 GPIO chip @0x120b6000 registered
pl061_gpio 120b7000.gpio_chip: PL061 GPIO chip @0x120b7000 registered
pl061_gpio 120b8000.gpio_chip: PL061 GPIO chip @0x120b8000 registered
pl061_gpio 120b9000.gpio_chip: PL061 GPIO chip @0x120b9000 registered
hisi-sfc hisi_spi_nor.0: SPI Nor ID Table Version 1.2
hisi-sfc hisi_spi_nor.0: The ID: 0x20 isn't in the BP table, Current device can't not protect
hisi-sfc hisi_spi_nor.0: xm25qh128a (Chipsize 16 Mbytes, Blocksize 64KiB)
3 cmdlinepart partitions found on MTD device hi_sfc
3 cmdlinepart partitions found on MTD device hi_sfc
Creating 3 MTD partitions on "hi_sfc":
0x000000000000-0x000000040000 : "boot"
0x000000040000-0x0000001e0000 : "kernel"
0x0000001e0000-0x000001000000 : "rootfs"
`

csev1755, 2025-06-15:
You can find some more information regarding locked bootloaders here: https://github.com/OpenIPC/wiki/blob/master/en/help-uboot.md#bypassing-password-protected-bootloader

Otherwise, if you can flash the chip directly, we do have a full image including the bootloader that might work for you: https://openipc.org/cameras/vendors/hisilicon/socs/hi3516cv300/download_full_image?flash_size=16&flash_type=nor&fw_release=ultimate

Make sure you have a flash backup if something goes wrong.

widgetii, 2026-08-16:
Doug — you don't need the password, and you don't need to pull the chip. Let's clear up a few things first.

### Your chip is a hi3516ev300

Your own boot log settles it:

```
U-Boot 2016.11 (Apr 01 2019 - 18:57:48 +0800)hi3516ev300
OF: fdt:Machine model: Hisilicon HI3516EV300 DEMO Board
```

Not cv300. And `HI2105CHIP` isn't a chip name — it's one of the stock bootloader *passwords* people try on Xiongmai-derived firmware (see #1494). Your flash is an `XM25QH128A`, 16 MB NOR.

### The password is irrelevant — go under U-Boot, not through it

Every Hi35xx has a mask-ROM bootrom that listens on UART0 for a serial download handshake at power-on, *before* U-Boot gets control. It's in silicon. The vendor can password-lock their U-Boot all day; they cannot touch the bootrom.

So you upload our U-Boot straight into RAM over the same UART you're already connected to, and you get an unlocked console on a camera whose flash still holds the locked vendor bootloader. Nothing is written. Nothing is at risk. That's also why the `Ctrl-C` spam and the "no delay" you're seeing don't matter — you're not trying to interrupt their U-Boot at all.

### Do it in your browser, right now

**https://openipc.github.io/defib/** — Chrome, Edge or Opera (it uses the WebSerial API, so Firefox and Safari won't work). No install, no Python, no drivers.

1. **Chip Model** → `hi3516ev300`
2. **U-Boot image** → *Download from OpenIPC* — pulled from our releases and SHA-256 verified
3. **Serial Port** → your USB-UART adapter
4. Tick **"Send Ctrl-C after upload (enter U-Boot console)"**
5. **Start Recovery** — when it prints `Waiting for bootrom... power-cycle the device now!`, cut power to the camera and reapply

You land in a `hisilicon #` prompt in the browser.

> If the U-Boot download step fails, GitHub is refusing the cross-origin fetch for release assets. The page prints the direct URL and the expected SHA-256 — grab the file by hand and select it.

### Back up first — it's one button now

Once the console is up, hit **Dump Flash**. It reads the whole 16 MB and drops `flash_hi3516ev300_<timestamp>.bin` in your Downloads. It's slow, because it comes out over `md.b`, but it's the backup everyone tells you to make and nobody has had an easy way to make. Do it before you erase anything.

### Your actual questions

**"Can I burn U-Boot directly to the chip, and where?"** — Yes: offset `0x0`. Your own mtdparts confirm the layout — `hi_sfc:256K(boot),1664K(kernel),14464K(rootfs)` — so U-Boot occupies `0x0`–`0x40000`.

**But don't flash only our U-Boot.** OpenIPC's partition layout is not your vendor's, so our bootloader sitting on top of their kernel and rootfs will not boot. Two clean options:

- **Programmer route** (you have the hardware and spare chips, so this is the safest for you): write a complete image — bootloader, kernel and rootfs already stitched together:
  `https://openipc.org/cameras/vendors/hisilicon/socs/hi3516ev300/download_full_image?flash_size=16&flash_type=nor&fw_release=lite`
- **UART route**: the browser tool above, then TFTP the kernel and rootfs from the guide at https://openipc.org/cameras/vendors/hisilicon/socs/hi3516ev300

**"Can I pull U-Boot off another chipset?"** — No. The bootloader is SoC-specific; DDR init alone will differ. Use ours for ev300.

**`upgrade(sd): check start...`** — that's the vendor's SD-card auto-upgrade hook. It only accepts the vendor's own signed package format, so it's not a route into OpenIPC. Ignore it.

Given you've got a programmer and spare chips, I'd dump the original, then write the full image and keep the dump. Shout if the bootrom handshake won't catch.

widgetii, 2026-08-16:
## Correction — the browser tool will not work for your chip

I need to walk back part of my previous comment. I checked defib's internals more carefully afterwards and got this wrong.

**hi3516ev300 needs a "frame-blast" handshake that the browser build does not implement.** Its profile carries a 64-byte `PRESTEP0` block that has to be sent *before* the DDR init step, and the handshake has to be an active `0xAA + HEAD` blast rather than passively waiting for the bootrom's `0x20` markers. The web build does neither — it waits passively and then sends DDR → SPL → U-Boot with the pre-step missing. So the sequence is incomplete for this SoC and the upload will not come up.

I missed it because the profile file for `hi3516ev300` is an alias — it contains the single line `hi3516ev200.json` rather than any data — so my first check for the pre-step read an empty result and I reported it as a plain-handshake chip. It isn't. Tracking the underlying defect as OpenIPC/defib#121.

### What to use instead

The **command-line** defib does implement the frame-blast path, and hi3516ev300 is hardware-verified on it:

```shell
uv tool install defib        # or: pipx install defib
defib burn -c hi3516ev300 -p /dev/ttyUSB0 -t
```

Start it first, then power-cycle the camera — same principle as before. `-t` drops you into the U-Boot console afterwards. On Windows, use the port name directly (`-p COM3`).

To back up the stock firmware before writing anything:

```shell
defib agent read -c hi3516ev300 -p /dev/ttyUSB0 -o flash_backup.bin
```

Everything else in my earlier comment stands — the bootrom genuinely does run ahead of the vendor bootloader and cannot be password-locked, so the method is right; it's only the browser front-end that can't drive your particular SoC yet.

Sorry for the detour.

closed 2026-09-26 by openipc-ai