OpenIPC/firmware #1875: gk7205v200 killed  ffmpeg??????

kangz543g, 2025-09-02:
./ffmpeg -rtsp_transport tcp -i rtsp://root:12345@192.168.219.105/stream=0 \
        -t 5 -s 640x360 -vcodec copy /mnt/sdcard/test.mp4

ffmpeg version 4.4.2 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 8.4.0 (Buildroot -g4783f48)
  configuration: --enable-cross-compile --cross-prefix=/home/kkk/firmware/output/per-package/ffmpeg-openipc/host/bin/arm-openipc-linux-musleabi- --sysroot=/home/kkk/firmware/output/per-package/ffmpeg-openipc/host/arm-buildroot-linux-musleabi/sysroot --host-cc='/home/kkk/firmware/output/per-package/ffmpeg-openipc/host/bin/ccache /usr/bin/gcc' --arch=arm --target-os=linux --pkg-config=/home/kkk/firmware/output/per-package/ffmpeg-openipc/host/bin/pkg-config --prefix=/usr --disable-everything --disable-x86asm --disable-w32threads --disable-os2threads --disable-alsa --disable-appkit --disable-avfoundation --disable-bzlib --disable-coreimage --disable-iconv --disable-libxcb --disable-libxcb-shm --disable-libxcb-xfixes --disable-libxcb-shape --disable-lzma --disable-asm --disable-sndio --disable-sdl2 --disable-xlib --disable-zlib --disable-amf --disable-audiotoolbox --disable-cuda --disable-cuvid --disable-d3d11va --disable-dxva2 --disable-nvdec --disable-nvenc --disable-v4l2-m2m --disable-vaapi --disable-vdpau --disable-videotoolbox --disable-avdevice --disable-swscale --disable-postproc --disable-doc --disable-runtime-cpudetect --disable-bsfs --disable-iconv --disable-ffprobe --enable-gpl --enable-version3 --enable-pthreads --disable-swresample --disable-avdevice --disable-filters --disable-encoders --disable-decoders --enable-decoder='h264,hevc' --disable-muxers --enable-muxer='flv,rtsp' --disable-demuxers --enable-demuxer='h264,rtsp' --disable-parsers --enable-parser='h264,hevc' --disable-protocols --enable-protocol='file,rtmp,tcp' --disable-programs --enable-ffmpeg --enable-small --cpu=cortex-a7
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavfilter     7.110.100 /  7.110.100
Killed

kangz543g, 2025-09-03:
send ffmjpeg to telegram

widgetii, 2026-08-16:
**`Killed` is the kernel OOM killer, not an ffmpeg bug.**

A bare `Killed` right after the banner, with no error, is the shell reporting SIGKILL. On these boards that's the out-of-memory killer.

The cause is the memory split. gk7205v200 has 64 MB of DDR, but Linux doesn't get it — most goes to the MMZ pool the video pipeline allocates from. The default for this SoC is `osmem=32M` of `totalmem=64M` (`general/package/goke-osdrv-gk7205v200/files/script/set_allocator`), so userspace gets ~27 MB, and there is no swap.

Measured on a gk7205v200 here, which is even bumped to `osmem=40M`:

```
# cat /proc/cmdline
mem=40M console=ttyAMA0,115200 ... mmz_allocator=gk
# free
              total        used        free      shared  buff/cache   available
Mem:          35232       10724       10128         128       14380       21396
Swap:             0           0           0
```

~21 MB available with majestic running. ffmpeg's RTSP input doesn't fit in that. Even with `-vcodec copy`, `avformat_find_stream_info()` instantiates the H.264 decoder to probe stream parameters, and that allocates a full DPB — at 1080p roughly 3 MB per frame times up to 16 reference frames. It dies during input open, before it ever reaches your output file.

Same board, OOM forced deliberately to show the signature you should be seeing:

```
dd invoked oom-killer: gfp_mask=0x24200ca(GFP_HIGHUSER_MOVABLE), nodemask=0, order=0, oom_score_adj=0
Out of memory: Kill process 4478 (dd) score 216 or sacrifice child
Killed process 4478 (dd) total-vm:9472kB, anon-rss:7824kB, file-rss:20kB, shmem-rss:0kB
```

Run `dmesg | grep -i "out of memory"` straight after ffmpeg dies and you'll see the same lines with `(ffmpeg)`.

**Two other problems in that command, independent of the OOM:**

1. `/mnt/sdcard/test.mp4` could never have worked. `ffmpeg-openipc` is built with `--disable-muxers --enable-muxer=flv,rtsp` (`general/package/ffmpeg-openipc/ffmpeg-openipc.mk:33`). There's no mp4/mov muxer. FLV is the only container you can write to a file here — and FLV can't carry H.265, so if your camera is set to h265 there is no usable on-box container at all.
2. `-s 640x360` is a no-op beside `-vcodec copy`. Scaling needs a re-encode, and this build has `--disable-swscale --disable-encoders --disable-filters`.

**For what you're actually after (clip to Telegram):**

- Still image: `curl -u root:12345 http://<cam>/image.jpg` — verified working, returns a real JPEG.
- Video: use majestic's own recorder (`records:` in `/etc/majestic.yaml` — `enabled`, `path`, `split`). It writes through the hardware encoder with no decode and no ffmpeg.
- Or pull the RTSP stream to a host that has RAM and do the muxing/sending there.

The camera's ~27 MB of Linux RAM is there to run majestic, not a general-purpose transcoder.

closed 2026-08-16 by widgetii