OpenIPC/firmware #1880: Ingenic T40N won't keep overlay after reboot from WebUI

tERB00, 2025-09-09:
Steps to reproduce:
1. firstboot
2. Log on to webui
3. Change password
4. Change timezone
5. Red sign appears offering a reboot for changes to apply
6. Click reboot
7. After reboot camera loses all the config changes (password, timezone, etc)

BUT! If you wont click on reboot and issue a reboot command from the console the changes are saved and the camera keeps them after reboot.

widgetii, 2026-11-10:
I tried to reproduce this on an emulated T40N running the current nightly, following your steps through the WebUI: changed the time zone, changed the password, then used the red Restart button. After the reboot both changes were still there: the new password logs in and the time zone is kept. Restarting from the console behaved the same.

One thing that can look like this: each WebUI save takes a moment, and a save sent while the previous one is still being applied can be refused, so that setting is never written at all and only seems to be lost after the restart.

Could you update to a current nightly and try again? Wait for each change to confirm it was saved before making the next one or restarting. If a setting still goes missing after a WebUI restart, please post the output of `cat /etc/TZ`, `ls -la /overlay/root/etc` and `logread | tail -100` from just before the restart.

tERB00, 2026-11-11:
Updated to the latest master+03ff937, 2026-11-10, tried several times with firstboot and time zone change and it seems to keep the settings after restart from WebUI.

closed 2026-11-11 by tERB00