OpenIPC/firmware #1884: About T40 and solar cameras

ortegafernando, 2025-09-14:
Hi, I have found several solar cameras with Ingenic T40. Does anyone know if any of them is supported? Thanks.

https://www.unifore.net/item/sqw-2.html
https://www.unifore.net/tag/ingenic-t40.html

tERB00, 2025-09-15:
T40 SOC is supported, but the rest of the camera is up to you to figure out. The wifi chip, gpio pins for the motors, ircut, white light and so on.

ortegafernando, 2025-09-17:
> T40 SOC is supported, but the rest of the camera is up to you to figure out. The wifi chip, gpio pins for the motors, ircut, white light and so on.

Thanks tERB00, I think this will be difficult for me.

Just another question. I have opened a dual camera Lenovo X6T (al least the easier len to open) and I see a INGENIC T40 SOC, but I can not know if it is T40A, N, NN or XP. How could I know that? I need to try to find serial console or something like that? Thanks

tERB00, 2025-09-17:
Yes, examining a serial console is the easiest way to determine a correct SOC.
Also dual sensor cams are not supported by Openipc at the moment.

ortegafernando, 2025-09-17:
> Yes, examining a serial console is the easiest way to determine a correct SOC. Also dual sensor cams are not supported by Openipc at the moment.

Thanks. And.... even with dual sensor ... if I flash the firmware could I use one or it will be totally useless?

widgetii, 2026-11-13:
@czsczs about the X6T's sensor "stuck on i2c and xvclk": on T40 the streamer and the sensor detection both assumed the reference board's wiring, with the sensor on I2C1, clocked from MCLK1, reset on GPIO 91. A board wired differently never sees its sensor, which matches what you describe. We reproduced exactly that on an emulated T40 with the sensor moved to I2C0.

The wiring is becoming configurable. Once these changes reach a T40 nightly, `/etc/majestic.yaml` takes:

```yaml
isp:
  sensorI2cBus: 1        # I2C adapter 0-3
  sensorMclk: 1          # sensor clock output MCLK0-2
  sensorResetPin: 91     # GPIO, -1 if not wired
  sensorPowerDownPin: -1 # GPIO, -1 if not wired
```

`load_ingenic` passes the bus and the two pins to the sensor detection (`sinfo`) too. It can only use MCLK1, so if your board uses MCLK0 or MCLK2, also set the sensor by hand: `fw_setenv sensor <name>`.

To fill those in, we need your board's wiring. The stock firmware usually shows it:

- the stock boot log and `dmesg`, especially the sensor driver's and `tx-isp` lines;
- if you can get a shell on the stock firmware, `cat /proc/cmdline`, `ls /sys/bus/i2c/devices/` and the parameters it loads the sensor module with (`cat /sys/module/sensor_*/parameters/*` or its start scripts);
- which sensor it is, if the log names it.

With those we can tell you the exact settings to try. Dual-sensor operation (both lenses at once) is a separate, larger piece of work. This would get one sensor working.