OpenIPC/firmware #1885: Unable to get ethernet to show up T20

theraspb3rry, 2025-09-16:
Hi, I have an ingenic t20 chip and i've taken the flash out and hardware flashed openipc onto it rather than doing uboot tftp shenanigans (because i accidentally corrupted the flash with a pastebuffer)

I got it booting, but cant get ethernet to show up in linux (or uboot for that matter, fails to setup the eth). attached is the uart/serial log of me messing about with it. any suggestions or debug steps to move forward? thanks.

[ipcam-1758009539.log](https://github.com/user-attachments/files/22359293/ipcam-1758009539.log)

SF: Detected W25Q64
In:    serial
Out:   serial
Err:   serial
Net:   ====>PHY not found!
gpio_request[wyze_usb_enable] = 47
gpio_request[wyze_cd_enable] = 43
gpio_request[wyze_mmc_enable] = 48
Board Net Initialization Failed
Jz4775-9161
Card did not respond to voltage select!

tERB00, 2025-09-17:
Was it working before, on the stock firmware?
It is either damaged or some very weird ethernet chip not supported by u-boot.
What is the ethernet chip? Also share some pcb photos (both sides) if you can.

theraspb3rry, 2025-09-17:
hmm, i tried to get a photo of the chip to confirm but it was really hard, haven't been able to make it out. I would've expected that if the ethernet physically comes up on the switch/computer on the otherside that the chip is fine but i guess maybe the data between it and the T20 is dead or not working for some reason.

tERB00, 2025-09-17:
You have a poe module over there - your switch might be reacting to that.

theraspb3rry, 2025-09-17:
thanks for this, i'll give it a shot when i have all the gear setup again to flash it and report back :)

widgetii, 2026-11-11:
Your PHY is fine. The log you attached shows a later boot finding it (`phy 0: 0x243-0xc54`, an IP101G), and the kernel still disabling Ethernet (`JZ GMAC disabled`).

The cause is in U-Boot. When a boot does not find a PHY, U-Boot assumes a camera without Ethernet (like the Wyze V2) and adds `nogmac` to the kernel arguments through the `extras` variable. That gets saved to flash: by U-Boot itself on a camera's first boot, or by any `saveenv` (as when you set `ethaddr`). From then on every boot passes `nogmac` to the kernel, even when the PHY answers. We reproduced exactly this on an emulated T20 and T31.

To get Ethernet back now:

```
fw_printenv extras      # shows extras=nogmac
fw_setenv extras        # clears it
reboot
```

(or `setenv extras; saveenv; reset` from the U-Boot prompt).

U-Boot is fixed in OpenIPC/u-boot-ingenic#9: a `nogmac` that U-Boot set itself is now dropped as soon as a later boot finds the PHY, while one set on purpose (Wi-Fi-only cameras use it) is kept. That protects cameras flashed with a U-Boot built from it. A camera that already saved `nogmac` still needs the `fw_setenv extras` above once.

The occasional missed PHY probe on a cold boot is a separate thing. If it keeps happening, a boot log from such a boot would help.